openwrt/openwrt #15470: Banana-pi BPI-R3 - unsupported SFP module with kernel 6.6.30 while ok with kernel 6.1.80 or below

gca31, 2024-05-12:
### Describe the bug

I bought a 2.5Gbps base T SFP module which was working well with kernel 6.1.80. It is now said to be an unsupported SFP module (no common interface modes) with kernel 6.6.30.

**Logs in kernel 6.6.30 are:**
[  297.717983] sfp sfp-2: module OEM              SFP-2.5G-T-R-RM  rev 1.0  sn 2311030046       dc 231103
[  297.727478] mt7530-mdio mdio-bus:1f sfp2: unsupported SFP module: no common interface modes
Sun May 12 23:08:13 2024 kern.info kernel: [  297.717983] sfp sfp-2: module OEM              SFP-2.5G-T-R-RM  rev 1.0  sn 2311030046       dc 231103
Sun May 12 23:08:13 2024 kern.err kernel: [  297.727478] mt7530-mdio mdio-bus:1f sfp2: unsupported SFP module: no common interface modes
[  297.767111] hwmon hwmon6: temp1_input not attached to any thermal zone
Sun May 12 23:08:13 2024 kern.info kernel: [  297.767111] hwmon hwmon6: temp1_input not attached to any thermal zone

**Logs in kernel 6.1.80 are:**
[   24.388739] sfp sfp-2: module OEM              SFP-2.5G-T-R-RM  rev 1.0  sn 2311030046       dc 231103                                                     
[   24.437948] hwmon hwmon3: temp1_input not attached to any thermal zone                                                                               
[   36.812348] mt7530-mdio mdio-bus:1f sfp2: configuring for inband/2500base-x link mode
[   36.822002] br-lan: port 5(sfp2) entered blocking state
[   36.827221] br-lan: port 5(sfp2) entered disabled state
[   36.834592] device sfp2 entered promiscuous mode                     
[   40.178602] mt7530-mdio mdio-bus:1f sfp2: Link is Up - 2.5Gbps/Full - flow control off                                                               
[   40.186631] IPv6: ADDRCONF(NETDEV_CHANGE): sfp2: link becomes ready    
[   40.193160] br-lan: port 5(sfp2) entered blocking state      
[   40.198373] br-lan: port 5(sfp2) entered forwarding state

### OpenWrt version

r26300-da0cd9d764

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Steps to reproduce

Use a 2.5G base-T SFP module. Mine comes from sinovoip store. Install latest kernel 5.15 or 6.1 Module should work. Then sysupgrade or fresh install of 6.6.30 and module is said to be unsupported (no common interface modes)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yuban10703, 2024-05-13:
https://forum.banana-pi.org/t/bpi-r4-and-sfp/16945/26

My BPI-R4 can be repaired by following this method

gca31, 2024-05-16:
It seems that only phylink.c and phylink.h files differs in openwrt kernel 6.6.30 from the 9 files in this method.
But not being an expert, I would rather request your help and the community's help to verify this and make the appropriate patches that I can test.

yuban10703, 2024-05-17:
> It seems that only phylink.c and phylink.h files differs in openwrt kernel 6.6.30 from the 9 files in this method. But not being an expert, I would rather request your help and the community's help to verify this and make the appropriate patches that I can test.

This is the path to the file

`openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.30/drivers/net/phy/phy.c `

I will not make openwrt related patches, this may have to wait for others to fix it

ch1993128, 2024-05-20:
same issue.

Headcrabed, 2024-05-27:
What is the output of ethtool -m? Maybe that’s a wrong module eeprom configuration and can be fixed by the tool here: https://forum.banana-pi.org/t/i2csfp-sfp-debugging-via-i2c/17761

gca31, 2024-05-28:
I succeeded in making this SFP module (OEM / SFP-2.5G-T-R-RM) to work with kernel 6.6.30 and this patch: 
[750-net_phy_sfp_add_support_for_more_sfp_modules.patch](https://github.com/openwrt/openwrt/files/15469697/750-net_phy_sfp_add_support_for_more_sfp_modules.patch)
Could you test this patch ?

ch1993128, 2024-05-29:
> This SFP module (OEM / SFP-2.5G-T-R-RM) seems to work with latest snapshot and this patch: [750-net_phy_sfp_add_support_for_more_sfp_modules.patch](https://github.com/openwrt/openwrt/files/15469697/750-net_phy_sfp_add_support_for_more_sfp_modules.patch) Could you test this patch ?

Thx,this patch works well on my SFP_2G5 model.

Headcrabed, 2024-05-30:
We can’t just apply this patch as some modules are using yt8821 instead of  rtl8221, and this quirk does not support yt8821 and will stop it working. More information here: https://forum.banana-pi.org/t/motorcomm-yt8821-on-sfp-module/17746/
To solve this problem, you should use the i2csfp tool mentioned in that topic to modify your module’s model name.

gca31, 2024-05-30:
Thanks to [Shengyu Qu](https://forum.banana-pi.org/u/Stat_headcrabed) from banana-pi forum:

> We can’t just apply this patch as some modules are using yt8821 instead of rtl8221, and this quirk does not support yt8821 and will stop it working. More information here: [...]

So for the time being I'm closing this issue, awaiting for a proper solution

closed 2024-05-30 by gca31

Headcrabed, 2024-05-30:
> Thanks to [Shengyu Qu](https://forum.banana-pi.org/u/Stat_headcrabed) from banana-pi forum:
>
>
> So for the time being I'm closing this issue, awaiting for a proper solution

I deleted my post because the problem shouldn't behave like what has beed reported in this issue. If the module is not in the quirk, nothing should happen, it would just behave like a normal sfp module. We might need more investigation into it.

Headcrabed, 2024-05-30:
@gca31 Also could you post the output of ```ethtool -m```  from ```ethtool-full``` package for your module?

gca31, 2024-05-30:
> @gca31 Also could you post the output of `ethtool -m` from `ethtool-full` package for your module?

I've change the vendor PN of the module. Before it was SFP-2.5G-T-R-RM. Now it is SFP-2.5G-T which is supported by kernel 6.6

root@BPI-R3:# ethtool -m sfp2
	Identifier                                : 0x03 (SFP)
	Extended identifier                       : 0x04 (GBIC/SFP defined by 2-wire interface ID)
	Connector                                 : 0x07 (LC)
	Transceiver codes                         : 0x00 0x01 0x00 0x00 0x00 0x00 0x02 0x00 0x00
	Transceiver type                          : SONET: OC-48, short reach
	Encoding                                  : 0x05 (SONET Scrambled)
	BR, Nominal                               : 2500MBd
	Rate identifier                           : 0x00 (unspecified)
	Length (SMF,km)                           : 0km
	Length (SMF)                              : 0m
	Length (50um)                             : 300m
	Length (62.5um)                           : 200m
	Length (Copper)                           : 0m
	Length (OM3)                              : 0m
	Laser wavelength                          : 850nm
	Vendor name                               : OEM
	Vendor OUI                                : 00:00:00
	Vendor PN                                 : SFP-2.5G-T
	Vendor rev                                : 1.0
	Option values                             : 0x00 0x1a
	Option                                    : RX_LOS implemented
	Option                                    : TX_FAULT implemented
	Option                                    : TX_DISABLE implemented
	BR margin, max                            : 0%
	BR margin, min                            : 0%
	Vendor SN                                 : 2311030046
	Date code                                 : 231103
	Optical diagnostics support               : Yes
	Laser bias current                        : 6.000 mA
	Laser output power                        : 0.5000 mW / -3.01 dBm
	Receiver signal average optical power     : 0.4000 mW / -3.98 dBm
	Module temperature                        : 42.99 degrees C / 109.39 degrees F
	Module voltage                            : 3.2987 V
	Alarm/warning flags implemented           : Yes
	Laser bias current high alarm             : Off
	Laser bias current low alarm              : Off
	Laser bias current high warning           : Off
	Laser bias current low warning            : Off
	Laser output power high alarm             : Off
	Laser output power low alarm              : Off
	Laser output power high warning           : Off
	Laser output power low warning            : Off
	Module temperature high alarm             : Off
	Module temperature low alarm              : Off
	Module temperature high warning           : Off
	Module temperature low warning            : Off
	Module voltage high alarm                 : Off
	Module voltage low alarm                  : Off
	Module voltage high warning               : Off
	Module voltage low warning                : Off
	Laser rx power high alarm                 : Off
	Laser rx power low alarm                  : Off
	Laser rx power high warning               : Off
	Laser rx power low warning                : Off
	Laser bias current high alarm threshold   : 15.000 mA
	Laser bias current low alarm threshold    : 1.000 mA
	Laser bias current high warning threshold : 13.000 mA
	Laser bias current low warning threshold  : 2.000 mA
	Laser output power high alarm threshold   : 1.9952 mW / 3.00 dBm
	Laser output power low alarm threshold    : 0.1584 mW / -8.00 dBm
	Laser output power high warning threshold : 1.5848 mW / 2.00 dBm
	Laser output power low warning threshold  : 0.1778 mW / -7.50 dBm
	Module temperature high alarm threshold   : 95.00 degrees C / 203.00 degrees F
	Module temperature low alarm threshold    : -50.00 degrees C / -58.00 degrees F
	Module temperature high warning threshold : 90.00 degrees C / 194.00 degrees F
	Module temperature low warning threshold  : -45.00 degrees C / -49.00 degrees F
	Module voltage high alarm threshold       : 3.6000 V
	Module voltage low alarm threshold        : 3.0000 V
	Module voltage high warning threshold     : 3.5000 V
	Module voltage low warning threshold      : 3.1000 V
	Laser rx power high alarm threshold       : 1.1220 mW / 0.50 dBm
	Laser rx power low alarm threshold        : 0.0199 mW / -17.01 dBm
	Laser rx power high warning threshold     : 1.0000 mW / 0.00 dBm
	Laser rx power low warning threshold      : 0.0223 mW / -16.52 dBm
root@BPI-R3:~#

Headcrabed, 2024-05-30:
Actually SFP-2.5G-T shouldn't be supported either....

Headcrabed, 2024-05-30:
@gca31  Can you change back to a module model name that is not supported by that fixup, re-plug the module and give me the output of ```ethtool sfp2```?

Headcrabed, 2024-05-30:
@gca31 Ok, No need for that, I found the issue, your module's BR, Nominal value is wrong. for 2500base-T module, it should be 3100(3125 actually), but it is set to 2500, that is the real problem. If you set that to 3100, it would work without changing module name.

CpServiceSpb, 2025-05-30:
Hi.
The same issue with HSGQ (ODI) RTK based xpon sfp.
That is there is no link.