openwrt/openwrt #15476: compex wpq864 Unable to mount root fs

stargieg, 2024-05-13:
### Describe the bug

Boot openwrt-ipq806x-generic-compex_wpq864-initramfs-fit-uImage.itb via tftpboot.
run sysupgrade -n openwrt-ipq806x-generic-compex_wpq864-squashfs-nand-sysupgrade.bin
next reboot halted with kernelpanic "Unable to mount root fs"

```
[    0.282485] nand: device found, Manufacturer ID: 0x2c, Chip ID: 0xaa
[    0.285837] qcom-pcie 1b900000.pci: host bridge /soc/pci@1b900000 ranges:
[    0.294021] nand: Micron MT29F2G08ABBEAH4
[    0.305370] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    0.311356] qcom-pcie 1b900000.pci:       IO 0x0035e00000..0x0035e0ffff -> 0x0000000000
[    0.317738] 23 fixed-partitions partitions found on MTD device qcom_nand.0
[    0.321922] qcom-pcie 1b900000.pci:      MEM 0x0032000000..0x0035dfffff -> 0x0032000000
[    0.325570] Creating 23 MTD partitions on "qcom_nand.0":
[    0.350355] 0x000000000000-0x000000040000 : "0:SBL1"
[    0.355004] qcom-pcie 1b700000.pci: iATU: unroll F, 8 ob, 8 ib, align 4K, limit 4G
[    0.363944] 0x000000040000-0x000000180000 : "0:MIBIB"
[    0.379876] 0x000000180000-0x0000002c0000 : "0:SBL2"
[    0.383172] qcom-pcie 1b500000.pci: iATU: unroll F, 8 ob, 8 ib, align 4K, limit 4G
[    0.400696] 0x0000002c0000-0x000000540000 : "0:SBL3"
[    0.462279] 0x000000540000-0x000000660000 : "0:DDRCONFIG"
[    0.467918] qcom-pcie 1b900000.pci: iATU: unroll F, 8 ob, 8 ib, align 4K, limit 4G
[    0.475056] 0x000000660000-0x000000780000 : "0:SSD"
[    1.037951] 0x000000780000-0x000000a00000 : "0:TZ"
[    1.044947] 0x000000a00000-0x000000c80000 : "0:RPM"
[    1.049955] 0x000000c80000-0x000001180000 : "0:APPSBL"
[    1.058911] 0x000001180000-0x000001200000 : "0:APPSBLENV"
[    1.060435] 0x000001200000-0x000001340000 : "0:ART"
[    1.066213] 0x000001340000-0x000005340000 : "ubi"
[    1.174931] 0x000005340000-0x0000053a0000 : "0:BOOTCONFIG"
[    1.176303] 0x0000053a0000-0x0000054e0000 : "0:SBL2_1"
[    1.182059] 0x0000054e0000-0x000005760000 : "0:SBL3_1"
[    1.189360] 0x000005760000-0x000005880000 : "0:DDRCONFIG_1"
[    1.192090] 0x000005880000-0x0000059a0000 : "0:SSD_1"
[    1.197646] 0x0000059a0000-0x000005c20000 : "0:TZ_1"
[    1.205235] 0x000005c20000-0x000005ea0000 : "0:RPM_1"
[    1.210132] 0x000005ea0000-0x000005f00000 : "0:BOOTCONFIG1"
[    1.211600] 0x000005f00000-0x000006400000 : "0:APPSBL_1"
[    1.224646] 0x000006400000-0x00000a400000 : "ubi_1"
[    1.331237] 0x00000a400000-0x000010000000 : "unused"
[    1.381408] qcom-pcie 1b700000.pci: Phy link never came up
[    1.382032] qcom-pcie 1b700000.pci: PCI host bridge to bus 0001:00
[    1.385929] pci_bus 0001:00: root bus resource [bus 00-ff]
[    1.391978] pci_bus 0001:00: root bus resource [io  0x10000-0x1ffff] (bus address [0x0000-0xffff])
[    1.397535] pci_bus 0001:00: root bus resource [mem 0x2e000000-0x31dfffff]
[    1.406583] pci 0001:00:00.0: [17cb:0101] type 01 class 0x060400
[    1.413417] pci 0001:00:00.0: supports D1
[    1.419495] pci 0001:00:00.0: PME# supported from D0 D1 D3hot
[    1.429942] PCI: bus0: Fast back to back transfers disabled
[    1.430460] PCI: bus1: Fast back to back transfers enabled
[    1.434528] pci 0001:00:00.0: PCI bridge to [bus 01-ff]
[    1.441085] qcom-pcie 1b500000.pci: Phy link never came up
[    1.445461] qcom-pcie 1b500000.pci: PCI host bridge to bus 0000:00
[    1.450660] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.456890] pci_bus 0000:00: root bus resource [io  0x0000-0xffff]
[    1.462285] pci_bus 0000:00: root bus resource [mem 0x08000000-0x0fdfffff]
[    1.468633] pci 0000:00:00.0: [17cb:0101] type 01 class 0x060400
[    1.474670] random: crng init done
[    1.475555] pci 0000:00:00.0: supports D1
[    1.484753] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    1.495262] PCI: bus0: Fast back to back transfers disabled
[    1.495815] PCI: bus1: Fast back to back transfers enabled
[    1.499980] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    1.506018] qcom-pcie 1b900000.pci: Phy link never came up
[    1.510872] qcom-pcie 1b900000.pci: PCI host bridge to bus 0002:00
[    1.516229] pci_bus 0002:00: root bus resource [bus 00-ff]
[    1.522271] pci_bus 0002:00: root bus resource [io  0x20000-0x2ffff] (bus address [0x0000-0xffff])
[    1.527885] pci_bus 0002:00: root bus resource [mem 0x32000000-0x35dfffff]
[    1.536869] pci 0002:00:00.0: [17cb:0101] type 01 class 0x060400
[... 188 lines cut ...]

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Compex WPQ864

### Image kind

Official downloaded image

### Steps to reproduce

tftpboot openwrt-ipq806x-generic-compex_wpq864-initramfs-fit-uImage.itb
sysupgrade -n openwrt-ipq806x-generic-compex_wpq864-squashfs-nand-sysupgrade.bin

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stargieg, 2024-05-13:
b03f60dc4a1a08741325f5b71abc0466af457099 Fix this Bug

namiltd, 2024-05-13:
Make Pull Request for your patch.