openwrt/openwrt #15477: compex wpq864 random ether mac addr

stargieg, 2024-05-13:
### Describe the bug

After every reboot the device has another random mac addr.

### OpenWrt version

r26300-da0cd9d764

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Compex WPQ864

### Image kind

Official downloaded image

### Steps to reproduce

reboot multiple times

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stargieg, 2024-05-13:
0bdad43c1976bfbf8c4151a29e5404533ac597f3 Fix this bug

namiltd, 2024-05-13:
Make Pull Request for your patch.