openwrt/openwrt #15478: compex wpq864 ubootenv config is missing

stargieg, 2024-05-13:
### Describe the bug

After firstboot the /etc/config/ubootenv was empty and fw_printenv dose not work.

### OpenWrt version

r26300-da0cd9d764

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Compex WPQ864

### Image kind

Official downloaded image

### Steps to reproduce

sysupgrade -n

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

stargieg, 2024-05-13:
b03f60dc4a1a08741325f5b71abc0466af457099 Fix this Bug

namiltd, 2024-05-13:
Make Pull Request for your patch.