openwrt/openwrt #15485: Can't connect to WAN after sysupgrade on Spectrum SAX1V1K (Askey RT5010W)

dangerweenie, 2024-05-14:
### Describe the bug

After installing OpenWRT from Meister's original work [here](https://github.com/MeisterLone/Askey-RT5010W-D187-REV6/tree/master/OpenWrt-r22241), sysupgrade to latest snapshot at OpenWRT firmware selector results in no WAN connection and Luci does not install (despite being included in the build)

### OpenWrt version

SNAPSHOT (r26302-4f87a4d84f)

### OpenWrt release

SNAPSHOT (r26302-4f87a4d84f)

### OpenWrt target/subtarget

SNAPSHOT (r26302-4f87a4d84f)

### Device

Askey RT5010W-D187

### Image kind

Official downloaded image

### Steps to reproduce

1. Install OpenWRT from [here](https://github.com/MeisterLone/Askey-RT5010W-D187-REV6/tree/master/OpenWrt-r22241)
2. Factory hard reset 10 seconds
3. Power cycle
4. Login to router via Luci
5. Flash firmware (do not keep settings)
5b. Also via SSH sysupgrade -v -n -O /tmp/sysupgrade.bin
6. Factory hard reset 10 seconds
7. Power cycle

### Actual behaviour

No WAN connection, no Luci

### Expected behaviour

WAN connection, Luci access

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-05-14:
Without full logs its impossible to guess

namiltd, 2024-05-14:
Try new snapshot generated today. Specify exactly what image you use in sysupgrade, attach logs.

dangerweenie, 2024-05-15:
closing and migrating to

#15493

closed 2024-05-15 by dangerweenie