openwrt/openwrt #15488: v4l-utils: ERROR: Unknown variable "ir_bpf_enabled".

alex9434, 2024-05-14:
### Describe the bug

Building for BCM2711 (Raspberry Pi 4). I get the following error although v3-utils is not enabled in make menuconfig:
./../../../build_dir/target-aarch64_cortex-a72_musl/v4l-utils-1.26.1/meson.build:408:32: ERROR: Unknown variable "ir_bpf_enabled".

### OpenWrt version

OpenWrt Snapshot 

### OpenWrt release

Snapshot

### OpenWrt target/subtarget

BCM2711

### Device

Raspbery Pi 4B

### Image kind

Official downloaded image

### Steps to reproduce

Build on x64 with official build image. Kernel video drivers enabled.

### Actual behaviour

Build stops with error message:
./../../../build_dir/target-aarch64_cortex-a72_musl/v4l-utils-1.26.1/meson.build:408:32: ERROR: Unknown variable "ir_bpf_enabled".

### Expected behaviour

No error message

### Diffconfig

```text
Posting only kmod packages because of space limitations in this form.

CONFIG_PACKAGE_kmod=y
# CONFIG_PACKAGE_kmod-3c59x is not set
# CONFIG_PACKAGE_kmod-6lowpan is not set
# CONFIG_PACKAGE_kmod-8139cp is not set
# CONFIG_PACKAGE_kmod-8139too is not set
CONFIG_PACKAGE_kmod-ac97=y
# CONFIG_PACKAGE_kmod-alx is not set
# CONFIG_PACKAGE_kmod-aoe is not set
# CONFIG_PACKAGE_kmod-ar5523 is not set
# CONFIG_PACKAGE_kmod-arptables is not set
CONFIG_PACKAGE_kmod-asn1-decoder=y
CONFIG_PACKAGE_kmod-asn1-encoder=y
# CONFIG_PACKAGE_kmod-at86rf230 is not set
# CONFIG_PACKAGE_kmod-ata-ahci is not set
# CONFIG_PACKAGE_kmod-ata-artop is not set
CONFIG_PACKAGE_kmod-ata-core=y
# CONFIG_PACKAGE_kmod-ata-nvidia-sata is not set
# CONFIG_PACKAGE_kmod-ata-pdc202xx-old is not set
# CONFIG_PACKAGE_kmod-ata-piix is not set
# CONFIG_PACKAGE_kmod-ata-sil is not set
# CONFIG_PACKAGE_kmod-ata-sil24 is not set
# CONFIG_PACKAGE_kmod-ata-via-sata is not set
CONFIG_PACKAGE_kmod-ath=y
# CONFIG_PACKAGE_kmod-ath10k is not set
# CONFIG_PACKAGE_kmod-ath10k-ct is not set
# CONFIG_PACKAGE_kmod-ath10k-ct-smallbuffers is not set
# CONFIG_PACKAGE_kmod-ath10k-smallbuffers is not set
[... 785 lines cut ...]
CONFIG_PACKAGE_kmod-video-gspca-stv06xx=y
CONFIG_PACKAGE_kmod-video-gspca-sunplus=y
CONFIG_PACKAGE_kmod-video-gspca-t613=y
CONFIG_PACKAGE_kmod-video-gspca-tv8532=y
CONFIG_PACKAGE_kmod-video-gspca-vc032x=y
CONFIG_PACKAGE_kmod-video-gspca-zc3xx=y
CONFIG_PACKAGE_kmod-video-mem2mem=y
CONFIG_PACKAGE_kmod-video-pwc=y
# CONFIG_PACKAGE_kmod-video-tw686x is not set
CONFIG_PACKAGE_kmod-video-uvc=y
CONFIG_PACKAGE_kmod-video-videobuf2=y
# CONFIG_PACKAGE_kmod-vmxnet3 is not set
CONFIG_PACKAGE_kmod-vxlan=y
# CONFIG_PACKAGE_kmod-w1 is not set
# CONFIG_PACKAGE_kmod-wdt-sp805 is not set
# CONFIG_PACKAGE_kmod-wil6210 is not set
CONFIG_PACKAGE_kmod-wireguard=y
# CONFIG_PACKAGE_kmod-wl12xx is not set
# CONFIG_PACKAGE_kmod-wl18xx is not set
# CONFIG_PACKAGE_kmod-wlcore is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

alex9434, 2024-05-14:
Yes, I used
```
git clone https://git.openwrt.org/openwrt/openwrt.git
cd openwrt
git pull
```
 according to https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem

neheb, 2024-05-17:
Wrong repo.

https://github.com/openwrt/packages/issues/24115

jpsollie, 2024-05-27:
I can confirm the patch specified in the upstream bug resport (this one: https://git.linuxtv.org/v4l-utils.git/commit/meson.build?id=fee976dcfe3be03cc72c099f9ed3bac938a34b58) fixes the issue

neheb, 2024-06-15:
@aparcar this was fixed. Please close.

closed 2024-06-17 by aparcar