openwrt/openwrt #15489: OpenWrt cannot install ca-certificates

yeyos1, 2024-05-14:
### Describe the bug

Please tell me how to install the system root certificate in the openwrt system. My ca certificate is in pem format.

### OpenWrt version

R23.8.5-D (2023-09-27)

### OpenWrt release

R23.8.5-D (2023-09-27)

### OpenWrt target/subtarget

R23.8.5-D (2023-09-27)

### Device

x86_64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2024-05-14:
Use the mailing list, IRC channels or forum for support requests.

closed 2024-05-14 by jow-

ynezz, 2024-05-14:
> Official downloaded image

Where did you downloaded such image?