openwrt/openwrt #15491: openvpn dco doesn't seem to work on the client side.

Xeon64, 2024-05-14:
### Describe the bug

OpenVPN DCO doesn't seem to work when using openvpn as a client on openwrt. I did not try running openwrt with openvpn server to see if dco works only client.

Kernel Version：5.15.137

Model：Hewlett-Packard HP Compaq Elite 8300 SFF

Architecture：Intel(R) Core(TM) i5-3470 CPU @ 3.20GHz

Firmware Version：OpenWrt 23.05.2 r23630-842932a63d

openvpn-openssl	2.5.8-5	~227.83 KiB	
kmod-ovpn-dco	5.15.137+2022-10-23-d1d53564-3	~25.58 KiB	
luci-app-openvpn	git-24.108.03100-3129664	~12.66 KiB	

lsmod shows dco kmod loaded.

The recommended "ip show/status" command that openvpn recommends to see if dco is working does not show up as DCO being used.  

top / ps doesn't show more than one thread being used.. while on the ubunutu KVM openvpn server (not openwrt) it shows dco loaded in lsmod and enabled by ip show command and multiple threads running for dco shows up under top/ps .

### OpenWrt version

 23.05.2 r23630-842932a63d

### OpenWrt release

 23.05.2 r23630-842932a63d

### OpenWrt target/subtarget

openvpn

### Device

x86 64-bit

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-14:
First try on version 23.05.3 and on snapshot.

Xeon64, 2024-05-14:
> First try on version 23.05.3 and on snapshot. See https://github.com/openwrt/openwrt/pull/153699 for information about other users' experiences with dco.

That link is 404 not found.  I'll try the newer version when I get a chance.

namiltd, 2024-05-14:
Try again.

MaxSKash, 2024-05-15:
Он вообще когда-нибудь работал? Особенно с OpenVPN 2.5...
В моментальном снимке он тоже не работает. Он запускается, но при попытке назначить IP происходит сбой. У меня получилось его запустить, если скомпилировать OpenWrt с этими рекомендациями: https://github.com/openwrt/packages/issues/24050

```
2024-05-15 14:39:38 net_route_v4_best_gw query: dst 0.0.0.0
2024-05-15 14:39:38 net_route_v4_best_gw result: via 10.19.126.1 dev eth1
2024-05-15 14:39:38 net_iface_new: add vpn-pim type ovpn-dco
2024-05-15 14:39:38 DCO device vpn-pim opened
2024-05-15 14:39:38 net_iface_mtu_set: mtu 1500 for vpn-pim
2024-05-15 14:39:38 net_iface_up: set vpn-pim up
2024-05-15 14:39:38 net_addr_v4_add: 192.168.56.55/25 dev vpn-pim
2024-05-15 14:39:38 /usr/libexec/openvpn-hotplug up Office vpn-pim 1500 0 192.168.56.55 255.255.255.128 init
2024-05-15 14:39:38 Data Channel: cipher 'AES-256-GCM', peer-id: 3
2024-05-15 14:39:38 Timers: ping 10, ping-restart 60
```

Xeon64, 2024-05-15:
Да,  with Gbps +++ networks and CPUs running in adaption mode idling in low mhz by default it's very important to get DCO to work. Even when CPUs are set to performance mode a single gigabit connection can saturate a single core on a encrypted openvpn connection.

> Он вообще когда-нибудь работал? Особенно с OpenVPN 2.5... В моментальном снимке он тоже не работает. Он запускается, но при попытке назначить IP происходит сбой. У меня получилось его запустить, если скомпилировать OpenWrt с этими рекомендациями: [openwrt/packages#24050](https://github.com/openwrt/packages/issues/24050)

csharper2005, 2024-05-15:
OpenVPN in OpenWrt by default without DCO support. So, you should build your own package with SDK.

Xeon64, 2024-05-15:
> OpenVPN in OpenWrt by default without DCO support. So, you should build your own package with SDK.

If that's true why can you install a precompiled kmod of DCO?  When I have chance I do plan on fixing it.. but doesn't mean it shouldn't be working for others as much as possible.

csharper2005, 2024-05-15:
> If that's true why can you install a precompiled kmod of DCO?

Why not? You can install kmod. OpenVPN with default build configuration doesn't use it. I don’t know if this is a bug or someone’s conscious decision.

Xeon64, 2024-05-15:
> > If that's true why can you install a precompiled kmod of DCO?
> 
> Why not? You can install kmod. OpenVPN with default build configuration doesn't use it. I don’t know if this is a bug or someone’s conscious decision.

Why not? That should be self evident.. If DCO support isn't intended to work then there shouldn't be a DCO module and this thread wouldn't be here to argue about such non-sense. At the very least it should have been marked as non-working in the module or openvpn package description.

csharper2005, 2024-05-17:
OpenWrt SNAPSHOT, 6.1 kernel. x86-64, OpenVPN server, dco - works fine
OpenWrt SNAPSHOT, 6.6 kernel. ramips/mt7621, OpenVPN client, no-dco - works fine

OpenWrt SNAPSHOT, 6.6 kernel. ramips/mt7621, OpenVPN client, dco - unusable. most sites don't open.

Xeon64, 2024-05-17:
My only concern at the moment is client config on x86-64..    are you defining "works fine" as using multiple threads/cores or just connecting to sites? The versions I've used.. everything works except for using multiple threads which is the whole purpose of DCO.

csharper2005, 2024-05-17:
> are you defining "works fine" as using multiple threads/cores or just connecting to sites?

Just connecting to web sites. Server - single core VM. The issue described above and yours are completely different. Sorry for offtopic.

Xeon64, 2024-05-18:
This could be one of the issues why DCO is not working.

configure: error: libnl-genl-3.0 package not found or too old. Is the development package and pkg-config (/usr/bin/pkg-config) installed? Must be version 3.4.0 or newer for DCO

Xeon64, 2024-05-19:
I just installed the lastest snapshot May 18, 2024

I can confirm DCO still doesn't work.

You can ignore my comment about the squashfs.. yes I did install the correct non-squash ext4 image.. but it wasn't the image that caused the problem.. it was reinstalling the packages via  "opkg install $(cat /etc/backup/installed.txt) not sure which package caused it.. but this process shouldn't exist on the X86 version of openwrt.. updates should work like most other linux distros in my opinion. The upgrade firmware image page shouldn't exist in luci on x86.. It would be nice to upgrade and downgrade seamlessly between stable and snapshot releases.

Does the openwrt kernel support kernel threads aka "kworkers"?  I don't see any listed in the typical (top / ps) command output.  If not I would assume that would be a number one reason why DCO is not working correctly..  the standard openvpn DCO kernel module depends on kworkers found on most linux distro kernels to work correctly.

csharper2005, 2024-05-19:
Of course. `[DCO]` is disabled by default in openvpn package.
https://github.com/openwrt/packages/blob/master/net/openvpn/Config-mbedtls.in#L38

OpenVPN with dco support looks like this:
` OpenVPN 2.6.10 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]`

Xeon64, 2024-05-19:
> Of course. `[DCO]` is disabled by default in openvpn package. https://github.com/openwrt/packages/blob/master/net/openvpn/Config-mbedtls.in#L38
> 
> OpenVPN with dco support looks like this: ` OpenVPN 2.6.10 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]`

Where do you see that?  I mean there is still a kmod ovpn dco module.. which shouldn't exist if dco is not working..

openssl , mbedtls ,wolfssl version all have this in their config file..

Which is only disable DCO if iproute2 is enabled by default which it is not.

	depends on !OPENVPN_mbedtls_ENABLE_IPROUTE2

! = NOT BEING ENABLED  unless I'm reading that wrong.

config OPENVPN_mbedtls_ENABLE_IPROUTE2
	bool "Enable support for iproute2"
	default n

config OPENVPN_mbedtls_ENABLE_DCO
	depends on !OPENVPN_mbedtls_ENABLE_IPROUTE2
	bool "Enable support for data channel offload"
	default n if OPENVPN_mbedtls_ENABLE_IPROUTE2
	help
	  enable data channel offload support
	  using the ovpn-dco-v2 kernel module

csharper2005, 2024-05-19:
@Xeon64 please show OpenVPN version string from the OpenWrt system log. Is there `[DCO]` capability there?

Xeon64, 2024-05-19:
> @Xeon64 please show OpenVPN version string from the OpenWrt system log. Is there `[DCO]` capability there?

Show me the  package description that DCO shouldn't work.. ? I may have overlooked it. If you included a DCO module in your distro.. DCO should work. Else what is the point of wasting space and confusing the end user? My whole point is that openvpn is not compiled correctly with DCO support.. so no it wouldn't show up with the [DCO] flag.. which goes back to what's the point of having a DCO kernel module that can't be used?

root@myrouter:~# openvpn --version
OpenVPN 2.5.8 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
library versions: OpenSSL 3.0.13 30 Jan 2024, LZO 2.10
Originally developed by James Yonan
Copyright (C) 2002-2022 OpenVPN Inc <sales@openvpn.net>
root@myrouter:~#

csharper2005, 2024-05-19:
@Xeon64 I agree with you, this needs to be fixed. Until this is done, rebuilding the openvpn package is the only workaround.

Xeon64, 2024-05-19:
I couldn't even get it to work with a rebuild.. if the module code to dco is stock and if openwrt kernel doesn't support kworker scheduler  I don't see any amount of rebuilding that's going to work.. without major modifications that is..  I think kworkers originally showed up in the 2.6.x  linux kernel .. but depending on openwrt .config and how much kernel code trimming openwrt has done.. kworkers may not work at all..   Perhaps openwrt developers should consider developing a luci/openwrt type package on top of other major linux distros? Where space isn't a issue with a squashfs/flashrom. It would be a openwrt interface but with compiled packages from a full linux distro..  because at this point I'm highly considering do that myself the old school way without the help of a gui.. just iptables, /etc/ ,  systemctl, and kernel magic.

csharper2005, 2024-05-19:
@Xeon64 I didn't try dco on 23.05 branch. Speed doubled on snapshot & mt7621 with dco.

Xeon64, 2024-05-19:
> @Xeon64 I didn't try dco on 23.05 branch. Speed doubled on snapshot & mt7621 with dco.

Yeh but was that because DCO actually was working using two cores for data encryption or was it because another factor such  kernel latency?  The mt7621 is only a dual core mips.. I doubt the nics that it run are very high end.. as it just wouldn't be able to handle throughput except direct nic engine to nic engine.  You have openvpn which uses 1 core/thread and you have the dco module that uses 1 core/thead even not working.. so thats 2 cores/threads.. but I'm talking about DCO using more than 1 thread + 1 openvpn thread on heavy connections.

Xeon64, 2024-05-22:
Summary: DCO still doesn't work.
Snapshot version = 05/22/2024
So I tried the latest snapshot again here are my findings..  
The default snapshot build

kmod-ovpn-dco  = loads in kernel no errors. 

openvpn-openssl package = works fine but not multi-threaded

compiled SDK version of openvpn-openssl with DCO support enabled (using precompiled snapshot dco kernel module) =  kworkers show up !  3 threads on a 4 core system. I was so excited until I found out it still doesn't work.  The dco kworkers (aka ovp)  behave nothing like on a standard distro with dco installed..  DCO is still useless.. performance is no better with or without it installed.. :(    

compiled SDK package version with compiled SDK openvpn DCO kernel module =  kworkers show up but still doesn't work.. performance no better then any other version with dco loaded or not.

vochong, 2024-05-25:
@Xeon64

Can you please add your comments to this bug:   https://github.com/openwrt/packages/issues/24050