openwrt/openwrt #15493: No WAN link of Spectrum SAX1V1K

dangerweenie, 2024-05-14:
### Describe the bug

Unable to grab a WAN link after flashing OpenWRT to Spectrum SAX1V1K (based on Askey RT5010W-D187-REV6 hardware)

### OpenWrt version

r26307-0868268c9f

### OpenWrt release

 SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Spectrum SAX1V1K

### Image kind

Official downloaded image

### Steps to reproduce

1. Boot recovery.img via TFTP
2. SCP sysupgrade.img 
3. Flash via serial /SSH: `sysupgrade -F -n -v /tmp/sysupgeade.img`

### Actual behaviour

Router boots, Luci is accessible, no WAN

### Expected behaviour

WAN link

### Additional info

Flashing image from [Meister's Github](https://github.com/MeisterLone/Askey-RT5010W-D187-REV6/tree/master/OpenWrt-r22241) results in a solid WAN link.

configs here:

`config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdd3:9347:3508::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option device 'wan'
        option proto 'dhcp'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'

root@OpenWrt:~# cat /etc/config/dhcp

config dnsmasq
        option domainneeded '1'
        option boguspriv '1'
        option filterwin2k '0'
        option localise_queries '1'
        option rebind_protection '1'
        option rebind_localhost '1'
        option local '/lan/'
        option domain 'lan'
        option expandhosts '1'
        option nonegcache '0'
        option cachesize '1000'
        option authoritative '1'
        option readethers '1'
        option leasefile '/tmp/dhcp.leases'
        option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
        option nonwildcard '1'
        option localservice '1'
        option ednspacket_max '1232'
        option filter_aaaa '0'
        option filter_a '0'

config dhcp 'lan'
        option interface 'lan'
        option start '100'
        option limit '150'
        option leasetime '12h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'

config dhcp 'wan'
        option interface 'wan'
        option ignore '1'

config odhcpd 'odhcpd'
        option maindhcp '0'
        option leasefile '/tmp/hosts/odhcpd'
        option leasetrigger '/usr/sbin/odhcpd-update'
        option loglevel '4'`

will update with the Diffconfig after reflashing the working custom firmware from Github.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-14:
Attach kernel log (dmesg).

robimarko, 2024-05-14:
Please, include a full log

dangerweenie, 2024-05-15:
this is the `logread` from yesterday (currently on working firmware, happy to flash back to the official (non-WAN working) snapshot to collect more data if that helps;.

dangerweenie, 2024-05-15:
> I would also like the logs from the version with working WAN (dmesg).

thanks for following up - i'll go grab it momentarily.

do you know a quick way to strip out MAC adddresses?  Maybe a sublime text macro or something?

dangerweenie, 2024-05-15:
> I would also like the logs from the version with working WAN (dmesg).

> I would also like the logs from the version with working WAN (dmesg).

Here is the kernel log for the **working** firmware ([here](https://github.com/MeisterLone/Askey-RT5010W-D187-REV6/tree/master/OpenWrt-r22241))

A couple things about the router when using this (WORKING) firmware:

1. There are warnings about legacy firewall rules conflicting (even after factory reset and `cp /rom/config/network/firewall /etc/config/network/firewall (or whatever the actual location is)`, so it seems like this FW is custom (it is the only working firmware I've been able to flash, and it comes from this Github (which ultimately led to the PR for this unit).  I will attach `cat /etc/config/firewall` below

2. While this FW has been the only one that appears to provide a funcitonal WAN link to the upstream router, the WAN does disconnect for about 1 minute, randomly, every few hours.  I have been unable to parse anything in either the system or kernel log that correspond to this event.

Kernel log for working firmware:

`

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.15.98 (ubuntu@DESKTOP-F7F31J2) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.2.0 r22236-a6fe02c018) 12.2.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Thu Mar 9 04:14:40 2023
[    0.000000] Machine model: Askey RT5010W-D187
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
[    0.000000]   node   0: [mem 0x0000000051100000-0x00000000bfffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x00000000bfffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 17 pages/cpu s30360 r8192 d31080 u69632
[    0.000000] pcpu-alloc: s30360 r8192 d31080 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 516096
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 mmc_mid=0x15 boot_signedimg mmc_mid=0x15 boot_signedimg mmc_mid=0x15 boot_signedimg root=/dev/mmcblk0p20 rootwait
[    0.000000] Unknown kernel command line parameters "boot_signedimg boot_signedimg boot_signedimg mmc_mid=0x15", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1916880K/2097152K available (7488K kernel code, 844K rwdata, 2052K rodata, 1216K init, 273K bss, 180272K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000118] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000134] pid_max: default: 32768 minimum: 301
[    0.000289] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000309] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001904] rcu: Hierarchical SRCU implementation.
[    0.002079] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.002593] smp: Bringing up secondary CPUs ...
[    0.003184] Detected VIPT I-cache on CPU1
[    0.003246] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.003864] Detected VIPT I-cache on CPU2
[    0.003904] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.004481] Detected VIPT I-cache on CPU3
[    0.004518] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.004584] smp: Brought up 1 node, 4 CPUs
[... 445 lines cut ...]
#       option dest_port        22
#       option proto            tcp

### FULL CONFIG SECTIONS
#config rule
#       option src              lan
#       option src_ip   192.168.45.2
#       option src_mac  00:11:22:33:44:55
#       option src_port 80
#       option dest             wan
#       option dest_ip  194.25.2.129
#       option dest_port        120
#       option proto    tcp
#       option target   REJECT

#config redirect
#       option src              lan
#       option src_ip   192.168.45.2
#       option src_mac  00:11:22:33:44:55
#       option src_port         1024
#       option src_dport        80
#       option dest_ip  194.25.2.129
#       option dest_port        120
#       option proto    tcp
root@routey:~#`

dangerweenie, 2024-05-15:
> Please, include a full log

sorry, accidentally closed with my previous reply.

log is posted above.

robimarko, 2024-05-15:
Please use code blocks in future to post logs.

When you say no WAN link, you mean you connect to the WAN port and there is no linkup or traffic?
Please also post `ethtool wan` output, and ideally also install mdio tools and post `mdio 9*` output

namiltd, 2024-05-15:
Paste files as an attachment or in between:
\```
text
\```

dangerweenie, 2024-05-15:
> Please use code blocks in future to post logs.
> 
> When you say no WAN link, you mean you connect to the WAN port and there is no linkup or traffic? Please also post `ethtool wan` output, and ideally also install mdio tools and post `mdio 9*` output

Sorry, I tried to use codeblocks for both - perhaps there were some codeblock escape characters in the log itself?

I am going to flash to the latest snapshot now (and will use onboard WiFI instead of ethernet).

I am new to Github / OpenWRT (not completely new, but new enough that I stumble a bit with posting protocol and issue sharing - thank you for bearing with me).  

This router has an interesting history, consolidated and represented below - this is based on crawling all the references to it on the OpenWRT forums and OpenWRT Github.

- There is the Spectrum (reseller) name (SAX1V1K),  on the Askey RT5010W. https://openwrt.org/toh/dynalink/dl-wrx36 OpenWRT firmware selector hosts snapshots for this model string.  The firmware is [here](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=qualcommax%2Fipq807x&id=spectrum_sax1v1k).  

**this firmware is not booting with a WAN connection (WAN won't grab a DHCP address).  The target is **qualcommax/ipq807x.**

- There is generic name (Askey RT5010W-D187-REV6).  No hits on OpenWRT firmware selector for this, but it **is the name** in the FW filename that @MeisterLone posted [here](https://github.com/MeisterLone/Askey-RT5010W-D187-REV6/tree/master/OpenWrt-r22241).  It should also be noted that this Github repo is referenced in the OpenWRT official Github guide to defeating and flashing this modem [here](https://github.com/openwrt/openwrt/pull/14914).

**this firmware is booting with a WAN connection (WAN grabs a DHCP address from upstream router).  The target is **generic/ipq807x.** Flashing to the aforementioned snapshot above results in a no WAN.  Only issues with this firmware:
     -WAN drops out inexplicably for a minute or two every few hours (absolutely no info on this in system or kernel log)
     -Legacy firewall rules warning (even after factory reset and cp /rom/etc/config/firewall /etc/config/firewall
     
-There is the the Dynalink DL-WRX36 - also based on the Askey RT5010W.  The difference between this model and the Spectrum SAX1V1K is the addition of a USB port, and different memory type (NAND)

    No hits on OpenWRT firmware selector for this string, but it does have a wiki site here, with links to files and flashing 
    instructions [here] 
    (https://openwrt.org/toh/dynalink/dl-wrx36).  Flashing the unit to this firmware softbricks it as it appears to be waiting to 
    connect to a particular boot partition (I reckon this is due to having that USB drive).
    
So clearly there is a bit of a diaspora of firmware images here, and I've been doing my best to backtrack from each respective model discussion in the forums to try to ascertain the correct FW to flash.  Seeing as how the SAX1V1K is the actual model name, I thought tha the openWRT firmware selector would be the best place to start - sadly, no working WAN link on this.  AFAIK there shouldn't be anything to configure after flashing, it should just grab a DHCP link from upstream.

The firmware that works (the one @MeisterLone posted), I haven't been able to find an updated version of.  

Thank you for bearing with here, I'm happy to provide any and all logs - going to flash to the latest snapshot now and will reply with whatever logs are useful.

namiltd, 2024-05-15:
Paste files as an attachment or in between
```
\```
text
\```
```

namiltd, 2024-05-15:
kernel 5.15:
```
[ 2.958003] Generic PHY 90000.mdio-1:00: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:00, irq=POLL)
[ 2.961674] nss-dp 3a001000.dp1 lan4: Registered netdev lan4(qcom-id:1)
[ 2.971615] Generic PHY 90000.mdio-1:01: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:01, irq=POLL)
[ 2.977841] nss-dp 3a001200.dp2 lan3: Registered netdev lan3(qcom-id:2)
[ 2.987750] Generic PHY 90000.mdio-1:02: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:02, irq=POLL)
[ 2.993948] nss-dp 3a001400.dp3 lan2: Registered netdev lan2(qcom-id:3)
[ 3.003910] Generic PHY 90000.mdio-1:03: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:03, irq=POLL)
[ 3.010164] nss-dp 3a001600.dp4 lan1: Registered netdev lan1(qcom-id:4)
[ 3.225434] QCA808X ethernet 90000.mdio-1:1c: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:1c, irq=POLL)
[ 3.225974] nss-dp 3a007000.dp6-syn wan: Registered netdev wan(qcom-id:6)
```
kernel 6.6:
```
[ 2.703648] dp2: ppe offload disabled: 0 for macid 2
[ 2.707391] dp2: Switch attached to macid 2 status: 0
[ 2.881792] dp3: ppe offload disabled: 0 for macid 3
[ 2.881824] dp3: Switch attached to macid 3 status: 0
[ 2.961348] dp4: ppe offload disabled: 0 for macid 4
[ 2.961381] dp4: Switch attached to macid 4 status: 0
[ 3.041351] dp6-syn: ppe offload disabled: 0 for macid 6
[ 3.041384] dp6-syn: Switch attached to macid 6 status: 0
```
I will try to find out where this difference comes from

robimarko, 2024-05-15:
@namiltd Those differences dont matter, its just newer SSDK and NSS-DP.

@dangerweenie Please provide ethtool, mdio and also `cat /sys/kernel/debug/clk/clk_summary`

dangerweenie, 2024-05-15:
> @namiltd Those differences dont matter, its just newer SSDK and NSS-DP and cosmetic differences.
> 
> @dangerweenie Please provide ethtool, mdio and also `cat /sys/kernel/debug/clk/clk_summary`

Thanks - will do.

To confirm - you want this output from the _working_ or the _non-working_ FW image?

Currently on the custom working FW image, can collect any data before flashing to the latest non-functioning fw.

(I also have the rootfs extracted for both images if that is helpful)

dangerweenie, 2024-05-15:
> cat /sys/kernel/debug/clk/clk_summary

on the working custom FW from Github:

```

root@routey:~#  cat /sys/kernel/debug/clk/clk_summary
                                 enable  prepare  protect                                duty  hardware
   clock                          count    count    count        rate   accuracy phase  cycle    enable
-------------------------------------------------------------------------------------------------------
 uniphy2_gcc_tx_clk                   1        1        0   125000000          0     0  50000         Y
    nss_port6_tx_clk_src              1        1        0   125000000          0     0  50000         Y
       nss_port6_tx_div_clk_src       2        2        0   125000000          0     0  50000         Y
          gcc_uniphy2_port6_tx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_nss_port6_tx_clk        1        1        0   125000000          0     0  50000         Y
 uniphy2_gcc_rx_clk                   1        1        0   125000000          0     0  50000         Y
    nss_port6_rx_clk_src              1        1        0   125000000          0     0  50000         Y
       nss_port6_rx_div_clk_src       2        2        0   125000000          0     0  50000         Y
          gcc_uniphy2_port6_rx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_nss_port6_rx_clk        1        1        0   125000000          0     0  50000         Y
 uniphy1_gcc_tx_clk                   0        0        0   125000000          0     0  50000         Y
 uniphy1_gcc_rx_clk                   0        0        0   125000000          0     0  50000         Y
 uniphy0_gcc_tx_clk                   2        2        0   125000000          0     0  50000         Y
    nss_port4_tx_clk_src              1        1        0   125000000          0     0  50000         Y
       nss_port4_tx_div_clk_src       2        2        0   125000000          0     0  50000         Y
          gcc_uniphy0_port4_tx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_nss_port4_tx_clk        1        1        0   125000000          0     0  50000         Y
    nss_port5_tx_clk_src              1        1        0   125000000          0     0  50000         Y
       nss_port5_tx_div_clk_src       3        3        0   125000000          0     0  50000         Y
          gcc_uniphy1_port5_tx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_uniphy0_port5_tx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_nss_port5_tx_clk        1        1        0   125000000          0     0  50000         Y
 uniphy0_gcc_rx_clk                   2        2        0   125000000          0     0  50000         Y
    nss_port4_rx_clk_src              1        1        0   125000000          0     0  50000         Y
       nss_port4_rx_div_clk_src       2        2        0   125000000          0     0  50000         Y
          gcc_uniphy0_port4_rx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_nss_port4_rx_clk        1        1        0   125000000          0     0  50000         Y
    nss_port5_rx_clk_src              1        1        0   125000000          0     0  50000         Y
       nss_port5_rx_div_clk_src       3        3        0   125000000          0     0  50000         Y
          gcc_uniphy1_port5_rx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_uniphy0_port5_rx_clk       1        1        0   125000000          0     0  50000         Y
          gcc_nss_port5_rx_clk        1        1        0   125000000          0     0  50000         Y
 usb3phy_0_cc_pipe_clk                1        1        0   125000000          0     0  50000         Y
    usb0_pipe_clk_src                 1        1        0   125000000          0     0  50000         Y
       gcc_usb0_pipe_clk              1        1        0   125000000          0     0  50000         Y
 bias_pll_nss_noc_clk                 1        1        0   416500000          0     0  50000         Y
    nss_noc_bfdcd_clk_src             1        1        0   416500000          0     0  50000         Y
       nss_noc_clk_src                2        2        0   416500000          0     0  50000         Y
          gcc_ubi1_nc_axi_clk         0        0        0   416500000          0     0  50000         N
          gcc_ubi1_axi_clk            0        0        0   416500000          0     0  50000         N
          gcc_ubi0_nc_axi_clk         0        0        0   416500000          0     0  50000         N
          gcc_ubi0_axi_clk            0        0        0   416500000          0     0  50000         N
          gcc_nss_noc_clk             1        1        0   416500000          0     0  50000         Y
          gcc_mem_noc_nss_axi_clk       1        1        0   416500000          0     0  50000         Y
 bias_pll_cc_clk                      1        1        0   300000000          0     0  50000         Y
    nss_ppe_clk_src                  15       15        0   300000000          0     0  50000         Y
       gcc_crypto_ppe_clk             0        0        0   300000000          0     0  50000         N
       gcc_port6_mac_clk              1        1        0   300000000          0     0  50000         Y
[... cut ...]
I can install ethtool and mdio on this working FW shell as it has access to WAN.  However, when I boot into the latest snapshot, there will be no WAN access - do you have any advice for installing these tools into the shell with no WAN?  If you could provide some guidance I could put them on my host machine and then apt to that, but I've never done that before.

Likewise I can add them to the fimware selector pkgs and request a build - are there any other tools I should add to the embedded packages before I grab it?

dangerweenie, 2024-05-15:
> @namiltd Those differences dont matter, its just newer SSDK and NSS-DP and cosmetic differences.
> 
> @dangerweenie Please provide ethtool, mdio and also `cat /sys/kernel/debug/clk/clk_summary`

`ethtool wan` on the working firmware:

```
root@routey:~# ethtool wan
Settings for wan:
        Supported ports: [ TP AUI BNC MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 28
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Link detected: yes
```

Having issues installing mdio-tools, please advise:

`root@routey:~# opkg install mdio-tools
Unknown package 'mdio-tools'.`

dangerweenie, 2024-05-15:
Flashed to latest snapshot from OpenWRT firmware-selector, `openwrt-qualcommax-ipq807x-spectrum_sax1v1k-squashfs-sysupgrade.bin`

ethtool wan:

'''
ethtool wan
Settings for wan:
        Supported ports: [ ]
        Supported link modes:   1000baseKX/Full
                                10000baseKX4/Full
                                10000baseKR/Full
                                10000baseSR/Full
                                10000baseLR/Full
                                10000baseER/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseKX/Full
                                10000baseKX4/Full
                                10000baseKR/Full
                                10000baseSR/Full
                                10000baseLR/Full
                                10000baseER/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Unknown! (255)
        Port: Twisted Pair
        PHYAD: 28
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Link detected: no
'''

mdio 9*

'''
root@routey:/# mdio 9*
 DEV      PHY-ID  LINK
0x00  0x004dd0b1  down
0x01  0x004dd0b1  down
0x02  0x004dd0b1  down
0x03  0x004dd0b1  down
0x04  0x004dd0b1  down
0x05  0x04820a05  down
0x1c  0x004dd101  down
0x1d  0x8aa21ac8  down
'''

cat /sys/kernel/debug/clk/clk_summary

'''
root@routey:/# cat /sys/kernel/debug/clk/clk_summary
                                 enable  prepare  protect                                dut                                                                                                                                                                                                                                                                                                                                               y  hardware                            connection
   clock                          count    count    count        rate   accuracy phase  cycl                                                                                                                                                                                                                                                                                                                                               e    enable   consumer                         id
--------------------------------------------------------------------------------------------                                                                                                                                                                                                                                                                                                                                               -------------------------------------------------
 uniphy2_gcc_tx_clk                  0       0        0        312500000   0          0                                                                                                                                                                                                                                                                                                                                                    50000      Y   deviceless                      no_connection_id
 uniphy2_gcc_rx_clk                  0       0        0        312500000   0          0                                                                                                                                                                                                                                                                                                                                                    50000      Y   deviceless                      no_connection_id
 uniphy1_gcc_tx_clk                  0       0        0        125000000   0          0                                                                                                                                                                                                                                                                                                                                                    50000      Y   deviceless                      no_connection_id
 uniphy1_gcc_rx_clk                  0       0        0        125000000   0          0                                                                                                                                                                                                                                                                                                                                                    50000      Y   deviceless                      no_connection_id
 uniphy0_gcc_tx_clk                  2       2        0        125000000   0          0                                                                                                                                                                                                                                                                                                                                                    50000      Y   deviceless                    [... cut ...]
 pcie1_pipe_clk_src                  0       0        0        0           0          0     50000      Y   deviceless                      no_connection_id
    gcc_pcie1_pipe_clk               0       0        0        0           0          0     50000      N      deviceless                      no_connection_id
 pcie0_pipe_clk_src                  0       0        0        0           0          0     50000      Y   deviceless                      no_connection_id
    gcc_pcie0_pipe_clk               0       0        0        0           0          0     50000      N      deviceless                      no_connection_id
root@routey:/#
'''

robimarko, 2024-05-15:
There is an mdio-tools package, it depends on `kmod-mdio-netlink` but it should be autoselected when you add `mdio-tools`.

You can just add it via firmware selector

dangerweenie, 2024-05-15:
> Contributor

Thanks for checking it out.  Not really sure what that means but I am down to troubleshoot!

robimarko, 2024-05-15:
Hm, UNIPHY2 clocks seem wrong to me.

I gotta see if any of mine QCA8081 devices have issues

dangerweenie, 2024-05-15:
> Hm, UNIPHY2 clocks seem wrong to me.
> 
> I gotta see if any of mine QCA8081 devices have issues

understood - leaving this firmware on for the time being in case you need any more diagnostic.

curious what the last [official] stable / working snapshot for this router was (and how to find it) --- there has to be one somewhere, right? else how could it have wound up on the firmware selector page?

robimarko, 2024-05-15:
Well, bad news is that its not a generic issue for QCA8081 and I cannot reproduce it on my devices.

Well, @gotofbi is the author or device support commit so I can only guess it worked for him.

To me it seems that SSDK or NSS-DP is not probing that port at all since the UNIPHY2 clocks are not even being enabled.

dangerweenie, 2024-05-15:
> Well, bad news is that its not a generic issue for QCA8081 and I cannot reproduce it on my devices.
> 
> Well, @gotofbi is the author or device support commit so I can only guess it worked for him.
> 
> To me it seems that SSDK or NSS-DP is not probing that port at all since the UNIPHY2 clocks are not even being enabled. Is there some way that I could remotely access the device and/or flash the firmware?

For sure - I'd be happy to open up a Teamviewer instance, if you don't mind tagging in for a bit.  Appreciate your generosity!  Sending you an email now (got your address on your profile)

dangerweenie, 2024-05-20:
update:

was able to get WAN to work by pulling a LAN port from BR-LAN and moving it into a BR-WAN device:

```
config interface 'wan'
        # option device 'wan'
        option device 'br-wan'
        option proto 'dhcp'

config interface 'wan6'
        # option device 'wan'
        option device 'br-wan'
        option proto 'dhcpv6'
       

Unfortunately this is not a long-term solution as it also drops WAN speed by half.

Local VM + router is available for SSH if anyone is open to checking it out.

gotofbi, 2024-05-24:
Sorry I was on vacation and missed entire issue.
My wan port works fine so as other 3 sax1v1k that I installed for dumb-ap
Dumb-ap's are connected through LAN port so didnt have any this issue.

On my main sax1v1k, WAN works totally fine but very rarely I do have some similar issue.
When I reboot device, very rarely, I do lose WAN port.
I had to power cycle to fix this problem but maybe its related to that.

@robimarko, @dangerweenie  Thank you so much for digging into the issue.
I will try if I can reproduce issue  and also try latest snapshot and see if the issue gets fixed.

PR #15541 (2024-05-21): qualcommax: ipq807x: sax1v1k: fix QCA8081 reset
It seems that on Spectrum SAX1V1K QCA8081 is being brought out of reset too quickly and thus causing it to get stuck in an invalid autoneg register configuration mode.

Setting an deasset delay seems to fix this, so lets set it.

Fixes: #15493