openwrt/openwrt #15496: MT7981/mediatek/filogic ethernet ports ping lost after reboot -f to force reboot

ptpt52, 2024-05-15:
### Describe the bug

tested on mt7981
this issue comes with kernel 6.6, I am sure that no such issue on kernel 6.1

how to reproduce:
1. force reboot via `reboot -f`
2. after reboot lan/wan ports all link up but ping lost, ifconfig shows rx=0, tx increasing.

ping @dangowrt @robimarko @nbd168 

### OpenWrt version

r26493-8a54c79a33e

### OpenWrt release

xxx

### OpenWrt target/subtarget

mediatek/filogic

### Device

xxx

### Image kind

Self-built image

### Steps to reproduce

1. force reboot via `reboot -f`
2. after reboot lan/wan ports all link up but ping lost, ifconfig shows rx=0, tx increasing.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2024-05-15:
after `reboot -f`, the issue happens. all ports lost connection (link up but no rx)
A hardware power outage and restart (unplug the power supply for a few seconds and then plug it in) can solve this problem and return to normal.

namiltd, 2024-05-15:
On what exact hardware/router did you observe this behavior?
Doesn't the MAC change after every reboot?

ptpt52, 2024-05-15:
@namiltd  CT3003
no mac changes

mrkiko, 2024-05-15:
So it happens even when you
echo "b"
to sysrq-trigger
?

ptpt52, 2024-05-15:
> So it happens even when you echo "b" to sysrq-trigger ?

sure for that. `echo b >/proc/sysrq-trigger` trigger a reboot also cause the issue.

ptpt52, 2024-05-15:
Finally, I found out that this issue was caused by a patch I added to the kernel's mtk_eth_soc driver when I adapted natflow. I don't know the real cause yet, but removing this patch can solve the problem. I'll do some research myself. Sorry for causing trouble to everyone

closed 2024-05-15 by ptpt52