openwrt/openwrt #15503: How to Build system setup for macOS (M1/M2)

atyouspace, 2024-05-15:
I would like to build system setup in a MacBook with arm64 kernel. I appreciate for your suggestion on these two issues:
1. What necessary packages can be installed on arm64?
2. When compiling by `gmake,` many issues occurred and parameters could not be found.

robimarko, 2024-05-15:
Its all documented here:
https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx

And I have tried it couple of days on M1 Mini sucesfully

atyouspace, 2024-05-16:
> Its all documented here: https://openwrt.org/docs/guide-developer/toolchain/buildroot.exigence.macosx
> 
> And I have tried it couple of days on M1 Mini sucesfully

Hi Robi, I was confused by "F. Create location indicator & build environment file" and met some issues. Could you show me more help?
I use macports-pkg-mngr in "G. Install necessary packages:" like this:
`● if you have MacPorts pkg-mngr then run below 2 commands:

source ~/MacPorts.sh
sudo port install libiconv gettext-runtime coreutils findutils gwhich gawk zlib pcre bzip2 ncurses grep getopt gettext-tools-libs gettext diffutils sharutils util-linux libxslt libxml2 help2man readline gtime gnutar unzip zip lzma xz libelf fastjar libusb libftdi0 expat sqlite3 openssl3 openssl kerberos5 dbus lz4 libunistring nettle icu gnutls p11-kit wget quilt subversion gmake pkgconfig libzip cdrtools ccache curl xxhashlib rsync libidn perl5 p5.28-xml-parser p5.30-xml-parser p5-extutils-makemaker p5-data-dumper boost-jam boost boost-build bash bash-completion binutils m4 flex intltool patchutils swig git-extras git openjdk17 openjdk7-zulu luajit libtool glib2 file python27 python310 libzzip mercurial asciidoc sdcc gnu-classpath`

However, some packages were missing, such as getopt and gnu-classpath (which may be not supported by macports for arm64). And how to Load build environment which confused me more, /Volumes/OpenWrt/env.sh and ~/MacPorts.sh are same? How to compile the project source, make?gmake?

I appreciate your help. @robimarko

robimarko, 2024-05-16:
Yeah, I just used homebrew and that worked after making `env.sh` that points to the correct tools.

I basically just use macOS for checking whether it still builds so my knowledge about it is minimal

atyouspace, 2024-05-16:
> Yeah, I just used homebrew and that worked after making `env.sh` that points to the correct tools.
> 
> I basically just use macOS for checking whether it still builds so my knowledge about it is minimal

How to compile? Make or gmake?

robimarko, 2024-05-16:
gmake, that is what the env.sh script does, aliases them

atyouspace, 2024-05-16:
> gmake, that is what the env.sh script does, aliases them

Now, I can compile source by make, but some more issues happen, like this
```
lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression of type 'void (void)' [-Wincompatible-function-pointer-types]
__attribute_noreturn__ void (*obstack_alloc_failed_handler) (void)
                              ^
1 error generated.
make[6]: *** [Makefile:2781: lib/obstack.o] Error 1
make[6]: Leaving directory '/Volumes/OpenWrt/source/build_dir/host/sed-4.8'
make[5]: *** [Makefile:3105: all-recursive] Error 1
make[5]: Leaving directory '/Volumes/OpenWrt/source/build_dir/host/sed-4.8'
make[4]: *** [Makefile:2321: all] Error 2
make[4]: Leaving directory '/Volumes/OpenWrt/source/build_dir/host/sed-4.8'
make[3]: *** [Makefile:49: /Volumes/OpenWrt/source/build_dir/host/sed-4.8/.built] Error 2
make[3]: Leaving directory '/Volumes/OpenWrt/source/tools/sed'
time: tools/sed/compile#16.14#9.27#49.65
    ERROR: tools/sed failed to build.
make[2]: *** [tools/Makefile:160: tools/sed/compile] Error 1
make[2]: Leaving directory '/Volumes/OpenWrt/source'
make[1]: *** [tools/Makefile:156: /Volumes/OpenWrt/source/staging_dir/host/stamp/.tools_compile_yyyyyynnyyyynyyyyyynyynnyyyynyyyyyyyyyyyyyyyyyynynnyyyyyyy] Error 2
make[1]: Leaving directory '/Volumes/OpenWrt/source'
make: *** [/Volumes/OpenWrt/source/include/toplevel.mk:230: world] Error 2
```

robimarko, 2024-05-16:
Master worked for me, I have not tested 23.05 or older

atyouspace, 2024-05-17:
> Master worked for me, I have not tested 23.05 or older
did you meet the problem about gdb error?

robimarko, 2024-05-17:
Not on master

BKPepe, 2024-05-22:
> 
> lib/obstack.c:351:31: error: incompatible function pointer types initializing 'void (*)(void) __attribute__((noreturn))' with an expression of type 'void (void)' [-Wincompatible-function-pointer-types]
> __attribute_noreturn__ void (*obstack_alloc_failed_handler) (void)

This looks like you are trying to compile OpenWrt 22.03, because the sed is 4.8 is there.  In any case, this was fixed there by https://github.com/openwrt/openwrt/commit/4a6911fe7911914eb1319d55dd0de02542a16247

closed 2024-05-22 by ynezz