openwrt/openwrt #15505: BT Home Hub 5A  once a day LAN ports are completely off

monsoft, 2024-05-15:
### Describe the bug

We weeks ago I switched from version 22.03.6 to 23.05.3 and all was working fine until this Monday, when all my devices connected to LAN ports goes completely off. Yesterday I had this same situation, and today as well. Devices connected to wifi are working, but connected to LAN not. 

There is completely no errors in syslog or kernel log files what can point me to anything. I was checking if this could be bridge issue related to changing ports state from forwarding to blocking, but no, all used LAN ports are in forwarding state.

I can't do much except to reboot router, and this are resolving issue.
I'm thinking to do router configuration wipe, and configure it again (a lot of things to configure) or if this will not help, I will try to back to previous system 22.03.6.

Is there anything what could be done, to try to figure out what causing this issue ?

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

lantiq/xrx200

### Device

BT Home Hub 5A

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-15:
It may be, for example, a hardware problem such as dried capacitors.

monsoft, 2024-05-15:
@namiltd : We are talking about max 4yo device which were working fine before upgrading to newest system. For me possible but unusual. Additionally, I'm doing software reboot not power cycle to fix this issue so I think (but I still can be wrong) that dried capacitors will be still in it's broken state until I will switch device off/on.

brada4, 2024-05-21:
> max 4yo device

Hh5 was manufactured 2013-2016, min 7 years to be accurate.

monsoft, 2024-06-01:
Maybe, but I'm not suspecting hardware issue related to dry capacitors only because I upgraded firmware.
I reset router to factory setup and restored configuration from backup. There was no issue for 14 days and issue with LAN ports come back. I noticed that when this happened, additionally on wifi there is no DHCP and DNS working. Maybe this is related to bridge issue on BT Home HUB 5A devices in this new firmware ?

mjaggard, 2025-01-15:
I think I'm getting the same issue with my BT Home Hub 5A devices (at least 2 of them). Although I didn't get it when I updated to 23.05 for quite a while - I only started seeing the issue when I changed my broadband provider and the router that I'm connecting my OpenWRT devices. Aside from that it seems identical.

I've ordered some network hubs to replace the role that these routers provide in joining my wired network together but I'm expecting to still have issues with Wifi mesh provided by the OpenWRT devices.

mjaggard, 2025-01-15:
I've enabled logging to a file so that I can view the logs after the device is rebooted and I'll report back here if I find anything.