openwrt/openwrt #15507: mac address assignment using nvmem-layout fails for eth0 on mt7981b devices

yogo1212, 2024-05-15:
### Describe the bug

During the porting work for the [cudy,m3000](https://github.com/openwrt/openwrt/pull/14393), we noticed that the mac address assignment for mac@0 doesn't work through `nvmem-layout`.
I noticed this happened for the yuncore,ax835 during a [cleanup commit](cfee7fa36682be3476b5b27e8affad99c8753934).

Example:

```
&eth {
  pinctrl-names = "default";
  pinctrl-0 = <&mdio_pins>;
    
  status = "okay";
    
  gmac0: mac@0 {
    compatible = "mediatek,eth-mac";
    reg = <0>;
    phy-mode = "2500base-x";
    phy-handle = <&rtl8221b_phy>;

    nvmem-cell-names = "mac-address";
    nvmem-cells = <&macaddr_bdinfo_de00 1>; /* <- doesn't work (even with a 0) */
  };

  gmac1: mac@1 {
    compatible = "mediatek,eth-mac";
    reg = <1>;
    phy-mode = "gmii";
    phy-handle = <&int_gbe_phy>;

    nvmem-cell-names = "mac-address";
    nvmem-cells = <&macaddr_bdinfo_de00 0>;
  };
};

&bdinfo {
          nvmem-layout {
          compatible = "fixed-layout";
          #address-cells = <1>;
          #size-cells = <1>;

          macaddr_bdinfo_de00: macaddr@de00 {
            #nvmem-cell-cells = <1>;
            compatible = "mac-base";
            reg = <0xde00 0x6>;
          };
        };
}l
```

`of_get_mac_address` is called twice for `mac@0`.
the first time, it finds a `local-mac-address` (that doesn't come from bdinfo but is consistent through reboots)
the second time, it finds that same address in `mac-address `.

```
[    1.066786] of_get_mac_address for mac@0
[    1.070710] of_get_mac_address mac-address -19
[    1.075140] of_get_mac_address local-mac-address 0
[    1.079942] of_add_mac_address ret 0: d6:37:bf:39:da:6a

[    1.084809] of_get_mac_address for mac@1
[    1.088725] of_get_mac_address mac-address -19
[    1.093155] of_get_mac_address local-mac-address -19
[    1.098109] of_get_mac_address address -19
[    1.102723] of_get_mac_address nvmem 0: 80:af:ca:15:84:04
[    1.108134] of_add_mac_address ret 0: 80:af:ca:15:84:04

[    1.113001] of_get_mac_address for mac@0
[    1.116920] of_get_mac_address mac-address 0
[    1.121186] of_add_mac_address ret 0: d6:37:bf:39:da:6a
```

The most confusing thing is, there is a `mac-address` in sysfs:
```
$ hexdump -C /sys/firmware/devicetree/base/soc/ethernet@15100000/ma
c@0/mac-address 
00000000  d6 37 bf 39 da 6a                                 |.7.9.j|
```

How does that get there?!

### OpenWrt version

doesn't matter

### OpenWrt release

doesn't matter

### OpenWrt target/subtarget

mediatek/filogic

### Device

doesn't matter

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yogo1212, 2024-05-15:
Was that put there by uboot?!

yogo1212, 2024-05-15:
got it. making a pr

yogo1212, 2024-05-15:
or not.
setting `local-mac-address = [ 00 00 00 00 00 00 ];` or `[ ff ff ff ff ff ff ]` doesn't seem to overwrite the bogus mac.

yogo1212, 2024-05-15:
the best i can think of atm is adding a `ignore-local-mac-address` node.
given the high number of `[ 00 00 00 00 00 00 ]` and `[ 000000000000 ]` nodes in the kernel, other people might also want that.

john-tho, 2024-05-16:
Yes, `mac-address` and `local-mac-address` can be set by u-boot (see [`fdt_fixup_ethernet`](https://elixir.bootlin.com/u-boot/latest/C/ident/fdt_fixup_ethernet)), and they take priority:
https://elixir.bootlin.com/linux/v6.6.30/source/net/core/of_net.c#L126

OpenWrt via [`netifd`](https://lxr.openwrt.org/ident?i=config_get_default_macaddr) will also change these addresses from userspace (if asked to), so make sure there are no `macaddr` settings in `/etc/config/network`
https://github.com/openwrt/openwrt/blob/20e4a18feb3f766b0f6ebec1afc154b345398a7a/target/linux/mediatek/filogic/base-files/etc/board.d/02_network#L139-L140
https://github.com/openwrt/openwrt/blob/20e4a18feb3f766b0f6ebec1afc154b345398a7a/target/linux/mediatek/filogic/base-files/etc/board.d/02_network#L187-L189

musashino205, 2024-05-16:
may be:

```
aliases {
    ethernetN = &gmacN;  <-- referred to find a ethernet adapter for applying 
};

...

gmacN: gmacN {
    local-mac-address = [xx xx xx xx xx xx];   <-- replaced (or added if not exists) with "ethNaddr" in environment variables of U-Boot
};
```

yogo1212, 2024-05-16:
@john-tho thanks for the info!

very unfortunate that it takes priority because:
`Warning: ethernet@15100000 (eth0) using random MAC address - 2a:d6:b9:ef:a3:00` (from `eth_post_probe`)
it looks like the vendor u-boot is pretty much the same as the one in the yuncore ax835 and it just doesn't have any address assignment info.

@musashino205 i'll try setting `eth0addr` to see what it does :+1: 
right now, i don't know whether there can be maintainable solution for openwrt.

yogo1212, 2024-05-16:
@john-tho do you have any idea why the override (`local-mac-addr: [ 00 00 00 00 00 00 ]`) doesn't work?

yogo1212, 2024-05-16:
@musashino205 this uboot has `ethaddr` set to the random address. i've tried persisting both `ethaddr` and `eth0addr` but it will still generate a random mac address and set the environment variable anew (overwriting my value if it's `ethaddr`).

yogo1212, 2024-05-16:
i'm pretty confident it's not `fdt_fixup_ethernet` that is setting the `local-mac-address`.
`fdtdec_set_ethernet_mac_address` wouldn't create the node, so something else would have to being that (up to the vendor).
the message comes from `eth_post_probe` and that uses `eth_write_hwaddr`. again, no idea if the vendor uboot has a driver that implements `write_hwaddr`.

regardless, the stock uboot doesn't do the detection right and needs replacing if this is to be avoided.
even if the env trick had worked, we wouldn't have been able to set it from Openwrt because uboot uses the default env (the `u-boot-env` mtd is empty). writing to it from openwrt would write the a different default environment.
not that i think it matters: uboot loads env vars at runtime depending on whether or not the button is pressed.

i think this is a matter of linux ignoring that mac address.

yogo1212, 2024-05-16:
i'll try to find out how u-boot can overwrite the device tree that openwrt brings.
if that doesn't work, i'll make a patch for `ignore-local-mac-address`.

yogo1212, 2024-05-16:
i didn't find any code that could overwrite `local-mac-address` after the flattened device tree has been loaded.
maybe there isn't anything like that. maybe uboot loads the fdt from the fit image and overwrites the mac address in there (rather than passing its own device tree to linux which then loads the fit dt later).

if someone could point me in the right direction, i would be very happy :-)

john-tho, 2024-05-16:
Okay, so we think (correct me if wrong):
- Vendor u-boot:
  - does not (consistently) load u-boot env (so setting `ethaddr` does not help)
  - generates and stores to env random MAC via [`net_random_ethaddr`](https://elixir.bootlin.com/u-boot/latest/C/ident/net_random_ethaddr)
    > `Warning: ethernet@15100000 (eth0) using random MAC address - 2a:d6:b9:ef:a3:00`
  - loads FIT FDT to memory, modifies it (stores this MAC address to `local-mac-address` via [`fdt_fixup_ethernet`](https://elixir.bootlin.com/u-boot/latest/C/ident/fdt_fixup_ethernet)), and passes (memory address of DTB) to Linux.
- Linux
  - [OpenWrt patch pending 683](https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.6/683-of_net-add-mac-address-to-of-tree.patch) copies `local-mac-address` to `mac-address`
  - [prefers `mac-address` over `local-mac-address`, `address`, then nvmem](https://elixir.bootlin.com/linux/v6.6.30/source/net/core/of_net.c#L126)

Not looked into _why_ Linux `of_get_mac_address` gets called more than once for the interface…

If you really want to avoid OpenWrt userspace setting MAC address, and cannot get u-boot to work with env, then yes, patch `of_get_mac_address` to ignore DTS addresses (possibly triggered via a DTS property).

yogo1212, 2024-05-17:
@john-tho 

About the u-boot env: One part is not being able to confirm it is loaded at boot. The only way to get at the prompt is by holding the reset button and the environment is strange (no bootcmd and such). Since u-boot u-boot-env mtd is empty by default, the other problem is that Openwrt wouldn't just write that one MAC address but its own default env which is unlikely to work (if it is even loaded).

Apart from it being much cleaner to be able to use the device tree, the principal means of changing the MAC address use UCI to change the corresponding interface section.
The (Linux) Ethernet interface still has that random MAC address, as any other UCI interfaces using eth0 also will.

PR #17201 (2024-12-09): mediatek: filogic: prevent faulty mac address assignment
This extends this fix #15507 also for Cudy WR3000

see also:
#15587
#15515
closed 2024-05-25 by openwrt-bot