openwrt/openwrt #15510: [23.05] mt76 fails to compile after May 16 commit (fix included)

Shine-, 2024-05-17:
### Describe the bug

@rmilecki 

After 2568b30285ff8715b45434f050ba4598dba1f34b (MT7603 fixes), mt76 fails to compile with the following error message:
```
[...]/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91/mt7603/dma.c:60:17: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-after-statement]
   60 |                 u8 qid = tid_to_ac[tid];
      |                 ^~
cc1: all warnings being treated as errors
```
As I have been backporting these MT7603 fixes into my personal 23.05 builds for quite a while already, I was aware of the error.

Here's a quick and dirty fix:
```
diff --git a/package/kernel/mt76/patches/0007-wifi-mt76-mt7603-fix-tx-queue-of-loopback-packets.patch b/package/kernel/mt76/patches/0007-wifi-mt76-mt7603-fix-tx-queue-of-loopback-packets.patch
index 56a2aea6f6..739da462cc 100644
--- a/package/kernel/mt76/patches/0007-wifi-mt76-mt7603-fix-tx-queue-of-loopback-packets.patch
+++ b/package/kernel/mt76/patches/0007-wifi-mt76-mt7603-fix-tx-queue-of-loopback-packets.patch
@@ -54,12 +54,12 @@ Signed-off-by: Felix Fietkau <nbd@nbd.name>
 -      if (ieee80211_is_data_qos(hdr->frame_control))
 +
 +      hwq = wmm_queue_map[IEEE80211_AC_BE];
-+      if (ieee80211_is_data_qos(hdr->frame_control)) {
++      if (ieee80211_is_data_qos(hdr->frame_control)) { u8 qid;
                tid = *ieee80211_get_qos_ctl(hdr) &
 -                    IEEE80211_QOS_CTL_TAG1D_MASK;
 -      skb_set_queue_mapping(skb, tid_to_ac[tid]);
 +                       IEEE80211_QOS_CTL_TAG1D_MASK;
-+              u8 qid = tid_to_ac[tid];
++              qid = tid_to_ac[tid];
 +              hwq = wmm_queue_map[qid];
 +              skb_set_queue_mapping(skb, qid);
 +      } else if (ieee80211_is_data(hdr->frame_control)) {
```

### OpenWrt version

commit 2568b30285ff8715b45434f050ba4598dba1f34b

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

any that compiles the MT76 driver

### Device

any that compiles the MT76 driver

### Image kind

Self-built image

### Steps to reproduce

Compile 23.05-SNAPSHOT for a mt76 device.
Tested on Alpine 3.19.1 and Debian Bookworm.

### Actual behaviour

Build fails at the MT76 driver.

### Expected behaviour

Build completes.

### Additional info

n/a

### Diffconfig

```text
n/a
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rmilecki, 2024-05-17:
Uh, I tested that commit in my local build without `CONFIG_PACKAGE_kmod-mt7603`. Sorry for that problem :/

Shine-, 2024-05-17:
No worries, I already had that fixed in my builds anyways, ever since I started backporting these commits. Now that it landed in STABLE, I wanted to raise this issue as a heads-up :-)

Borromini, 2024-05-17:
Just bumped into this issue as well. Can confirmed the modified patch makes mt76 compile again.

```
/var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91/mt7603/dma.c: In function 'mt7603_rx_loopback_skb':
/var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91/mt7603/dma.c:60:17: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-after-statement]
   60 |                 u8 qid = tid_to_ac[tid];
      |                 ^~
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:289: /var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91/mt7603/dma.o] Error 1
make[4]: *** [scripts/Makefile.build:552: /var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91/mt7603] Error 2
make[3]: *** [Makefile:1908: /var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91] Error 2
make[3]: Leaving directory '/var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.15.158'
make[2]: *** [Makefile:638: /var/media/data/code/lede/23.05/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2024-02-03-6124ea91/.built] Error 2
make[2]: Leaving directory '/var/media/data/code/lede/23.05/package/kernel/mt76'
time: package/kernel/mt76/compile#5.59#0.75#6.20
    ERROR: package/kernel/mt76 failed to build.
make[1]: *** [package/Makefile:129: package/kernel/mt76/compile] Error 1
```

rmilecki, 2024-05-18:
```
[2024-05-17] [09:41:31 CEST] <rmilecki> nbd: there seems to be a problem with mt7603 compilation
[2024-05-17] [09:41:37 CEST] <rmilecki> caused by https://git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/commit/?id=b473c0e47f04d3b4ee9d05d2e79234134aad14d5 ("wifi: mt76: mt7603: fix tx queue of loopback packets")
[2024-05-17] [09:41:57 CEST] <rmilecki> mt7603/dma.c: In function 'mt7603_rx_loopback_skb':
[2024-05-17] [09:41:57 CEST] <rmilecki> mt7603/dma.c:60:17: error: ISO C90 forbids mixed declarations and code [-Werror=declaration-after-statement]
[2024-05-17] [09:41:57 CEST] <rmilecki> 60 | u8 qid = tid_to_ac[tid];
[2024-05-17] [09:41:57 CEST] <rmilecki> | ^~
[2024-05-17] [09:46:30 CEST] <nbd> rmilecki: will look into it later
```

piaddic7ed, 2024-05-18:
I ran into issues building images using the imagebuilder as well. Current 23.05-SNAPSHOT images are borked due to missing kmods. For whatever reason buildbot didn't complain about compilation errors. 
`Collected errors:
 * opkg_install_cmd: Cannot install package kmod-mt7915e.
 * opkg_install_cmd: Cannot install package kmod-mt7981-firmware.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:144: _call_image] Error 2
make: *** [Makefile:262: image] Error 2
`

Shine-, 2024-05-18:
It has been fixed in openwrt/mt76@513c131c6309712a51502870b041f45b4bd6a6d4.

closed 2024-05-18 by rmilecki