openwrt/openwrt #15516: failed to make source

WenisOK, 2024-05-17:
### Describe the bug

tmp/.config-package.in:114391: syntax error
tmp/.config-package.in:114390: unknown statement "---help---"
tmp/.config-package.in:114391: unknown statement "Ntfs-3g"
tmp/.config-package.in:114392:warning: ignoring unsupported character ','
tmp/.config-package.in:114392: unknown statement "If"
tmp/.config-package.in:114393: unknown statement "etc."
tmp/.config-package.in:114394: unknown statement "of"

### OpenWrt version

23.05

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

.

### Device

x86

### Image kind

Official downloaded image

### Steps to reproduce

./scripts/feeds update -a
./scripts/feeds install -a

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-17:
Please correct version and device tags from your openwrt device.
What behaviour you expect?

WenisOK, 2024-05-17:
i clone this repository. i want to build the openwrt's packages. if i exec any make commands, the console output this error messages. i execed the `./scripts/feeds update -a` and `./scripts/feeds install -a` too.

WenisOK, 2024-05-17:
my system is ubuntu 24.04. i fllowed the offical document. exec the commadns:
```shelll
sudo apt update
sudo apt install build-essential clang flex bison g++ gawk \
gcc-multilib g++-multilib gettext git libncurses5-dev libssl-dev \
python3-setuptools rsync swig unzip zlib1g-dev file wget
```

brada4, 2024-05-17:
First buiild toolset:
https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem
Then enable package you need via menuconfig, and
https://openwrt.org/docs/guide-developer/toolchain/single.package

WenisOK, 2024-05-17:
> First buiild toolset: https://openwrt.org/docs/guide-developer/toolchain/use-buildsystem Then enable package you need via menuconfig, and https://openwrt.org/docs/guide-developer/toolchain/single.package

i exec the commad`make menuconfig`, the console print the error. I made sure I followed the official documentation
```shell
tmp/.config-package.in:114391: syntax error
tmp/.config-package.in:114390: unknown statement "---help---"
tmp/.config-package.in:114391: unknown statement "Ntfs-3g"
tmp/.config-package.in:114392:warning: ignoring unsupported character ','
tmp/.config-package.in:114392: unknown statement "If"
tmp/.config-package.in:114393: unknown statement "etc."
tmp/.config-package.in:114394: unknown statement "of"
tmp/.config-package.in:120124:warning: ignoring type redefinition of 'PACKAGE_jshn' from 'bool' to 'tristate'
tmp/.config-package.in:120401:warning: ignoring type redefinition of 'PACKAGE_libjson-script' from 'bool' to 'tristate'
make: *** [/home/wenisok/openwrt-sdk-23.05.3-x86-64_gcc-12.3.0_musl.Linux-x86_64/include/toplevel.mk:136: menuconfig] Error 1
```
i make sure i don't change the any files

brada4, 2024-05-17:
You do not need to check out this repository if you use SDK
https://openwrt.org/docs/guide-developer/toolchain/using_the_sdk

closed 2024-05-17 by WenisOK