openwrt/openwrt #15528: bmips: SNAPSHOT build - attendedsysupgrade or auc to latest version not possible

GoetzGoerisch, 2024-05-19:
### Describe the bug

Running the below SNAPSHOT version not able to upgrade to the latest version.

auc error:
``` text
Error: Received inncorrect version r26355-52a225cba4 (requested r26353-a96354bcfb)
Bad message (77)
```

attendedsysupgrade error:

``` text
Package list missing or not up-to-date, generating it.

Building package index...
make[3]: *** [Makefile:168: package_index] Error 1
make[2]: *** [Makefile:181: package_reload] Error 2
make[1]: *** [Makefile:160: _call_manifest] Error 2
make: *** [Makefile:314: manifest] Error 2
```

### OpenWrt version

r25978-ea609fe486

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bmips/bcm6368

### Device

Netgear DGND3800B

### Image kind

Official downloaded image

### Steps to reproduce

run auc or attendedsysupgrade

### Actual behaviour

error received, see above

### Expected behaviour

image is build and applied

### Additional info

Request Data:

``` json
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r25978-ea609fe486",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "9a1c5a1cb161ebb434e9f4c63e09780c",
    "sha256_unsigned": "",
    "client": "luci/24.079.58964~7943616",
    "packages": {
        "luci-theme-material": "24.085.24999~41c1821",
        "iwinfo": "2024.03.23~79a96150-r1",
        "libc": "1.2.5-r4",
        "wpad-basic-mbedtls": "2024.03.09~695277a5-r1",
        "busybox": "1.36.1-r1",
        "ip-full": "6.7.0-r1",
        "nano": "7.2-r4",
        "kmod-veth": "6.1.86-r1",
        "procd-ujail": "2024.03.30~946552a7-r1",
        "base-files": "1589~ea609fe486",
        "libustream-mbedtls": "2024.04.19~524a76e5-r1",
        "firewall4": "2023.11.03~698a5335-r1",
        "kmod-usb-ohci": "6.1.86-r1",
        "procd": "2024.03.30~946552a7-r1",
        "kmod-usb2": "6.1.86-r1",
        "odhcp6c": "2023.05.12~bcd28363-r20",
        "ethtool": "6.6-r1",
        "uci": "2023.08.10~5781664d-r1",
        "luci-ssl": "23.035.26083~7550ad6",
        "dropbear": "2022.83-r1",
        "luci-theme-openwrt-2020": "24.079.45461~be212dd",
        "mtd": "26",
        "odhcpd-ipv6only": "2023.10.24~d8118f6e-r1",
        "procd-seccomp": "2024.03.30~946552a7-r1",
        "urandom-seed": "3",
        "kmod-leds-gpio": "6.1.86-r1",
        "kmod-gpio-button-hotplug": "6.1.86-r3",
        "logd": "2024.01.24~2c5887cb-r2",
        "luci-app-attendedsysupgrade": "24.079.58964~7943616",
        "auc": "0.3.2-r1",
        "ca-bundle": "20240203-r1",
        "kmod-b43": "6.1.86.6.6.15-r1",
        "luci": "23.051.66410~a505bb1",
        "kmod-usb-ledtrig-usbport": "6.1.86-r1",
        "kernel": "6.1.86~a50ea60044632de1cb4b4eaffe47722d-r1",
        "acme-acmesh-dnsapi": "3.0.7-r2",
        "jool-tools-netfilter": "4.1.11-r1",
        "luci-app-acme": "23.267.29406~2edb201",
        "urngd": "2023.11.01~44365eb1-r1",
        "dnsmasq-full": "2.90-r2"
    },
    "profile": "netgear,dgnd3800b",
    "target": "bmips/bcm6368",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

STDERR
``` text
Package list missing or not up-to-date, generating it.

Building package index...
make[3]: *** [Makefile:168: package_index] Error 1
make[2]: *** [Makefile:181: package_reload] Error 2
make[1]: *** [Makefile:160: _call_manifest] Error 2
make: *** [Makefile:314: manifest] Error 2
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

KanjiMonster, 2024-05-19:
Should be fixed by https://github.com/openwrt/openwrt/commit/5d37d8dc824f4fdef690b80ceae1bde6a6f16913 after the next buildbot cycle.

closed 2024-05-19 by KanjiMonster

GoetzGoerisch, 2024-05-21:
Confirmed for luci-attended-sysupgrade ✅ 

`auc` is still broken

```
Error: Received inncorrect version r26366-2eef4f7354 (requested r26363-2b8e875a85)
Bad message (77)

```