openwrt/openwrt #15529: Wifi not authenticating via VLAN on OpenWRT v23.05.3 on Netgear R8000

pixelocdguy, 2024-05-19:
### Describe the bug

Wifi will not authenticate via a VLAN using any type of encryption. An Open network with no password is fine.

Alternatively, removing the VLAN and assigning the WiFi directly to the default lan switch ports bridge using encryption works fine.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Netgear R8000 (BCM4709)

### Image kind

Official downloaded image

### Steps to reproduce

Starting with a default config:

Configure the default Bridge device: br-lan to have 1 or more VLAN IDs
Edit the default Zone LAN to one of the above VLANs, e.g. Software VLAN br-lan.1
Save

At this point everything is OK via wired/ethernet connection including downstream devices via both wired or WIFi connections using dumb AP config with VLAN added where appropriate.

Add a new Wireless config, mode Access point, Network LAN interface from above, wireless security WPA-PSK2 (same issues with others except No Encryption/Open Network), key - 8+ chars
Save

This won't auth, despite password being correct on client devices.

### Actual behaviour

Client devices will not authenticate via WiFi, even thought typed in password is correct.

### Expected behaviour

Client devices authenticate and can browse the local network and/or Internet if WAN is configured.

### Additional info

OpenWRT forums post:

[https://forum.openwrt.org/t/wifi-not-authenticating-via-vlan-after-update-to-openwrt-v23-05-3-on-netgear-r8000/195067/5]

This post also contains the Network and Wireless config files.

Found 3 other posts with the same issue, 2 were unsolved and a 3rd resolved it by setting encryption from WPA-PSK3 to WPA-PSK2.  Was unable to replicate this - all encryption had the same issue.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-19:
The long thread ends with your config fixed, then month later you say its broke again? The issue is clearly in random stuff you ñaste in your config files

pixelocdguy, 2024-05-20:
Hi, it was never fixed, unfortunately. I thought I replied to the poster at the time but either didn't click submit or it timed out.

I haven't had a chance to look at it again until now hence then month long lag in time.

I am confident this can be replicated by anyone else with this router. I have also reset the router multiple times and replicated the issue with a minimal setup. It isn't just me having this problem.

Here is another user with the same issue:

[https://forum.openwrt.org/t/r8000-vlan-wifi-encryption/170496]

If you look at their last post, they have switched to WPA2 but still have the same issue when adding a VLAN.

And possibly this:

[https://forum.openwrt.org/t/wireless-works-but-only-without-password/160649/7]
 
They don't say they are using a VLAN but never pasted their network config so I suspect they have configured something along the same lines as the above user.

brada4, 2024-05-20:
Burden of proof lies with you.

pixelocdguy, 2024-05-21:
NO problem, what else should I provide?

I am thinking a hardware reset followed by a vanilla config, add the wifi to the default bridge to show it works. Add a VLAN - wifi will stop authenticating at this point.

VLAN will work fine via wired/ethernet connections. Downstream devices work with VLANs and Wifi (different modem routers setup as dumb APs).

brada4, 2024-05-21:
> add the wifi to the default bridge

donnt do that, wifi network is bridged by hostap/supplicant from wireless config

pixelocdguy, 2024-05-22:
Sorry, bad wording. That's what I meant. I should get time to run through this in the next day or so

brada4, 2024-05-25:
Wifi does not authenticate vlans, and you refer to well solved issue. If you introduced new issue head to forum,  Your repeater effectively blocks dhcp in br-lan, no bug at al, works as expected.

pixelocdguy, 2024-05-25:
OK, I've replicated the issue.

**Steps:**

- Reset router settings using paper clip
- (Changed IP range of lan interface from 192.168.1.1 to 10.0.0.1 - so it doesn't conflict with WAN, not required if using some other WAN config)
- Enable WiFi, change Encrption to WPA2-PSK
- Set key
- Save & Apply

Everything is working, Internet works via wired or WiFi connection. Config files below minus unneeded info.

----------------------------------

root@OpenWrt:~# cat /etc/config/network

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

-----------------------------------------------

root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '18000000.axi/bcma0:7/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '149'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key '********'

-------------------------------------------------------------------

**Next, configure VLAN:**

- Devices > br-lan > Enable VLAN filtering
- Add VLAN ID 1
- All untagged & default for testing
- Back to Interfaces
- Edit lan
- Device Software VLAN br-lan.1
- Save > Save & Apply

Internet Traffic via LAN switch points is fine. Wifi disconnects on client devices stops authing at this point.

---------------------------

root@OpenWrt:~# cat /etc/config/network

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan.1'
        option proto 'static'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'

-------------------------------------

root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '18000000.axi/bcma0:7/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '149'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key '********'

brada4, 2024-05-25:
Yes, it is one untagged network and 4095 virtual networks, you are connecting wifi to otherwise unconnected br-lan.
Next ch149 will wait for any transmission in channel before transmitting, and if your client is in EU regdom then at 13dbm permitted you need to be like in 1-2m from AP to connect.

pixelocdguy, 2024-05-27:
I don't think it is the config. The above was just for demonstration purposes. This is close to my usual config:

root@OpenWrt:~# cat /etc/config/network

```
config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan.1'
        option proto 'static'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:t*'

config bridge-vlan
        option device 'br-lan'
        option vlan '2'
        list ports 'lan4:t'
```

---

I also changed the WiFi as suggested, but again, it works perfectly with no VLAN and well over 2m.

root@OpenWrt:~# cat /etc/config/wireless

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path '18000000.axi/bcma0:7/pci0000:00/0000:00:00.0/0000:01:00.0'
        option channel '157'
        option band '5g'
        option htmode 'VHT80'
        option cell_density '0'
        option country 'AU'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key '********'

```
---

Also, I don't have it connected right now, but other  downstream connected to port 4 worked the correct VLAN and via WiFi.

---

If this is not a correct config, then please advise what should be changed.

I am after is a fairly simple setup:

2 Network interfaces each with their own private IP range
2 VLANs
Each network interface assigned to one of the above VLANs
Only 1 switch port tagged (this is used to go to dumb AP). Other 3 untagged and default to VLAN 1
1 WiFi config connected to VLAN 1 (Trusted devices, LAN/Internet access)
2nd WiFi  config connected to VLAN 2 (Untrusted devices, e.g. IoT & Guests - Internet Access only, isolated Wifi)

Firewall config is working so not required.

pixelocdguy, 2024-05-27:
As a test, I've tried to same setup on a Netgear R6220 (previously setup as a dumb AP) - this works!

Vanilla config:

```
root@OpenWrt:~# cat /etc/config/network

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
```
---

```
root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key '********'

```
---

Also works via dummy VPN:

---

```
root@OpenWrt:~# cat /etc/config/network

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config interface 'lan'
        option device 'br-lan.1'
        option proto 'static'
        option ipaddr '10.0.0.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'

```
---

```
root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:02.0/0000:02:00.0'
        option channel '6'
        option band '2g'
        option htmode 'HT40'
        option cell_density '0'
        option country 'AU'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'psk2'
        option key '********'

```
---

Given the above I am still leaning towards a bug in the firmware / device for the Netgear R8000

brada4, 2024-05-27:
Value in config/wifi is network as on first tab network/interfaces which is rendered as bridge device br-lan normally but should be br-lan.1 in your setup into `/var/run/hostapd*conf`

pixelocdguy, 2024-05-28:
Do you mean this?

config wifi-iface 'default_radio0'
        option device 'radio0'
        **option network 'lan'**

lan is the name of the network interface. It points to br-lan.1 and is working correctly on the other device R6220:

config interface 'lan'
        **option device 'br-lan.1'**

I am still not sure this is a config issue, why would it work correctly on the R6220 but not the R8000?

brada4, 2024-05-28:
Please check generated configuration files

pixelocdguy, 2024-05-28:
Sorry, I'm  not sure what config files you mean. They were setup the same via Luci, the network & wireless files are the same as per above. The only difference is the Wifi radios, but that is not the issue as the R8000 WiFi works correctly without a VPN or turning off authentication.

What are the paths to any other files you want me to post?

brada4, 2024-05-28:
> /var/run/hostapd*conf

thanks.

pixelocdguy, 2024-05-29:
```
root@OpenWrt:/var/run# cat hostapd-phy0.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=AU
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=157
chanlist=157

tx_queue_data2_burst=2.0

#num_global_macaddr=1
ieee80211n=1
ht_coex=0
ht_capab=[HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=155
vht_capab=[SHORT-GI-80][SU-BEAMFORMER][SU-BEAMFORMEE][SOUNDING-DIMENSION-3][BF-ANTENNA-3][MAX-A-MPDU-LEN-EXP0][VHT-LINK-ADAPT-3]

interface=phy0-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
wpa_passphrase=********
wpa_psk_file=/var/run/hostapd-phy0-ap0.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=OpenWrt
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
dynamic_vlan=0
vlan_naming=1
vlan_no_bridge=1
vlan_file=/var/run/hostapd-phy0-ap0.vlan
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=3c:37:86:c6:45:94
#default_macaddr
```

brada4, 2024-05-29:
Everything is correct.
```
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1
```

pixelocdguy, 2024-05-30:
I made the change but it's overwritten after a restart of the radio. It also had the same issue before and after:
```

root@OpenWrt:~# vim /var/run/hostapd-phy0.conf
root@OpenWrt:~# cat /var/run/hostapd-phy0.conf | grep br-lan
bridge=br-lan.1
snoop_iface=br-lan.1
root@OpenWrt:~# /sbin/wifi down radio0
root@OpenWrt:~# /sbin/wifi up radio0
root@OpenWrt:~# cat /var/run/hostapd-phy0.conf | grep br-lan
bridge=br-lan
snoop_iface=br-lan.1
```

I also found that via Luci web config that only br-lan is the option, it doesn't show the vlans such as br-lan.1

pixelocdguy, 2024-05-31:
```
root@OpenWrt:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-lan          7fff.3c3786c64594       no              lan4
                                                        lan2
                                                        phy0-ap0
                                                        lan3
                                                        lan1
```

pixelocdguy, 2024-06-03:
I think I have the same issue as this:

`https://github.com/openwrt/openwrt/issues/10219`

I'll see if I can implement the work around in the next few days.

pixelocdguy, 2024-06-06:
I can confirm the work around in the above thread works.

Basically, instead of setting your Interface device to a VLAN directly, you are creating a Bridge which contains the VLAN and setting your Interface device to set to the Bridge.

This appears to be a superfluous configuration requirement and as the config works fine without this on a Netgear R6220 but it is required on a Netgear R8000 (both support DSA). I am not sure why it effects some devices but not others. In both cases the config was done the same via Luci.

 I think this is a bug, and not by design. I will leave this issue open.

pixelocdguy, 2024-06-06:
I also forgot to mention, I tried rolling back to v22, same issue. Also tried the latest Daily build, same issue.

I also "demoted" my R8000 to a "dumb switch" (it still had the issue and needs to the workaround), and setup the R6220 as the router despite being an older/slower device. My logic was mainly that the R8000 is in a better part of the house for Wifi coverage and I wasn't sure if there was any security implications for the additional bridges in the workaround.

raenye, 2024-06-12:
R6220 is Mediatek, R8000 is Broadcom. Different bug set.

In any case, this is a duplicate of #10219, so beyond creating the link between them by referencing that issue in this paragraph (your link above did not do that) I don't see a reason to keep this open.

k0nstant1n0s, 2024-11-16:
I had the same issue for versions: 21.02.2 and 23.05.5 running on Raspberry Pi 4.
I can confirm the workaround of #10219 worked for 23.05.5 (I have not tested for 21.02.2).

Abdoymani, 2025-10-25:
Despite having `brcmfmac-firmware-43602a1-pcie` installed, the driver reports missing `.txt`, `.clm_blob`, and `.txcap_blob` files.

---

### What I have tried

- Installed all related packages: `kmod-brcmfmac`, `brcmfmac-firmware-43602a1-pcie`.
- Extracted multiple Netgear stock firmwares (`R8000-V1.0.4.xx.chk`) using `binwalk` — **no CLM or TXCAP blobs were found**.
- Tested firmwares from Asus RT-AC88U and Linksys EA8500 (same BCM43602 chipset) — only partial success.
- Attempted to download blobs from `linux-firmware.git` and several mirrors, but these specific files are missing for BCM43602 PCIe.
- Created a custom NVRAM `.txt` file to extend channel support; partial improvement, but TX queue timeouts persist.

---

### What I need

Could anyone please:
1. Share working versions of the following files (for BCM43602 PCIe):
   - `brcmfmac43602-pcie.txt`
   - `brcmfmac43602-pcie.clm_blob`
   - `brcmfmac43602-pcie.txcap_blob`
2. Or confirm whether these blobs are embedded in the proprietary Broadcom firmware and not distributed publicly?

Any help or pointers would be greatly appreciated.
This issue is reproducible on OpenWrt (latest snapshot, kernel 6.6.x) and affects Wi-Fi channel availability and TX stability.

---

### References
- Similar discussions on OpenWrt and linux-firmware mailing lists mention the same missing blobs for BCM43602.
- Device tree and dmesg logs available upon request.