openwrt/openwrt #15531: OPEN-MAIN  have a wrong while buliding shadowsocks-libev

suyoulin, 2024-05-19:
### Describe the bug

ERROR MESSAGE IS . checking for mbedtls_cipher_setup in -lmbedcrypto... yes
configure: error: MBEDTLS_CIPHER_MODE_CFB required
make[3]: *** [Makefile:130: /workdir/openwrt/build_dir/target-x86_64_musl/shadowsocks-libev-3.3.5/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1

The wrong hapened with Master and Main . I use 23.05 is good . 
So I use 23.05'MBEDTLS replace into Master/Main . It is working too . 
But I got another fault that I can not open make menuconfig . 

Could you trying to see if there has something you can do for me ?
Thank you . 

### OpenWrt version

main/master

### OpenWrt release

ipq400

### OpenWrt target/subtarget

generc

### Device

mr8300

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-19:
DUP: https://github.com/openwrt/packages/issues/24168 (upstream defect)

suyoulin, 2024-05-20:
> So I use 23.05'MBEDTLS replace into Master/Main . It is working too .
> But I got another fault that I can not open make menuconfig .

brada4, 2024-05-20:
Generally build system works, likely posting error message in forum will help to resolve your setup.