openwrt/openwrt #15536: filogic: xhci-mtk: reset usb disk device when write data

patrykk, 2024-05-20:
### Describe the bug

When we try to write data on the external usb disk the block device reset like below. sync parameter extend life of the device but reduce write performance to 20MB/s . Reads are  stable. Problem occur on router asus TUF AX4200 and asus TUF AX6000. Sometime we have to write 2GB, sometime more like 20GB. With sync option we can write 50GB. Problem occur on snapshots older with kernel 5.5 and newer with kernels 6.1 and 6.6 on both devices and with two different usb cases ( one with external power, both described later). 

```
[81401.651922] EXT4-fs (sda1): mounted filesystem a41c7752-47a4-49da-a475-d4256af6a2ee r/w with ordered data mode. Quota mode: disabled.
[81413.150891] usb 2-1: USB disconnect, device number 2
[81413.156189] xhci-mtk 11200000.usb: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[81413.165592] sd 0:0:0:0: [sda] tag#9 uas_zap_pending 0 uas-tag 1 inflight: CMD 
[81413.172805] sd 0:0:0:0: [sda] tag#9 CDB: opcode=0x2a 2a 00 01 fc 20 e8 00 04 00 00
[81413.180360] sd 0:0:0:0: [sda] tag#3 uas_zap_pending 0 uas-tag 2 inflight: CMD 
[81413.187566] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x2a 2a 00 01 fc 3c e8 00 04 00 00
[81413.195120] sd 0:0:0:0: [sda] tag#4 uas_zap_pending 0 uas-tag 3 inflight: CMD 
[81413.202321] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 01 fc 40 e8 00 04 00 00
[81413.209869] sd 0:0:0:0: [sda] tag#5 uas_zap_pending 0 uas-tag 4 inflight: CMD 
[81413.217070] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x2a 2a 00 01 fc 14 00 00 04 00 00
[81413.224621] sd 0:0:0:0: [sda] tag#6 uas_zap_pending 0 uas-tag 5 inflight: CMD 
[81413.231828] sd 0:0:0:0: [sda] tag#6 CDB: opcode=0x2a 2a 00 01 fc 18 00 00 04 00 00
[81413.239382] sd 0:0:0:0: [sda] tag#7 uas_zap_pending 0 uas-tag 6 inflight: CMD 
[81413.246585] sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x2a 2a 00 01 fc 1c 00 00 04 00 00
[81413.254136] sd 0:0:0:0: [sda] tag#8 uas_zap_pending 0 uas-tag 7 inflight: CMD 
[81413.261342] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x2a 2a 00 01 fc 20 00 00 00 e8 00
[81413.268891] sd 0:0:0:0: [sda] tag#11 uas_zap_pending 0 uas-tag 8 inflight: CMD 
[81413.276182] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x2a 2a 00 01 fc 24 e8 00 04 00 00
[81413.283821] sd 0:0:0:0: [sda] tag#2 uas_zap_pending 0 uas-tag 9 inflight: CMD 
[81413.291024] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x2a 2a 00 01 fc 28 e8 00 04 00 00
[81413.298577] sd 0:0:0:0: [sda] tag#10 uas_zap_pending 0 uas-tag 10 inflight: CMD 
[81413.305956] sd 0:0:0:0: [sda] tag#10 CDB: opcode=0x2a 2a 00 01 fc 2c e8 00 04 00 00
[81413.313595] sd 0:0:0:0: [sda] tag#12 uas_zap_pending 0 uas-tag 11 inflight: CMD 
[81413.320973] sd 0:0:0:0: [sda] tag#12 CDB: opcode=0x2a 2a 00 01 fc 30 e8 00 04 00 00
[81413.328610] sd 0:0:0:0: [sda] tag#0 uas_zap_pending 0 uas-tag 12 inflight: CMD 
[81413.335901] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 01 fc 34 e8 00 04 00 00
[81413.343453] sd 0:0:0:0: [sda] tag#1 uas_zap_pending 0 uas-tag 13 inflight: CMD 
[81413.350742] sd 0:0:0:0: [sda] tag#1 CDB: opcode=0x2a 2a 00 01 fc 38 e8 00 04 00 00
[81413.358387] sd 0:0:0:0: [sda] tag#9 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.367943] sd 0:0:0:0: [sda] tag#9 CDB: opcode=0x2a 2a 00 01 fc 20 e8 00 04 00 00
[81413.375520] I/O error, dev sda, sector 33300712 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.384772] sd 0:0:0:0: [sda] tag#3 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.394335] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x2a 2a 00 01 fc 3c e8 00 04 00 00
[81413.401908] I/O error, dev sda, sector 33307880 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.411185] sd 0:0:0:0: [sda] tag#4 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.420748] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 01 fc 40 e8 00 04 00 00
[81413.428296] I/O error, dev sda, sector 33308904 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.437576] sd 0:0:0:0: [sda] tag#5 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.447136] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x2a 2a 00 01 fc 14 00 00 04 00 00
[81413.454715] I/O error, dev sda, sector 33297408 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.463957] sd 0:0:0:0: [sda] tag#6 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.473510] sd 0:0:0:0: [sda] tag#6 CDB: opcode=0x2a 2a 00 01 fc 18 00 00 04 00 00
[81413.481065] I/O error, dev sda, sector 33298432 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.490309] sd 0:0:0:0: [sda] tag#7 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.499876] sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x2a 2a 00 01 fc 1c 00 00 04 00 00
[81413.507431] I/O error, dev sda, sector 33299456 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.516702] sd 0:0:0:0: [sda] tag#8 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.516784] sd 0:0:0:0: [sda] tag#9 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
[81413.526266] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x2a 2a 00 01 fc 20 00 00 00 e8 00
[81413.526270] I/O error, dev sda, sector 33300480 op 0x1:(WRITE) flags 0x0 phys_seg 29 prio class 2
[81413.535811] sd 0:0:0:0: [sda] tag#9 CDB: opcode=0x2a 2a 00 01 fc 44 e8 00 03 18 00
[81413.543356] EXT4-fs warning (device sda1): ext4_end_bio:343: I/O error 10 writing to inode 14 starting block 4162560)
[81413.552207] I/O error, dev sda, sector 33309928 op 0x1:(WRITE) flags 0x0 phys_seg 99 prio class 2
[81413.559763] sd 0:0:0:0: [sda] tag#11 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.588842] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x2a 2a 00 01 fc 24 e8 00 04 00 00
[81413.596508] I/O error, dev sda, sector 33301736 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.605756] sd 0:0:0:0: [sda] tag#2 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=0s
[81413.615339] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x2a 2a 00 01 fc 28 e8 00 04 00 00
[81413.622900] I/O error, dev sda, sector 33302760 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 2
[81413.632171] EXT4-fs warning (device sda1): ext4_end_bio:343: I/O error 10 writing to inode 14 starting block 4163741)
[81413.642845] Buffer I/O error on device sda1, logical block 4162333
[81413.646139] EXT4-fs warning (device sda1): ext4_end_bio:343: I/O error 17 writing to inode 14 starting block 4165760)
[81413.649032] Buffer I/O error on device sda1, logical block 4162334
[81413.665796] Buffer I/O error on device sda1, logical block 4162335
[... 82 lines cut ...]
### Actual behaviour

Reset usb block device.

### Expected behaviour

Stability

### Additional info

Testred with two disks and two routers :
routers : Asus TUF AX4200
Asus TUF AX6000

disks:
nvme disk in the RTL9210B-CG usb case ( without external power )
3,5 WD black disk in the 152d:0578 JMicron USB to ATA/ATAPI Bridge usb case with external power 

Both disks are stable with RPI4 or x86 device. 

Tested on official snapshot bild with kernel 5.x 6.1.x and 6.6.x

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

patrykk, 2024-05-20:
> Is this from a clean repository?

Yes

brada4, 2024-05-21:
kmod-iosched-bfq will sync more often and likely help with fake NCQ over USB

patrykk, 2024-05-21:
> kmod-iosched-bfq will sync more often and likely help with fake NCQ over USB

Hi, thanks for advice but it will not solve root cause and probably will improve only stability a little bit.

brada4, 2024-05-21:
Uninstall uas driver, no ncq no tears

patrykk, 2024-05-21:
> Uninstall uas driver, no ncq no tears

Without uas I cant detect these disks. Is there other kernel drivers which supports these device classes ?

brada4, 2024-05-21:
usb-storage driver must support each and every uac disk.

patrykk, 2024-05-21:
> usb-storage driver must support each and every uac disk.
When I remove uas :

root@router3:~# lsusb -t
/:  Bus 001.Port 001: Dev 001, Class=root_hub, Driver=xhci-mtk/2p, 480M
/:  Bus 002.Port 001: Dev 001, Class=root_hub, Driver=xhci-mtk/1p, 20000M/x2
    |__ Port 001: Dev 002, If 0, Class=[unknown], **Driver=[none],** 5000M/x2
root@router3:~# lsusb 
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.30 xhci-hcd xHCI Host Controller
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.30 xhci-hcd xHCI Host Controller
Bus 002 Device 002: ID 0bda:9210 Realtek RTL9210B-CG

brada4, 2024-05-21:
same enclosure has problems... 
https://github.com/raspberrypi/linux/issues/4130

brada4, 2024-05-21:
> Realtek RTL9210B-CG

yeah,  totally different, just strange you dont enjoy your wildly enhanced uas mode.

brada4, 2024-05-21:
rebuild uas with patch. 
https://patchwork.kernel.org/project/linux-usb/patch/20240117141132.1023457-1-bruno@clisp.org/

patrykk, 2024-05-22:
> Alternatively try with some "rolling" linux distribution just to confirm that there is no UAS for this device and it works.

```
linux-6.9.1]$ grep -R 0x9210 * -A 2 -B 2 | grep 0bda -A 2 -B 2
--
drivers/usb/storage/uas-detect.h-	 */
drivers/usb/storage/uas-detect.h-	if (le16_to_cpu(udev->descriptor.idVendor) == 0x0bda &&
drivers/usb/storage/uas-detect.h:			le16_to_cpu(udev->descriptor.idProduct) == 0x9210 &&
drivers/usb/storage/uas-detect.h-			(udev->manufacturer && !strcmp(udev->manufacturer, "HIKSEMI")) &&
$
```

brada4, 2024-05-22:
It has to be backporyed in this case...

brada4, 2024-05-29:
The newest kernel disables uas for particulat usb target, whose problem exhibits independently of host.

brada4, 2024-05-29:
It is happening on raspberry if you google. I am baffled why storage driver does not attach without uas though.

patrykk, 2024-05-31:
ok, so probably you are right with  RTL9210B-CG. This device works without UAS. The second one unfortunately not. 
Openwrt usb_storage module ignore parameters when you load module with modprobe or modprobe.conf defined parameters. The workaround looks like :
a) remove file: /etc/modules.d/usb-storage-uas
b) add to the rc.local:
```
echo -n "quirks=0bda:9210:u" > /sys/module/usb_storage/parameters/quirks 
usbreset 0bda:9210
echo "0bda 9210" > /sys/module/usb_storage/drivers/usb:usb-storage/new_id
```

BlaY0, 2024-12-12:
Adding `quirks=<vid>:<pid>:u` to `/etc/modules.d/usb-storage` also works.