openwrt/openwrt #15560: kmod-hwmon-rpi-poe-fan for 23.05.3

cannonfodder728, 2024-05-25:
### Describe the bug

I see that this package is only available for 22.05. Wondering if one will be released for 23.05.3?

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.03

### OpenWrt target/subtarget

bcm27xx/bcm2711

### Device

Raspberry Pi 4 Model B Rev 1.1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

cannonfodder728, 2024-05-25:
yes according to this page it is maintained by openwrt team
https://openwrt.org/packages/pkgdata/kmod-hwmon-rpi-poe-fan

i upgraded to 23.05.3 from 22.03.5 and this module is no longer available. It controls the fan on the poe hat for rpi, so im now facing thermal throttling issues without the fan module.

robimarko, 2024-05-29:
I think that it was replaced with `kmod-pwm-raspberrypi-poe`