openwrt/openwrt #15561: Kernel crash on OOM on 5.15, 6.1, not 6.6

darkk, 2024-05-25:
### Describe the bug

The router crashes under memory pressure: either `dnsmasq` with a large blocklist or simple `stress`. Watchdog reboots the router.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ath79/mikrotik

### Device

MikroTik RouterBOARD mAPL-2nD (mAP lite)

### Image kind

Official downloaded image

### Steps to reproduce

```
opkg update && opkg install stress
free -h
over=$(awk '/MemTotal/ {print 16 + int($2 / 1024)}' < /proc/meminfo)
under=$(( $over - 32 ))
( set -o xtrace; for i in `seq $under $over`; do stress --timeout 1 --vm 1 --vm-bytes ${i}m; done )
```

### Actual behaviour

Kernel crashes at some point and the loop does not finish: the shell is non-responding, the router does not respond to ICMP pings anymore, nothing is logged `logread -f` or `netconsole`. Looks like crash. The router is eventually rebooted. Looks like watchdog to me.

### Expected behaviour

OOM killer killing the process instead of kernel crash happening, Memory allocation failure is also a good thing to expect.

### Additional info

Following boards are affected, the issue is 100% reproducable:
- bcm27xx/bcm2708, Raspberry Pi Model B Rev 2, 23.05.0, r23497-6637af95aa, 5.15.134 #0 Mon Oct 9 21:45:35 2023 armv6l
- ath79/mikrotik, MikroTik RouterBOARD mAPL-2nD (mAP lite), 23.05.3, r23809-234f1a2efa, 5.15.150 #0 Fri Mar 22 22:09:42 2024 mips
- ramips/mt7620, SNR-CPE-W4N (rev.M), 23.05.3, r23809-234f1a2efa, 5.15.150 #0 Fri Mar 22 22:09:42 2024 mips
- ath79/generic, TP-Link TL-WR842N v3, 23.05.3, r23809-234f1a2efa, 5.15.150 #0 Fri Mar 22 22:09:42 2024 mips
- ath79/generic, TP-Link TL-WR842N v3, SNAPSHOT, r26404-a55ab9e134, 6.1.89 #0 Sat May 25 10:41:26 2024 mips

I fail to reproduce the issue on the following boards, it's unclear if they're affected:
- bcm27xx/bcm2709, Raspberry Pi 3 Model B Rev 1.2, 23.05.3, r23809-234f1a2efa, 5.15.150 #0 SMP Fri Mar 22 22:09:42 2024 armv7l
- ramips/mt7620, SNR-CPE-W4N (rev.M), SNAPSHOT, r26404-a55ab9e134 6.6.30 #0 Sat May 25 10:41:26 2024 mips

SNR and WR842N boards are the same boards running different versions, so it's unlikely a hardware issue.

```
vm.overcommit_kbytes = 0
vm.overcommit_memory = 0
vm.overcommit_ratio = 50
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-25:
Looks like netifd was killed.
`echo -17 > /proc/<pid of netifd>/oom_adj`
Also will help to disable stress oom_adj manipulation.

brada4, 2024-05-26:
What I suspect happens - init (procd, pid 0) gets killed and system reboots with timer.
I think this makes it stay and restart processes, you may need whole process chain leading to your logs "protected"
` echo -17 > /proc/0/oom_adj`

darkk, 2024-05-26:
Thanks for hypotheses!

> like `netifd` was killed

I doubt that as `kill -9 $(pidof netifd)` does not make SSH session freeze.

> disable stress oom_adj manipulation

As far as I can see, `stress` does not tune oom_*_adj variables.

> `init` (`procd`, pid 0) gets killed

I doubt that as well as OOM killer does not usually kill pid=1 according to [oom_kill.c](https://github.com/torvalds/linux/blob/c13320499ba0efd93174ef6462ae8a7a2933f6e7/mm/oom_kill.c#L162-L170).

`echo -17 > /proc/1/oom_adj` does not help as a workaround, `stress` still triggers the issue.

brada4, 2024-05-26:
You need to exclude everything involved in your logging chain, procd is supposed to restart rest if those are killed. Probably zram-swap  can let your task finish