openwrt/openwrt #15573: AVM FRITZ Repeater 1200 - Using more than one of these devices breaks complete LAN

ojaksch, 2024-05-27:
### Describe the bug

Having three of these devices. Simple setup w/o extras like VLAN etc but wpad-mesh-mbedtls to extend my long-time working existing mesh (consisting of TL-WR1043ND-xx). Using **one** AVM 1200 is great, no problem at all. As soon as I add another one, both/all AVM 1200 are sporadically reporting

> received packet on [network device] with own address as source address

as well as my server (with a bridged network device). The complete LAN is unreachable then.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Repeater 1200

### Image kind

Official downloaded image

### Steps to reproduce

Connect LAN cable between the AVM 1200 and a network switch, setup an AP on OpenWRT. Repeat with another AVM 1200.

### Actual behaviour

Kernel messages of

> received packet on [network device] with own address as source address

appearing sporadically which bothers the complete network

### Expected behaviour

No errors nor problems on such a simple setup.

### Additional info

/etc/config/network

```
config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd15:ecf9:87da::/48'

config device
        option name 'br-lan'
        option type 'bridge'

config interface 'lan'
        option device 'br-lan'
        option proto 'dhcp'
```
Also tested with `option stp '1'` and/or `option multicast '0'` and/or `option packet_steering '1'`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jmv2009, 2024-05-27:
It appears you are having a mesh AND a switch, suggesting you have a route back to the devices. So having packets come back to the devices should not be that surprising. So were you always running 23.05.3 on all your devices? @robimarko

jmv2009, 2024-05-29:
I would try to capture the packets going through the networking, try to understand the differences that are occurring, AND under the different circumstances. Try to determine
a) Under what circumstances do packets get back to the device itself (always/never/sometimes/only when ...), on which interface with which mac address with with mac address.
b) Whether the second device causes the packets to go back to the first devices, either through the router or through the mesh.
c) Does it really need both the router and the mesh?
In the end, try to portray the path that the packets takes.
It may be important to have a promiscuous capture adaptor on the wireless and the wired network. Be aware that a switch (vs a hub) only forwards the packets to the intended destination.

ojaksch, 2024-05-30:
Thanks for any suggestions. While inspecting my network isn't practical for reasons (no time, to less knowledge about such tasks), I began with point c:

> c) Does it really need both the router and the mesh?

No. The AVM 1200's are primarily intended for Smart Home devices _and_ normal operation with WiFi devices like tablets etc. I thought to myself, while I'm at it, I could add the mesh to them too. 
I disabled the mesh on two AVM 1200 this morning and network is error-free for hours now.

And that's what I don't understand: Why is the mesh working fine for years with three (meshed) TL-WR1043ND's, but struggling as soon as I add >=2 AVM 1200?

robimarko, 2024-06-21:
Try the latest images that contain:
https://github.com/openwrt/openwrt/commit/0b6b54eca939a2cb757f854594863548fb93f21f

ojaksch, 2024-06-22:
> Try the latest images that contain: [0b6b54e](https://github.com/openwrt/openwrt/commit/0b6b54eca939a2cb757f854594863548fb93f21f)

Thanks for the headup. I tried the latest snapshot this morning, but this lead to another problem/bug as wpa_supplicant isn't working reliably, at least for the mesh, and is complaining about

> Sat Jun 22 06:14:47 2024 daemon.err wpa_supplicant[1025]: Line 12: invalid number "2412.0"
Sat Jun 22 06:14:47 2024 daemon.err wpa_supplicant[1025]: Line 12: failed to parse frequency '2412.0'.
Sat Jun 22 06:14:47 2024 daemon.err wpa_supplicant[1025]: Line 18: failed to parse network block.
Sat Jun 22 06:14:47 2024 daemon.err wpa_supplicant[1025]: Failed to read or parse configuration '/var/run/wpa_supplicant-mesh0.conf'.

which is
```country=DE
network={
        ssid="mymesh"
        key_mgmt=SAE
        mode=5
        mesh_fwding=1
        mesh_rssi_threshold=0
        fixed_freq=1
        frequency=2412.0
        ht40=1
        disable_vht=1
        noscan=1
        sae_password="mypw"
        beacon_int=100
}
```
The culprit here is `frequency=2412.0` which normally reads `frequency=2412`, but this values changes automagically back to the wrong value after reboot or command wifi.
Back to stable w/o mesh for now.

robimarko, 2024-06-22:
You have hit a fresh bug, fix is already posted:
https://github.com/openwrt/openwrt/pull/15770

ojaksch, 2024-06-22:
Ha! I'll try the next nightly then and report back...

ojaksch, 2024-06-24:
Started testing with two boxes this morning, nine hours ago, and nothing worth reporting happen'd 'til now.
Will add a 3rd tomorrow.

ojaksch, 2024-07-04:
Mesh continuously working with all three WR1043ND and three AVM 1200, no `received packet on [network device] with own address as source address` ever seen and the complete local network is stable. But what I'm observing are "micro disconnects" from WAN. The Internet connection stalls for a few seconds on the complete LAN and then continues as normal, but no (kernel) message everywhere and router (a FritzBox) hasn't recorded any disconnection or inaccessibility.

I disabled the mesh on all AVM 1200 for now to see if that helps or to catch a new/unknown Gizmo here...

ojaksch, 2024-08-13:
I'm closing this issue as it looks like that all problems are gone. All three AVM 1200 are working fine with the 23.05.4 release. Dunno where these "micro disconnects" came from but they're also gone.

Thanks to all participants for tips and suggestions! :heart:

closed 2024-08-13 by ojaksch