openwrt/openwrt #15577: syslog server not receiving any rejected messages

user23498723452, 2024-05-27:
### Describe the bug

When properly configured, no rejected traffic messages are sent to the remote syslog server, even though the relevant messages appear in the system log (and kernel log).

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Asus RT-AC68U (BCM4708)

### Image kind

Official downloaded image

### Steps to reproduce

1. Configure remote syslog (tcp port 514) to a remote IP, with the following:
`
config system
        option hostname 'OpenWrt'
        option log_size '300'
        option log_ip '192.168.1.11'
        option log_proto 'tcp'
        option conloglevel '8'
        option cronloglevel '0'
        option log_port '514'
`
2. Restart log and system services.
3. Note that there is a syslog mssage received at the remote system like this: `<30>1 2024-05-27T17:26:51-04:00 OpenWrt logread 13023 - - logread[13023]: Logread connected to 192.168.1.11:514 via tcp`
4. Enable logging on all firewall zones (which should log denied traffic by default)
5. setup a rule to reject specific network traffic. 
6. perform the relevant network traffic (which is rejected)
7. The system log shows the rejected messages.

### Actual behaviour

the remote syslog server does not receive the rejected traffic messages.

### Expected behaviour

the remote syslog server should receive the rejected traffic messages

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

user23498723452, 2024-05-28:
Digging into this more, the issue appears to be related to a a non-standard syslog message format used for the firewall messages.  A non standard format is being sent as best I can tell.  It is not RFC-3164 nor RFC-5424.

here is an example of one of these firewall messages.  You can note how the format is NOT the same as the one received in my ticket description.  This is the issue.  Migrate the openwrt firewall logs to RFC5424 and this will be solved.  I have no clue what "format" this is,  but its not any standard.  It might be possible to get there by starting with updating the date format used.  I have no idea if I can control that or not.

`<4>Mon May 27 09:42:57 2024 kern.warn kernel: [68681.183689] reject wan out: IN=br-lan.2 OUT=wan MAC=e0:3f:49:f2:2a:d8:24:5e:be:51:6c:34:08:00 SRC=192.168.1.11 DST=1.2.3.4 LEN=52 TOS=0x00 PREC=0x00 TTL=63 ID=57238 DF PROTO=TCP SPT=39966 DPT=443 WINDOW=65535 RES=0x00 SYN URGP=0`

user23498723452, 2024-05-30:
I want to note that I have tried to install and configure rsyslog to fix this,  but I don't think it works at all.  I know that is dealt with elsewhere,  but it needed to be stated.

john-tho, 2024-06-08:
To try to work out where the parsing failure is could you please provide samples of one of these rejected packet log messages before it is sent to remote, possibly from `dmesg` and `logread`, as well as when being sent to remote via `tcpdump`, and at a remote via `ncat`?

kernel [`nf_log_syslog.c`](https://elixir.bootlin.com/linux/v6.6.32/source/net/netfilter/nf_log_syslog.c) does the _logging_
OpenWrt [`ubox` `logread`](https://github.com/openwrt/openwrt/blob/b9650de249c83685254973c9349fba3d48f3253c/package/system/ubox/files/log.init#L80) does the remote sending [`LOG_NET`](https://lxr.openwrt.org/ident?i=LOG_NET)

Adding rfc3164 remote logging: https://github.com/openwrt/ubox/commit/5c45b560bc8c9e13682269ed963a8a4a65959518

https://github.com/openwrt/ubox/blob/85f1053019caf4cd333795760950235ee4529ba7/log/logread.c#L159-L190:

On main/snapshot, some additional processing of kernel timestamp was added here: https://github.com/openwrt/ubox/commit/c08709cceb554cba02c935d1442f6a042fe6b2a8