openwrt/openwrt #15585: EA8500 (ipq806x)  not recieving dhcp offers on wan

dandudikof, 2024-05-29:
### Describe the bug

fresh sysupgrade from 23.05.3 with reset

not receiving or dhcp on wan interface.
(can see requests and responses on dhcp server device)

### OpenWrt version

r26436-7514d9bb6f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

Linksys EA8500 WiFi Router

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-05-29:
`ethtool wan` -> is the speed negotiated ok

dandudikof, 2024-05-29:
ethtool unavailable as i have not installed it manually 
(no tcpdump ether on snapshots image)

(same cable and switch port works fine on 23.05.3)

logread shows good link negotiations (same with ip link)
(and lights on switch confirm)

tcpdumping the other side of udp (dhcp requests)
are getting trough and are being responded to.

tcpdumping the other side of icmps (pings) 
trigger ARP requests and are being responded to.

ErwanMAS, 2024-11-02:
@dandudikof what is your change ?

You change the mac address for wan , for a different value than lan ?

egc112, 2025-01-08:
Ran into the same problem updating form 23.05.5 to 24.10 RC5
Although it was easy to increase the WAN MAC value with one and after that everything worked fine.

mircea-vutcovici, 2025-01-26:
For my Linksys EA8500 OpenWrt 24.10.0-rc6 the fix was to delete the MAC from the config:
https://github.com/openwrt/openwrt/issues/17157#issuecomment-2614250161

commit eba2fbf6 (2025-02-03): ipq806x: fix broken WAN on Linksys EAX and Asrock G10

Fix broken WAN on Linksys EAX and Asrock G10 by incrementing the WAN
interface MAC address + 1. This caused conflicting entry in the FDB
table and caused the WAN port to malfunction with the DSA conversion.

Fixes: #17157
Fixes: #15585
Fixes: #16604
Link: https://github.com/openwrt/openwrt/pull/17839
commit cd747cb3 (2025-02-03): ipq806x: fix broken WAN on Linksys EAX and Asrock G10

Fix broken WAN on Linksys EAX and Asrock G10 by incrementing the WAN
interface MAC address + 1. This caused conflicting entry in the FDB
table and caused the WAN port to malfunction with the DSA conversion.

Fixes: #17157
Fixes: #15585
Fixes: #16604
Link: https://github.com/openwrt/openwrt/pull/17839

(cherry picked from commit eba2fbf63814bbcf8c99c98177818476fca808a7)
closed 2025-02-03 by openwrt-bot