openwrt/openwrt #15588: an accidental error when configuring sane led to a partial problem with the router's operation as a repeater

saber716rus, 2024-05-30:
### Describe the bug

the problem is that due to an accidental sane error, which led to a partial malfunction of the router as a repeater. even resetting to factory settings does not help solve this problem. the devices do not want to connect to the repeater, everything was fine before, and if they do, then there may not be an Internet connection. you need to reboot both the repeater and the main router.

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

1)configure the repeater mode using relayd according to the instructions.
2)Configure p910nd
3)try to set up sane.

### Actual behaviour

it led to an accidental error, which in turn partially disrupted the operation of the repeater mode. Also, if I connect to the repeater's access point, I can't log in either to the web interface or via ssh. 

### Expected behaviour

both the print server and the network scanner should work, and clients should connect without problems and work quietly on the Internet. 

### Additional info

logs uploaded to Yandex.disk 
https://disk.yandex.ru/d/ljROE8sQt7EjqA
https://disk.yandex.ru/d/Wcb1B5slLUf0GA

### Diffconfig

```text
-ash: ./scripts/diffconfig.sh: not found
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

saber716rus, 2024-05-30:
> Nothing in the logs illustrate any stain of saned being used.

I had to reset it, but the problem with the repeater mode and access points remained.

brada4, 2024-05-30:
There is nothing in the logs in that regard. Increase hostap and supplicant log amounts (level 1 ipo 2)

saber716rus, 2024-05-30:
> There is nothing in the logs in that regard. Increase hostap and supplicant log amounts (level 1 ipo 2)

I did it on this wiki

https://openwrt.org/docs/guide-developer/debugging

saber716rus, 2024-05-30:
There must be an extended log. After the reboot, everything works until I leave the repeat zone.
```
Thu May 30 18:50:49 2024 kern.info kernel: [    0.560355] mt7621_gpio 1e000600.gpio: registering 32 gpios
Thu May 30 18:50:49 2024 kern.info kernel: [    0.566374] mt7621_gpio 1e000600.gpio: registering 32 gpios
Thu May 30 18:50:49 2024 kern.info kernel: [    0.572543] mt7621-pci 1e140000.pcie: host bridge /pcie@1e140000 ranges:
Thu May 30 18:50:49 2024 kern.info kernel: [    0.579216] mt7621-pci 1e140000.pcie:   No bus range found for /pcie@1e140000, using [bus 00-ff]
Thu May 30 18:50:49 2024 kern.info kernel: [    0.587983] mt7621-pci 1e140000.pcie:      MEM 0x0060000000..0x006fffffff -> 0x0060000000
Thu May 30 18:50:49 2024 kern.info kernel: [    0.596091] mt7621-pci 1e140000.pcie:       IO 0x001e160000..0x001e16ffff -> 0x0000000000
Thu May 30 18:50:49 2024 kern.info kernel: [    0.961348] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
Thu May 30 18:50:49 2024 kern.info kernel: [    0.968243] mt7621-pci 1e140000.pcie: PCIE0 enabled
Thu May 30 18:50:49 2024 kern.info kernel: [    0.973126] mt7621-pci 1e140000.pcie: PCIE1 enabled
Thu May 30 18:50:49 2024 kern.info kernel: [    0.977937] PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
Thu May 30 18:50:49 2024 kern.info kernel: [    0.985096] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
Thu May 30 18:50:49 2024 kern.info kernel: [    0.991391] pci_bus 0000:00: root bus resource [bus 00-ff]
Thu May 30 18:50:49 2024 kern.info kernel: [    0.996813] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.003669] pci_bus 0000:00: root bus resource [io  0x0000-0xffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.009822] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
Thu May 30 18:50:49 2024 kern.info kernel: [    1.015772] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.021994] pci 0000:00:00.0: reg 0x14: [mem 0x60600000-0x6060ffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.028241] pci 0000:00:00.0: supports D1
Thu May 30 18:50:49 2024 kern.info kernel: [    1.032175] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
Thu May 30 18:50:49 2024 kern.info kernel: [    1.038733] pci 0000:00:01.0: [0e8d:0801] type 01 class 0x060400
Thu May 30 18:50:49 2024 kern.info kernel: [    1.044747] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x7fffffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.050926] pci 0000:00:01.0: reg 0x14: [mem 0x60610000-0x6061ffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.057240] pci 0000:00:01.0: supports D1
Thu May 30 18:50:49 2024 kern.info kernel: [    1.061158] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
Thu May 30 18:50:49 2024 kern.info kernel: [    1.069020] pci 0000:01:00.0: [14c3:7916] type 00 class 0x000280
Thu May 30 18:50:49 2024 kern.info kernel: [    1.075051] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.082245] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x00003fff 64bit pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.089379] pci 0000:01:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.096692] pci 0000:01:00.0: supports D1 D2
Thu May 30 18:50:49 2024 kern.info kernel: [    1.100869] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
Thu May 30 18:50:49 2024 kern.info kernel: [    1.107503] pci 0000:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:00.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
Thu May 30 18:50:49 2024 kern.info kernel: [    1.123627] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.128788] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.134879] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.141603] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x602fffff pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.148744] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
Thu May 30 18:50:49 2024 kern.info kernel: [    1.155648] pci 0000:02:00.0: [14c3:7915] type 00 class 0x000280
Thu May 30 18:50:49 2024 kern.info kernel: [    1.161675] pci 0000:02:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.168830] pci 0000:02:00.0: reg 0x18: [mem 0x00000000-0x00003fff 64bit pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.176044] pci 0000:02:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.183321] pci 0000:02:00.0: supports D1 D2
Thu May 30 18:50:49 2024 kern.info kernel: [    1.187500] pci 0000:02:00.0: PME# supported from D0 D1 D2 D3hot D3cold
Thu May 30 18:50:49 2024 kern.info kernel: [    1.194145] pci 0000:02:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x1 link at 0000:00:01.0 (capable of 4.000 Gb/s with 5.0 GT/s PCIe x1 link)
Thu May 30 18:50:49 2024 kern.info kernel: [    1.210325] pci 0000:00:01.0: PCI bridge to [bus 02-ff]
Thu May 30 18:50:49 2024 kern.info kernel: [    1.215542] pci 0000:00:01.0:   bridge window [io  0x0000-0x0fff]
[... cut ...]
[   31.521979] br-lan: port 5(phy1-ap0) entered blocking state
[   31.527567] br-lan: port 5(phy1-ap0) entered forwarding state
[   35.520696] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[   35.527836] br-lan: port 4(phy0-ap0) entered blocking state
[   35.533526] br-lan: port 4(phy0-ap0) entered forwarding state
[  430.391557] mt7530-mdio mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
[  430.399533] br-lan: port 1(lan1) entered blocking state
[  430.404805] br-lan: port 1(lan1) entered forwarding state
```

saber716rus, 2024-05-30:
in general, after investigating a little, I found that the 5GHz access point is not yet functioning on the repeater, and if I connect to it, then either there is no Internet at all or the speed is very low.

https://www.speedtest.net/result/16316688824 - this measurement was made on a 5 GHz network.
```
Thu May 30 20:30:22 2024 daemon.notice hostapd: SAE: 3a:6f:70:ef:32:05 indicates support for SAE H2E, but did not use it
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2412 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2417 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2422 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2427 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2432 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2437 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2442 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2447 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
Thu May 30 20:30:22 2024 daemon.notice hostapd: ACS: Survey for freq 2452 is missing noise floor
[... cut ...]
Thu May 30 20:32:08 2024 daemon.debug hostapd: phy0-ap0: STA 84:3e:1d:28:af:80 WPA: received EAPOL-Key frame (4/4 Pairwise)
Thu May 30 20:32:08 2024 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 84:3e:1d:28:af:80 auth_alg=sae
Thu May 30 20:32:08 2024 daemon.debug hostapd: phy0-ap0: STA 84:3e:1d:28:af:80 IEEE 802.1X: authorizing port
Thu May 30 20:32:08 2024 daemon.info hostapd: phy0-ap0: STA 84:3e:1d:28:af:80 WPA: pairwise key handshake completed (RSN)
Thu May 30 20:32:08 2024 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 84:3e:1d:28:af:80
```

saber716rus, 2024-05-30:
> Check the box in br-lan to bring up empty bridge. logread includes dmesg, no need to post later smaller log. There is client disconnect followed by client successful connect in the logs' end?

I think so. But there is also one thing that the repeater periodically disconnects from the main router, then connects back. There have been 3 or 4 times today, but so far there have been no such outages in the evening.

[image]

when connecting to the repeater access point, I cannot access the web interface.

[image]

[image]

saber716rus, 2024-05-30:
connected to the 5GHz repeater network, the Internet leaves much to be desired. He's right next to me.
https://www.speedtest.net/result/16317122346 - my Internet connection speed is 100mbps.

```
SSID:	SecNET
Protocol: Wi-Fi 5 (802.11ac)
Security type: WPA personal
Manufacturer: Realtek Semiconductor Corp.
Description: Realtek 8822CE Wireless LAN 802.11n PCI-E NIC
Driver version: 2024.0.9.232
Network range: 5 GHz
Network channel: 60
Line speed (reception and transmission): 585/585 (Mbps)
Local IPv6 channel address: fe80::12a4:f942:7f17:d70a%4
IPv4 address: 192.168.1.81
IPv4 DNS servers: 192.168.1.1 (without encryption)
Physical Address (MAC): 84-3E-1D-28-AF-80
```

saber716rus, 2024-05-30:
information about the repeater and main router signal
[image]
[image]

saber716rus, 2024-05-30:
just a couple of minutes ago, the repeater's radio streams were rebooted, the Internet has stabilized a little, but the behavior is still strange. 

```
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: authentication OK (open system)
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-AUTHENTICATE.indication(3a:6f:70:ef:32:05, OPEN_SYSTEM)
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-DELETEKEYS.request(3a:6f:70:ef:32:05)
Thu May 30 22:49:36 2024 daemon.info hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: authenticated
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: No PMKSA cache entry found for SAE
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: start SAE authentication (RX commit, status=0 (SUCCESS))
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: SAE authentication (RX confirm, status=0 (SUCCESS))
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-AUTHENTICATE.indication(3a:6f:70:ef:32:05, unknown)
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-DELETEKEYS.request(3a:6f:70:ef:32:05)
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: event 0 notification
Thu May 30 22:49:36 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: No PMKSA cache entry found for SAE
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: authentication OK (open system)
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-AUTHENTICATE.indication(3a:6f:70:ef:32:05, OPEN_SYSTEM)
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-DELETEKEYS.request(3a:6f:70:ef:32:05)
Thu May 30 22:49:40 2024 daemon.info hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: authenticated
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: start SAE authentication (RX commit, status=0 (SUCCESS))
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: SAE authentication (RX confirm, status=0 (SUCCESS))
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-AUTHENTICATE.indication(3a:6f:70:ef:32:05, unknown)
Thu May 30 22:49:40 2024 daemon.debug hostapd: phy0-ap0: STA 3a:6f:70:ef:32:05 WPA: event 0 notification
Thu May 30 22:49:40 2024 daemon.notice hostapd: SAE: 3a:6f:70:ef:32:05 indicates support for SAE H2E, but did not use it
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: authentication OK (open system)
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: event 0 notification
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-AUTHENTICATE.indication(3a:6f:70:ef:32:05, OPEN_SYSTEM)
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-DELETEKEYS.request(3a:6f:70:ef:32:05)
Thu May 30 22:49:45 2024 daemon.info hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: authenticated
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: PMKID found from PMKSA cache eap_type=0 vlan=0
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: association OK (aid 1)
Thu May 30 22:49:45 2024 daemon.info hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: associated (aid 1)
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-REASSOCIATE.indication(3a:6f:70:ef:32:05)
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 MLME: MLME-DELETEKEYS.request(3a:6f:70:ef:32:05)
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.11: binding station to interface 'phy1-ap0'
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: event 1 notification
Thu May 30 22:49:45 2024 daemon.notice hostapd: phy0-ap0: Prune association for 3a:6f:70:ef:32:05
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: start authentication
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.1X: unauthorizing port
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: sending 1/4 msg of 4-Way Handshake
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: received EAPOL-Key frame (2/4 Pairwise)
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: sending 3/4 msg of 4-Way Handshake
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: received EAPOL-Key frame (4/4 Pairwise)
Thu May 30 22:49:45 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 3a:6f:70:ef:32:05 auth_alg=open
Thu May 30 22:49:45 2024 daemon.debug hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 IEEE 802.1X: authorizing port
Thu May 30 22:49:45 2024 daemon.info hostapd: phy1-ap0: STA 3a:6f:70:ef:32:05 WPA: pairwise key handshake completed (RSN)
Thu May 30 22:49:45 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 3a:6f:70:ef:32:05
[... cut ...]
Thu May 30 22:50:59 2024 daemon.debug hostapd: phy1-ap0: STA 84:3e:1d:28:af:80 WPA: received EAPOL-Key frame (4/4 Pairwise)
Thu May 30 22:50:59 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 84:3e:1d:28:af:80 auth_alg=sae
Thu May 30 22:50:59 2024 daemon.debug hostapd: phy1-ap0: STA 84:3e:1d:28:af:80 IEEE 802.1X: authorizing port
Thu May 30 22:50:59 2024 daemon.info hostapd: phy1-ap0: STA 84:3e:1d:28:af:80 WPA: pairwise key handshake completed (RSN)
Thu May 30 22:50:59 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 84:3e:1d:28:af:80
```

brada4, 2024-05-30:
non-greenfield 2.4ghz 40Mhz is most likely at fault combined with week signal. 
Please make your killometer long logs into attachments and describe your claim in original post , as it sounds absolutely absurd that sane damaged router permanently.

saber716rus, 2024-05-31:
> non-greenfield 2.4ghz 40Mhz is most likely at fault combined with week signal. 
> Please make your killometer long logs into attachments and describe your claim in original post , as it sounds absolutely absurd that sane damaged router permanently.

the error appeared with the repeater, after an accidental error when setting up sane and then resetting to factory settings, but the problem still persisted.

saber716rus, 2024-05-31:
> non-greenfield 2.4ghz 40Mhz is most likely at fault combined with week signal. 
> Please make your killometer long logs into attachments and describe your claim in original post , as it sounds absolutely absurd that sane damaged router permanently.

https://disk.yandex.ru/d/1zt_8Q3_fFi-JQ
https://disk.yandex.ru/d/HFzGwCJLCTqeBw
https://disk.yandex.ru/d/IS36yNrTJfDO6Q

I filled in the logs from the posts on Yandex. disk. There will be a new one now.

saber716rus, 2024-05-31:
https://disk.yandex.ru/d/TUR-o3HeyJIQKA

I made a new log. with this log, the connection was stable.

saber716rus, 2024-05-31:
at 9.02 there was a 5-second client shutdown. I am attaching the log.
https://disk.yandex.ru/d/JQDtnGsGAeckZQ
[image]

brada4, 2024-05-31:
Attach files to this issue and replace pictures with english ones (or better text logs)

saber716rus, 2024-05-31:
> Attach files to this issue and replace pictures with english ones (or better text logs)

the most interesting thing is the failure in the log. i.e. from 7 a.m. to 2 p.m. the repeater was working.

[image]

saber716rus, 2024-05-31:
I'm attaching logs[image]

[image]

[image]

[image]

[image]

[image]

[image]

brada4, 2024-05-31:
What is exact timeline of events logged by openwrt around events in keeneeteex.

saber716rus, 2024-05-31:
> What is exact timeline of events logged by openwrt around events in keeneeteex. 

Internet loss occurs every 15-20 minutes.

brada4, 2024-05-31:
Like in 1h dhcp lease time on your non-openwrt router?

saber716rus, 2024-05-31:
> Like in 1h dhcp lease time on your non-openwrt router?

I have reserved an ip address. I've turned on the repeater now. and I'm dropping the log from now on. https://www.speedtest.net/result/16320289736
[image]
[system log .txt](https://github.com/user-attachments/files/15515482/system.log.txt)

I'm sorry, but for keenetic os I don't know where the language is changing.
[image]

I also noticed that the devices are connected and disconnected from the main router several times.

saber716rus, 2024-05-31:
+  if you are connected to the repeater, I can't get to the luci interface via browser or ssh

saber716rus, 2024-05-31:
[log.txt](https://github.com/user-attachments/files/15515638/log.txt)

this is the log of the main keenetic router

saber716rus, 2024-05-31:
[MWS_LOG__31.05.2024__18_18_39.csv](https://github.com/user-attachments/files/15515668/MWS_LOG__31.05.2024__18_18_39.csv)

the transition log

saber716rus, 2024-05-31:
I disabled FT roaming on keenetic and this is what I get in the repeater log.

[system log with roaming mode turned off on keenatic.txt](https://github.com/user-attachments/files/15515759/system.log.with.roaming.mode.turned.off.on.keenatic.txt)

brada4, 2024-05-31:
wpa3 rekey chokes, set one to wpa2

saber716rus, 2024-05-31:
> wpa3 rekey chokes, set one to wpa2

by the way, as soon as I turned off FT mode on the main router, I can immediately log back into the repeater if I am connected to it.

saber716rus, 2024-05-31:
> wpa3 rekey chokes, set one to wpa2

on the access point or the client?

saber716rus, 2024-05-31:
> wpa3 rekey chokes, set one to wpa2

that's what I got, the problem persists. But when I turned off FT on the main router, the situation became better, but the unstable Internet persists.
[system log wpa3 to wpa2.txt](https://github.com/user-attachments/files/15515968/system.log.wpa3.to.wpa2.txt)

saber716rus, 2024-05-31:
@brada4 today I sent all the collected information to keenetic support.

saber716rus, 2024-05-31:
> @brada4 today I sent all the collected information to keenetic support. they said that it is possible to cache PMKID

brada4, 2024-05-31:
Nice, maybe they are better at guessing timezone offsets in assorted logs.

saber716rus, 2024-05-31:
> Nice, maybe they are better at guessing timezone offsets in assorted logs.

it seems to have gone, but I can't get into the repeater again and the Internet speed does not rise above 8mbps

Incoming connection
7.74 Mbps=967.10 Kbytes/s
Outgoing connection
7.52 Mbps=939.82 Kbytes/s

saber716rus, 2024-05-31:
writes that the page is unavailable
[image]
[image]

brada4, 2024-05-31:
You have to describe a bug and how to repeat it, otherwise forums are for random mishaps during initial setup. 
Does not help posting random assorted stuff across all screens, it is not crowd of  superhuman robots that will serve as your helpdesk.

closed 2024-05-31 by saber716rus