openwrt/openwrt #15596: MR52 dts defines "red:user", "green:user", "blue:user" for the system LED but ls -al /sys/class/leds/ does not show any of them.

vochong, 2024-05-31:
### Describe the bug

MR52 dts defines "red:user", "green:user", "blue:user" for the system LED but 'ls -al /sys/class/leds/' does not show any of them. The problem is seen in 23.05.3 as well as the latest master snapshots.

MR52 dts:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8068-mr52.dts;h=e1c233254d25dddbc8fb6f9cf00fade407180926;hb=67f52012760ce553ba6393aaba77359a458deeda

 118         tlc591xx@49 {
 119                 #address-cells = <1>;
 120                 #size-cells = <0>;
 121                 compatible = "ti,tlc59108";
 122                 reg = <0x49>;
 123 
 124                 red@0 {
 125                         label = "red:user";
 126                         reg = <0x0>;
 127                 };
 128 
 129                 green@1 {
 130                         label = "green:user";
 131                         reg = <0x1>;
 132                 };
 133 
 134                 blue@2 {
 135                         label = "blue:user";
 136                         reg = <0x2>;
 137                 };
root@mr52:~# ls -al /sys/class/leds/
lrwxrwxrwx    1 root     root             0 May 29 09:01 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 May 29 09:01 ath10k-phy1 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 May 29 09:01 ath10k-phy2 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active

For MR42, similar definitions are present in the MR42 dts and they show up properly in ls -la /sys/class/leds/

https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8068-mr42.dts;h=ccf2554ca319fa2b36cd37f4da1859d9e7699c5f;hb=67f52012760ce553ba6393aaba77359a458deeda

root@mr42:~# ls -la /sys/class/leds/
lrwxrwxrwx    1 root     root             0 May 29 00:48 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 May 29 00:48 ath10k-phy1 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 May 29 00:48 ath10k-phy2 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user -> ../../devices/platform/soc/16500000.gsbi/16580000.i2c/i2c-1/1-0040/leds/blue:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user -> ../../devices/platform/soc/16500000.gsbi/16580000.i2c/i2c-1/1-0040/leds/green:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user -> ../../devices/platform/soc/16500000.gsbi/16580000.i2c/i2c-1/1-0040/leds/red:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

ipq806x/generic

### Device

Cisco Meraki MR52

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-05-31:
@robimarko 
@namiltd

robimarko, 2024-05-31:
Please post full bootlog at least

namiltd, 2024-05-31:
```
opkg install i2c-tools
i2cdetect -l
i2cdetect -y 0
```

vochong, 2024-06-01:
@robimarko 

Here's the information you requested.   Thanks a lot for taking a look at this issue.

```
mr52:~# i2cdetect -l
i2c-0	i2c       	QUP I2C adapter                 	I2C adapter

mr52:~# i2cdetect -y 0
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                                                 
20:                                                 
30: -- -- -- -- -- -- -- --                         
40:                                                 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60:                                                 
70:                                                 

mr52:~# i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
```

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.15.155 (Cisco_Meraki_MR52_ath10k@openwrt.lan) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.3.0 r0+23852-5a90369931) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon May 6 16:54:57 2024
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Meraki MR52
[    0.000000] OF: fdt: Ignoring memory range 0x41500000 - 0x44000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000] percpu: Embedded 12 pages/cpu s19532 r8192 d21428 u49152
[    0.000000] pcpu-alloc: s19532 r8192 d21428 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113426
[    0.000000] Kernel command line: loader=u-boot part=bootkernel2 console=ttyMSM0,115200n8 ubi.mtd=ubi ubi.mtd=art
[    0.000000] Bootloader command line not present
[    0.000000] Unknown kernel command line parameters "loader=u-boot part=bootkernel2", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 441932K/457728K available (6761K kernel code, 618K rwdata, 1768K rodata, 1024K init, 240K bss, 15796K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000002] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000024] Switching to timer-based delay loop, resolution 160ns
[    0.000347] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000381] CPU: Testing write buffer coherency: ok
[    0.000443] pid_max: default: 32768 minimum: 301
[    0.001294] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001322] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.003349] qcom_scm: convention: smc legacy
[    0.004697] Setting up static identity map for 0x44300000 - 0x44300060
[    0.004900] rcu: Hierarchical SRCU implementation.
[    0.005520] smp: Bringing up secondary CPUs ...
[    0.007555] smp: Brought up 1 node, 2 CPUs
[    0.007582] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[... 462 lines cut ...]
[   36.650051] device phy2-ap0 entered promiscuous mode
[   37.010604] IPv6: ADDRCONF(NETDEV_CHANGE): phy2-ap0: link becomes ready
[   37.010762] br-lan: port 2(phy2-ap0) entered blocking state
[   37.016094] br-lan: port 2(phy2-ap0) entered forwarding state
[   37.022554] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   43.423846] ath10k_pci 0001:01:00.0: Unknown eventid: 36933
[   43.428456] br-lan: port 3(phy1-ap0) entered blocking state
[   43.428487] br-lan: port 3(phy1-ap0) entered disabled state
[   43.433981] device phy1-ap0 entered promiscuous mode
[   43.786579] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[   43.786731] br-lan: port 3(phy1-ap0) entered blocking state
[   43.791999] br-lan: port 3(phy1-ap0) entered forwarding state
[   44.435444] br-lan: port 3(phy1-ap0) entered disabled state
[   47.456902] phy1-sta0: authenticate with 10:da:43:14:9b:57
[   47.726076] phy1-sta0: send auth to 10:da:43:14:9b:57 (try 1/3)
[   47.734251] phy1-sta0: authenticated
[   47.734917] phy1-sta0: associate with 10:da:43:14:9b:57 (try 1/3)
[   47.749571] phy1-sta0: RX AssocResp from 10:da:43:14:9b:57 (capab=0x11 status=0 aid=4)
[   47.751763] ath10k_pci 0001:01:00.0: pdev param 0 not supported by firmware
[   47.756483] ath10k_pci 0001:01:00.0: failed to enable peer stats info: -95
[   47.763425] phy1-sta0: associated
[   47.885484] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-sta0: link becomes ready
[   47.891416] br-lan: port 3(phy1-ap0) entered blocking state
[   47.891446] br-lan: port 3(phy1-ap0) entered forwarding state
```

namiltd, 2024-06-01:
```
i2cdetect -y -r 0
```

vochong, 2024-06-02:
mr52:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- 4c -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  

mr52:~# i2cget 0 0x4c
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will read from device file /dev/i2c-0, chip address 0x4c, current data
address, using read byte.
Continue? [Y/n] y
0x1e

mr52:~# i2cget 0 0x4c
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will read from device file /dev/i2c-0, chip address 0x4c, current data
address, using read byte.
Continue? [Y/n] y
0x1f

namiltd, 2024-06-02:
It looks like the tlc591xx address was 4c and not 49.

robimarko, 2024-06-02:
Hm, it doesn't have to be like that as UU only means that kernel has reserved that address.

Since TLC59108 doesn't have an ID register it's hard to see for sure whether it's the thing you are talking to.
Can you try unloading the tlc591xx kmod and then loading it, I would love to see the error code it returns.

vochong, 2024-06-02:
I change the tlc591xx's address from 0x49 to 0x4c (2 changes in the MR52 dts), and the green:user, blue:user, red:user show up in /sys/class/led/, so it looks like the 0x4c is the real address of the tlc591xx.   However, I still cannot control the colors using echo 0/1  on the brightness (this works properly on MR42).   It's very likely that some additional settings must also be changed accordingly.

```
AFTER A POWER-CYCLE RUNNING A 6.6.32 IMAGE WITH THE tlc591xx address change.

root@mr52:~# ls -la /sys/class/leds/
lrwxrwxrwx    1 root     root             0 Jun  2 12:26 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 Jun  2 12:26 ath10k-phy1 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 Jun  2 12:26 ath10k-phy2 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active

root@mr52:~# i2cdetect -l
i2c-0	i2c       	QUP I2C adapter                 	I2C adapter

root@mr52:~# i2cdetect -y 0
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                                                 
20:                                                 
30: -- -- -- -- -- -- -- --                         
40:                                                 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60:                                                 
70:                                                 
                                    
            
root@mr52:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- UU -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

root@mr52:~# lsmod | grep tlc591xx
leds_tlc591xx          12288  0 
regmap_i2c             12288  3 ina2xx,at24,leds_tlc591xx

root@mr52:~# rmmod leds_tlc591xx

root@mr52:~# lsmod | grep tlc591xx

root@mr52:~# i2cdetect -y 0
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                                                 
10:                                                 
20:                                                 
30: -- -- -- -- -- -- -- --                         
40:                                                 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60:                                                 
70:

root@mr52:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- 4c -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     

All green:user/blue:user/red:user are now gone.

root@mr52:~# ls -la /sys/class/leds/
lrwxrwxrwx    1 root     root             0 Jun  2 12:26 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 Jun  2 12:26 ath10k-phy1 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 Jun  2 12:26 ath10k-phy2 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active

```

namiltd, 2024-06-02:
Check the remaining subaddresses (reg = <0x3> to reg = <0x7>)

vochong, 2024-06-02:
I don't see the subaddresses you mentioned.  Can you please be more specific.  Thanks.

 124                 red@0 {
 125                         label = "red:user";
 126                         reg = <0x0>;
 127                 };
 128 
 129                 green@1 {
 130                         label = "green:user";
 131                         reg = <0x1>;
 132                 };
 133 
 134                 blue@2 {
 135                         label = "blue:user";
 136                         reg = <0x2>;

namiltd, 2024-06-02:
Lines 126, 131 and 136

vochong, 2024-06-02:
But the color control seems to work fine on MR42 DTS for these reg addresses

```
MR42

        tlc591xx@40 {
                #address-cells = <1>;
                #size-cells = <0>;
                compatible = "ti,tlc59108";
                reg = <0x40>;

                red@0 {
                        label = "red:user";
                        reg = <0x0>;
                };
                
                green@1 {
                        label = "green:user";
                        reg = <0x1>;
                };

                blue@2 {
                        label = "blue:user";
                        reg = <0x2>;
                };

----------------------------------------
MR52:

        tlc591xx@4c {
                #address-cells = <1>;
                #size-cells = <0>;
                compatible = "ti,tlc59108";
                reg = <0x4c>;

                red@0 {
                        label = "red:user";
                        reg = <0x0>;
                };

                green@1 {
                        label = "green:user";
                        reg = <0x1>;
                };

                blue@2 {
                        label = "blue:user";
                        reg = <0x2>;
                };
```

namiltd, 2024-06-02:
The TLC59108 system has 8 pins for connecting LEDs. MR52 has three diodes, so you need to check which pins these are.

vochong, 2024-06-03:
I tried 3 images (each with 0/1/2,   3/4/5,   5/6/7), but echo0/1 to brightness do NOT work for red:user/blue:user/green:user.

For image with all pin settings (per your post above), user2 and user3 do not show up:

```
root@mr52:/sys/class/leds# lsmod | grep tlc
leds_tlc591xx          12288  0 
regmap_i2c             12288  3 ina2xx,at24,leds_tlc591xx

root@mr52:/sys/class/leds# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- UU -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

root@mr52:/sys/class/leds# ls -al | grep user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user

root@mr52:/sys/class/leds# ls -al 
drwxr-xr-x    2 root     root             0 Dec 31  1969 .
drwxr-xr-x   39 root     root             0 Dec 31  1969 ..
lrwxrwxrwx    1 root     root             0 Jun  2 17:38 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 Jun  2 17:38 ath10k-phy1 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 Jun  2 17:38 ath10k-phy2 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active

```

robimarko, 2024-06-03:
Try using 255 for brightness to turn the LED ON since this is a 256 step driver so brightness of 1 should basically not be visible.

vochong, 2024-06-03:
@namiltd 

It looks like the sysupgrade to the image with all pins enabled actually failed, so I tried upgrading again.    After that,  it showed all the DTS-configured settings.  However, echo 255 to any of them did not produce any color.  I also tried to turn off (echo 0) the white color (white:active) before applying echo 255 to the *.user* entries to no avail.

```
root@mr52:/tmp# ls -al /sys/class/leds/ | grep user

lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user2 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user2 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user3 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user3
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user2 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user3 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user3

root@mr52:/tmp# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- UU -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

root@mr52:/tmp# lsmod | grep tlc
leds_tlc591xx          12288  0 
regmap_i2c             12288  3 ina2xx,at24,leds_tlc591xx

```

@robimarko

"echo 255" did not help either.   On MR42,  "echo 1" produces max brightness for each color.

vochong, 2024-06-03:
Strange thing:  with the device running the image with all pins enabled,  I powered it down, waited 5 seconds and powered it back up.   It did not show any user entry.    Note that the address 0x4c was NOT showing UU.

```
root@mr52:~# uptime
 08:47:04 up 2 min,  load average: 0.00, 0.09, 0.06

root@mr52:~# lsmod | grep tlc
leds_tlc591xx          12288  0 
regmap_i2c             12288  3 ina2xx,at24,leds_tlc591xx

root@mr52:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- 4c -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

root@mr52:~# ls -la /sys/class/leds/
drwxr-xr-x    2 root     root             0 Dec 31  1969 .
drwxr-xr-x   33 root     root             0 Dec 31  1969 ..
lrwxrwxrwx    1 root     root             0 Jun  3 08:43 ath10k-phy0 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 Jun  3 08:43 ath10k-phy1 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 Jun  3 08:43 ath10k-phy2 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active
```

vochong, 2024-06-03:
I did another power down/up again (with 30 seconds waiting in between) and all user entries showed up again.  Unfortunately,  no change in user color behavior.

vochong, 2024-06-03:
@robimarko
@namiltd 

Some clue:  It looks like all the tlc591xx register settings are wrong.   The dmesg shows that "Setting an LED's brightness failed (-6)" after echo any value to either of the user entries.

```
root@mr52:~# lsmod | grep tlc
leds_tlc591xx          12288  0 
regmap_i2c             12288  3 ina2xx,at24,leds_tlc591xx

root@mr52:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- UU -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  

root@mr52:~# echo 1 > /sys/class/leds/blue:user/brightness
root@mr52:~# ls -la /sys/class/leds/ | grep user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 blue:user2 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/blue:user2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user2 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:user3 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/green:user3
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user2 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 red:user3 -> ../../devices/platform/soc/16600000.gsbi/16680000.i2c/i2c-0/0-004c/leds/red:user3

root@mr52:~# dmesg | grep brightness
[  164.382652] leds red:user: Setting an LED's brightness failed (-6)
[  427.886876] leds red:user: Setting an LED's brightness failed (-6)
[  429.893735] leds red:user2: Setting an LED's brightness failed (-6)
[  431.900280] leds red:user3: Setting an LED's brightness failed (-6)
[  433.907042] leds green:user: Setting an LED's brightness failed (-6)
[  435.913308] leds green:user2: Setting an LED's brightness failed (-6)
[  437.919853] leds green:user3: Setting an LED's brightness failed (-6)
[  439.927054] leds blue:user: Setting an LED's brightness failed (-6)
[  441.932946] leds blue:user2: Setting an LED's brightness failed (-6)
```

```
root@mr52:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.6.32 (ernguyen@ubuntu24) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 13.3.0 r26497+1-37ff0ea726) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun Jun  2 21:21:39 2024
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Meraki MR52
[    0.000000] OF: fdt: Ignoring memory range 0x41500000 - 0x44000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] OF: reserved mem: 0x40000000..0x40ffffff (16384 KiB) nomap non-reusable nss@40000000
[    0.000000] OF: reserved mem: 0x41000000..0x411fffff (2048 KiB) nomap non-reusable smem@41000000
[    0.000000] OF: reserved mem: 0x41200000..0x414fffff (3072 KiB) nomap non-reusable rsvd@41200000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000044000000-0x000000005fefffff]
[    0.000000] percpu: Embedded 13 pages/cpu s21012 r8192 d24044 u53248
[    0.000000] pcpu-alloc: s21012 r8192 d24044 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Kernel command line: loader=u-boot part=bootkernel2 console=ttyMSM0,115200n8 ubi.mtd=ubi ubi.mtd=art
[    0.000000] Bootloader command line not present
[    0.000000] Unknown kernel command line parameters "loader=u-boot part=bootkernel2", will be passed to user space.
[... 543 lines cut ...]
[  793.489366] leds blue:user: Setting an LED's brightness failed (-6)
[  833.852178] leds blue:user: Setting an LED's brightness failed (-6)
[  903.176600] leds blue:user: Setting an LED's brightness failed (-6)
[  944.806856] leds red:user: Setting an LED's brightness failed (-6)
[  946.812832] leds red:user2: Setting an LED's brightness failed (-6)
[  948.819434] leds red:user3: Setting an LED's brightness failed (-6)
[  950.827669] leds green:user: Setting an LED's brightness failed (-6)
[  952.834839] leds green:user2: Setting an LED's brightness failed (-6)
[  954.840825] leds green:user3: Setting an LED's brightness failed (-6)
[  956.846900] leds blue:user: Setting an LED's brightness failed (-6)
[  958.852758] leds blue:user2: Setting an LED's brightness failed (-6)
root@mr52:~# dmesg | grep user
[... cut ...]
[  429.893735] leds red:user2: Setting an LED's brightness failed (-6)
[  431.900280] leds red:user3: Setting an LED's brightness failed (-6)
[  433.907042] leds green:user: Setting an LED's brightness failed (-6)
[  435.913308] leds green:user2: Setting an LED's brightness failed (-6)
[  437.919853] leds green:user3: Setting an LED's brightness failed (-6)
[  439.927054] leds blue:user: Setting an LED's brightness failed (-6)
[  441.932946] leds blue:user2: Setting an LED's brightness failed (-6)
```

vochong, 2024-06-03:
@rx78gp01

Hope you may have some suggestion.

namiltd, 2024-06-04:
What exact command caused the "leds red:user: Setting an LED's brightness failed (-6)" error.
Try:
```
uci add system led # =cfg058bba
uci set system.@led[-1].sysfs='green:user'
uci set system.@led[-1].trigger='timer'
uci set system.@led[-1].delayon='500'
uci set system.@led[-1].delayoff='500'
uci commit system
```

vochong, 2024-06-04:
Hello,

echo X > /sys/class/leds/green:user/brightness

X = 1 or 255

namiltd, 2024-06-06:
Try:
```
uci add system led
uci set system.@led[-1].sysfs='green:user'
uci set system.@led[-1].trigger='heartbeat'
uci commit system
```

namiltd, 2024-06-09:
I think it would be worth testing whether controlling these diodes works on MT42.

appleeimac, 2024-06-09:
get me a image, and I will.

namiltd, 2024-06-10:
@appleeimac https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ipq806x%2Fgeneric&id=meraki_mr42

appleeimac, 2024-06-10:
This hasn't been committed to snapshot.

vochong, 2024-06-10:
@namiltd

As I clearly mentioned earlier in the bug report, there's no problem with RGB control in MR42.  The problem is only applicable for MR52.

namiltd, 2024-06-10:
MR42 and MR52 have the same LED chip and the same driver. So they should behave the same. The control of individual LEDs should be identical.

appleeimac, 2024-06-10:
Correct, but the MR42 only displays White, everything else for LED config under LuCI (even in snapshot) is blank on the MR42

vochong, 2024-06-11:
@appleeimac

You can file a separate LUCI bug for MR42.   As for RGB control on MR42, it works fine if you use the echo 1 > /sys/class/led/xxx:xxx/brightness method.   

Unfortunately the echo 1 method still does not work for MR52 after we fixed the wrong i2c address of the tlc591xx LED sink driver.

Leo-PL, 2024-10-23:
```
[  164.382652] leds red:user: Setting an LED's brightness failed (-6)
[  427.886876] leds red:user: Setting an LED's brightness failed (-6)
[  429.893735] leds red:user2: Setting an LED's brightness failed (-6)
[  431.900280] leds red:user3: Setting an LED's brightness failed (-6)
[  433.907042] leds green:user: Setting an LED's brightness failed (-6)
[  435.913308] leds green:user2: Setting an LED's brightness failed (-6)
[  437.919853] leds green:user3: Setting an LED's brightness failed (-6)
[  439.927054] leds blue:user: Setting an LED's brightness failed (-6)
[  441.932946] leds blue:user2: Setting an LED's brightness failed (-6)
[  722.439222] leds red:user: Setting an LED's brightness failed (-6)
[  724.445552] leds red:user2: Setting an LED's brightness failed (-6)
[  726.452185] leds red:user3: Setting an LED's brightness failed (-6)
[  728.458138] leds green:user: Setting an LED's brightness failed (-6)
[  730.465153] leds green:user2: Setting an LED's brightness failed (-6)
[  732.471613] leds green:user3: Setting an LED's brightness failed (-6)
[  734.477742] leds blue:user: Setting an LED's brightness failed (-6)
[  736.483664] leds blue:user2: Setting an LED's brightness failed (-6)
[  766.213261] leds red:user: Setting an LED's brightness failed (-6)
[  793.489366] leds blue:user: Setting an LED's brightness failed (-6)
[  833.852178] leds blue:user: Setting an LED's brightness failed (-6)
[  903.176600] leds blue:user: Setting an LED's brightness failed (-6)
[  944.806856] leds red:user: Setting an LED's brightness failed (-6)
[  946.812832] leds red:user2: Setting an LED's brightness failed (-6)
[  948.819434] leds red:user3: Setting an LED's brightness failed (-6)
[  950.827669] leds green:user: Setting an LED's brightness failed (-6)
[  952.834839] leds green:user2: Setting an LED's brightness failed (-6)
[  954.840825] leds green:user3: Setting an LED's brightness failed (-6)
[  956.846900] leds blue:user: Setting an LED's brightness failed (-6)
[  958.852758] leds blue:user2: Setting an LED's brightness failed (-6)
````
That precisely means the I²C transactions for those were NACKed (-6 stands for -ENXIO). Aren't we dealing with damaged hardware here?  The case of damaged MX65 from the forum suggests it's possible, sadly.
I haven't yet looked at the controller driver, but for it, it may be enough for the I²C device to respond to address, not necessary knowing that the chip is correct.
Please double check pin straps setting the address on the LED controller - for me having them set differently on MR42 vs MR52 is quite odd, and I believe @clayface did test that before submitting patches. 
Also: can you control the LEDs manually using `i2c-tools` somehow? Maybe you can take a look at a datasheet to get register map for the chip.
I don't have any of those devices, so that's basically all I can figure out.

rkboni, 2024-11-01:
For what it's worth, here's what my MR52 reports running a fairly recent `23.05-SNAPSHOT`:

```
root@schiphol:~# i2cdetect -l
i2c-0	i2c       	QUP I2C adapter                 	I2C adapter
root@schiphol:~# i2cdetect -y 0
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30: -- -- -- -- -- -- -- --
40:
50: -- -- UU -- -- -- -- -- -- -- 5a -- -- -- -- --
60:
70:
root@schiphol:~# i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
root@schiphol:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- 4c -- -- --
50: -- -- UU -- -- -- -- -- -- -- 5a -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@schiphol:~# i2cget 0 0x4c
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will read from device file /dev/i2c-0, chip address 0x4c, current data
address, using read byte.
Continue? [Y/n]
0x2a
root@schiphol:~# i2cget 0 0x5a
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will read from device file /dev/i2c-0, chip address 0x5a, current data
address, using read byte.
Continue? [Y/n]
0xff
root@schiphol:~# rmmod leds_tlc591xx
root@schiphol:~# lsmod | grep tlc591xx
root@schiphol:~# i2cdetect -y 0
Warning: Can't use SMBus Quick Write command, will skip some addresses
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:
10:
20:
30: -- -- -- -- -- -- -- --
40:
50: -- -- UU -- -- -- -- -- -- -- 5a -- -- -- -- --
60:
70:
root@schiphol:~# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- 4c -- -- --
50: -- -- UU -- -- -- -- -- -- -- 5a -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@schiphol:~# ls -la /sys/class/leds/
drwxr-xr-x    2 root     root             0 Dec 31  1969 .
drwxr-xr-x   38 root     root             0 Dec 31  1969 ..
lrwxrwxrwx    1 root     root             0 Oct 31 21:13 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 Oct 31 21:13 ath10k-phy1 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 Oct 31 21:13 ath10k-phy2 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Dec 31  1969 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Dec 31  1969 white:active -> ../../devices/platform/leds/leds/white:active
root@schiphol:~# ubus call system board
{
	"kernel": "5.15.167",
	"hostname": "schiphol",
	"system": "ARMv7 Processor rev 0 (v7l)",
	"model": "Meraki MR52",
	"board_name": "meraki,mr52",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05-SNAPSHOT",
		"revision": "r24125-607091d4fe",
		"target": "ipq806x/generic",
		"description": "OpenWrt 23.05-SNAPSHOT r24125-607091d4fe"
	}
}
```

vochong, 2024-11-02:
On my three MR52, no i2c device is detected @0x5a at all.  This is really weird.

root@OpenWrt:/sys/class/leds# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- UU -- -- -- -- -- -- 4c -- -- -- 
50: -- -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                       

root@OpenWrt:/sys/class/leds# ls -la
drwxr-xr-x    2 root     root             0 Jan  1  1970 .
drwxr-xr-x   41 root     root             0 Jan  1  1970 ..
lrwxrwxrwx    1 root     root             0 Nov  2 00:14 ath10k-phy0 -> ../../devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/leds/ath10k-phy0
lrwxrwxrwx    1 root     root             0 Nov  2 00:14 ath10k-phy1 -> ../../devices/platform/soc/1b900000.pci/pci0002:00/0002:00:00.0/0002:01:00.0/leds/ath10k-phy1
lrwxrwxrwx    1 root     root             0 Nov  2 00:14 ath10k-phy2 -> ../../devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/leds/ath10k-phy2
lrwxrwxrwx    1 root     root             0 Jan  1  1970 green:lan1 -> ../../devices/platform/leds/leds/green:lan1
lrwxrwxrwx    1 root     root             0 Jan  1  1970 green:lan2 -> ../../devices/platform/leds/leds/green:lan2
lrwxrwxrwx    1 root     root             0 Jan  1  1970 orange:lan1 -> ../../devices/platform/leds/leds/orange:lan1
lrwxrwxrwx    1 root     root             0 Jan  1  1970 orange:lan2 -> ../../devices/platform/leds/leds/orange:lan2
lrwxrwxrwx    1 root     root             0 Jan  1  1970 orange:power -> ../../devices/platform/leds/leds/orange:power
lrwxrwxrwx    1 root     root             0 Jan  1  1970 white:active -> ../../devices/platform/leds/leds/white:active

root@OpenWrt:/sys/class/leds# uname -a
Linux OpenWrt 6.6.58 #0 SMP Sat Nov  2 00:11:41 2024 armv7l GNU/Linux

rkboni, 2024-11-05:
@vochong when you changed the address of the LED controller to `0x4c` did you have any other side effects, or did it boot fine?  Maybe there was something wrong with my build, but when I installed it, the MR52 rebooted with the order of the ethernet interfaces reversed, which meant that it was effectively DOA.

On a related note, I extracted at least the MR42 DTB out of the Meraki image, but the actual device tree look like this:

```
                fdt@1 {
                        description = "MR42 Device Tree";
                        data = [d0 0d fe ed 00 00 14 99 00 00 00 38 00 00 12 d0
00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00 00 00 01 c9 00 00 12 98 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 03 00 0
0 00 04 00 00 00 00 00 00 00 01 00 00 00 03 00 00 00 04 00 00 00 0f 00 00 00 01
00 00 00 03 00 00 00 14 00 00 00 1b 4d 65 72 61 6b 69 20 4d 52 34 32 20 28 59 6f
 77 69 65 29 00 00 00 00 03 00 00 00 29 00 00 00 21 6d 65 72 61 6b 69 2c 79 6f 7
7 69 65 00 6d 65 72 61 6b 69 2c 63 72 79 70 74 69 64 00 71 63 6f 6d 2c 69 70 71
38 30 36 78 00 00 00 00 00 00 00 01 63 68 6f 73 65 6e 00 00 00 00 00 02 00 00 00
 01 61 6c 69 61 73 65 73 00 00 00 00 02 00 00 00 01 6d 65 6d 6f 72 79 00 00 00 0 ...
```

is there any tooling to extract these data declarations into files I can further parse?

vochong, 2024-11-05:
@rkboni 

If you read my posts above, you can see that I have no problem with images with the LED controller address set to 0x4c.  MR52 booted normally and I could see all the 8 labels "colorX:userX" in /sys/class/leds/ .     However, echo 1 > colorX:userX" (for each of the 8 settings) does not change any color.     I did not have any boot issue with custom-built images in either 23.05.x (mentioned in my previous posts) or in the most recent master snapshots (6.6.58).

I still don't understand why none of my three MR52 devices can see any i2c device @0x5a like your MR52.

```
	tlc591xx@4c {
		#address-cells = <1>;
		#size-cells = <0>;
		compatible = "ti,tlc59108";
		reg = <0x4c>;

                red@0 {
                        label = "red:user0";
                        reg = <0x0>;
                };
                
                green@1 {
                        label = "green:user1";
                        reg = <0x1>;
                };
                 
		blue@2 {
                        label = "blue:user2";
                        reg = <0x2>;
                };

                red@3 {
                        label = "red:user3";
                        reg = <0x3>;
                };
               
                green@4 {
                        label = "green:user4";
                        reg = <0x4>;
                };

                blue@5 {
                        label = "blue:user5";
                        reg = <0x5>;
                };

                red@6 {
                        label = "red:user6";
                        reg = <0x6>;
                };
                
                green@7 {
                        label = "green:user7";
                        reg = <0x7>;

		};
	};
```

rkboni, 2024-11-05:
@vochong I got the `data` blobs out of the Meraki DTB extracted and decoded and the LED controller register is *definitely not* at 0x4c:

```
                tricolor@49 {
                        #address-cells = <0x01>;
                        #size-cells = <0x00>;
                        compatible = "ti,tlc59108";
                        reg = <0x49>;

                        red@0 {
                                label = "tricolor0:red";
                                reg = <0x00>;
                                max-poe-val = <0xef>;
                                max-dc-val = <0xff>;
                        };

                        green@1 {
                                label = "tricolor0:green";
                                reg = <0x01>;
                                max-poe-val = <0xff>;
                                max-dc-val = <0xff>;
                        };

                        blue@2 {
                                label = "tricolor0:blue";
                                reg = <0x02>;
                                max-poe-val = <0xff>;
                                max-dc-val = <0xff>;
                        };
                };
```

so I still think we're barking up the wrong tree.

Re: the odd behavior of trying to move the register to `0x4c`, I'm just reporting what happened to my system... there's definitely a lot of entropy still in the system, because I think I did get the ethernet interfaces back in the right order after powering off for 30s+ again... but it was late and I was super annoyed that my primary living-area AP was now down, so that might be a less-than-ideally-recorded data point.

rkboni, 2024-11-05:
But back to the relevant points:

1. The OEM FDT has no devices at either `0x4c` or `0x5a`.
2. The registers for the RGB LEDs.
3. As pointed out, the differences between the MR42 and MR52 are greater than just "added an extra ethernet interface":

```
--- mr42.dts    2024-11-05 11:48:59
+++ mr52.dts    2024-11-05 11:52:46
@@ -3,8 +3,8 @@
 / {
        #address-cells = <0x01>;
        #size-cells = <0x01>;
-       model = "Meraki MR42 (Yowie)";
-       compatible = "meraki,yowie", "meraki,cryptid", "qcom,ipq806x";
+       model = "Meraki MR52 (Bigfoot)";
+       compatible = "meraki,bigfoot", "meraki,cryptid", "qcom,ipq806x";

        chosen {
        };
@@ -35,8 +35,8 @@

                pin@2 {
                        pin = <0x02>;
-                       active = <0x00 0x00 0x03 0x01>;
-                       suspended = <0x00 0x00 0x03 0x01>;
+                       active = <0x00 0x03 0x00 0x00>;
+                       suspended = <0x00 0x03 0x00 0x00>;
                };

                pin@6 {
@@ -45,60 +45,102 @@
                        suspended = <0x00 0x00 0x01 0x02>;
                };

+               pin@7 {
+                       pin = <0x07>;
+                       active = <0x00 0x00 0x01 0x02>;
+                       suspended = <0x00 0x00 0x01 0x02>;
+               };
+
                pin@8 {
                        pin = <0x08>;
-                       active = <0x00 0x00 0x00 0x00>;
-                       suspended = <0x00 0x00 0x00 0x00>;
+                       active = <0x00 0x00 0x03 0x00>;
+                       suspended = <0x00 0x00 0x03 0x00>;
                };

-               pin@15 {
-                       pin = <0x0f>;
-                       active = <0x00 0x00 0x00 0x00>;
-                       suspended = <0x00 0x00 0x00 0x00>;
+               pin@9 {
+                       pin = <0x09>;
+                       active = <0x00 0x00 0x03 0x00>;
+                       suspended = <0x00 0x00 0x03 0x00>;
                };

-               pin@16 {
-                       pin = <0x10>;
-                       active = <0x00 0x00 0x00 0x00>;
-                       suspended = <0x00 0x00 0x00 0x00>;
+               pin@12 {
+                       pin = <0x0c>;
+                       active = <0x01 0x00 0x03 0x00>;
+                       suspended = <0x01 0x00 0x03 0x00>;
                };

-               pin@24 {
[... 300 lines cut ...]
+               reg = <0x02>;
+       };
+
        ble {
                compatible = "ti,cc2540";
-               gpios = <0x01 0x08 0x00 0x01 0x3d 0x00 0x01 0x3e 0x00>;
+               gpios = <0x01 0x15 0x01 0x01 0x11 0x00 0x01 0x0f 0x00>;
        };

        board-data {
@@ -337,11 +436,6 @@
                };

                partition@13 {
-                       label = "panic";
-                       reg = <0x7f80000 0x40000>;
-               };
-
-               partition@14 {
                        label = "all";
                        reg = <0x00 0x8000000>;
                };
```

It does look like this largely matches the DTS's in OpenWRT as created by @clayface -- I haven't verified exhaustively though.

vochong, 2024-11-05:
Clayface's original DTS:

MR42    (RGB is working)
```
        tlc591xx@49 {
                #address-cells = <1>;
                #size-cells = <0>;
                compatible = "ti,tlc59108";
                reg = <0x49>;

                red@0 {
                        label = "red:user";
                        reg = <0x0>;
                };

                green@1 {
                        label = "green:user";
                        reg = <0x1>;
                };

                blue@2 {
                        label = "blue:user";
                        reg = <0x2>;
                };
        };
```

MR52    (no "color:user" label shows up in /sys/class/led)
```
        tlc591xx@49 {
                #address-cells = <1>;
                #size-cells = <0>;
                compatible = "ti,tlc59108";
                reg = <0x49>;

                red@0 {
                        label = "red:user";
                        reg = <0x0>;
                };

                green@1 {
                        label = "green:user";
                        reg = <0x1>;
                };

                blue@2 {
                        label = "blue:user";
                        reg = <0x2>;
                };
        };

```

rkboni, 2024-11-05:
Those are both the same -- in actuality MR42 has `tlc591xx@40` vs MR52's `tlc591xx@49`.

rkboni, 2024-11-05:
> Oops sorry bad copy/paste for MR42. I have fixed it.

Ok, but I think I'm still missing your point... what's that supposed to tell us?

vochong, 2024-11-06:
If you believe that Clayface’s original DTS for the MR52 is correct for the I2C address 0x49 of the TLC591xx (as he likely obtained this information from extracting or decoding the Meraki DTB for the MR52), then why is no ‘color:user’ label created under /sys/class/led? The labels do appear (though non-functional) when the address is changed to 0x4C.

rkboni, 2024-11-06:
> If you believe that Clayface’s original DTS for the MR52 is correct for the I2C address 0x49 of the TLC591xx (as he likely obtained this information from extracting or decoding the Meraki DTB for [...]

I confirmed the DTS by extracting and decoding the Meraki DTB (that's what I quoted above), but that doesn't mean I have any clue why it doesn't work... I just don't think changing the address is going to do anything for us when we know it should be correct... as shown by your attempts to change it and the writes all failing.

vochong, 2024-11-07:
Any idea why i2cdetect could not see any device @0x49 in MR52?

rkboni, 2024-11-07:
> @rkboni Any idea why i2cdetect could not see any device @0x49 in MR52?

Nope :/. Someone should debug-log the heck out of the `tlc591xx` driver -- I was wondering if the `i2c` pin setup could be wrong, but given that the EEPROM is also on that bus and we're reading it fine, that seems to also say that isn't the issue.

vochong, 2024-11-08:
Do we know which device may be @0x4c ?

namiltd, 2024-11-08:
mr42:
```
	i2c0_pins: i2c0_pins {
		mux {
			pins = "gpio24", "gpio25";
			function = "gsbi2";
			drive-strength = <2>;
			bias-pull-up;
			input;
		};
	};
	i2c1_pins: i2c1_pins {
		mux {
			pins = "gpio29", "gpio30";
			function = "gsbi6";
			drive-strength = <2>;
			bias-pull-up;
			input;
		};
	};
```
mr52:
```
	i2c_pins: i2c_pins {
		mux {
			pins = "gpio8", "gpio9";
			function = "gsbi7";
			drive-strength = <2>;
			bias-pull-up;
			input;
		};
	};
```
What if mr52 like mr42 has 2xi2c and the missing Led control is on this missing i2c?

rkboni, 2024-11-08:
thanks for the suggestion @namiltd -- and MR52 does have 2 I2C's but the LED controller is still on the first one.

Attached is the extracted OEM DTS: 
[mr52.dts.txt](https://github.com/user-attachments/files/17680715/mr52.dts.txt)

rkboni, 2024-12-31:
> If you believe that Clayface’s original DTS for the MR52 is correct for the I2C address 0x49 of the TLC591xx (as he likely obtained this information from extracting or decoding the Meraki DTB for [...]

I haven't had a lot of time to look at this recently, but I did install 22.03.0 (AFAIR the earliest release that supported the MR42/MR52) and the LEDs don't work there either, so this is unlikely to be a "broken by upgrades around it" issue and probably a day 0 issue.  I also did add some debugging to the TLC591xx driver and the probe of that driver fails at https://github.com/torvalds/linux/blob/4bbf9020becbfd8fc2c3da790855b7042fad455b/drivers/leds/leds-tlc591xx.c#L181 with a return of `-ENXIO`, which I'm guessing comes from https://github.com/torvalds/linux/blob/ccb98ccef0e543c2bd4ef1a72270461957f3d8d0/drivers/i2c/busses/i2c-qup.c#L814 as a result of the transaction being NACK'ed by the controller?

I think it's worth adding the config for the 2nd I2C controller as suggested by @namiltd in https://github.com/openwrt/openwrt/issues/15596#issuecomment-2463940810 and at least seeing what appears on that bus.  I'll see if I have some time to try that now that I've got a local build environment set up for easier round trips to the device.

rkboni, 2025-01-06:
> I think it's worth adding the config for the 2nd I2C controller as suggested by @namiltd in [#15596 (comment)](https://github.com/openwrt/openwrt/issues/15596#issuecomment-2463940810) and at least seeing what appears on that bus. I'll see if I have some time to try that now that I've got a local build environment set up for easier round trips to the device.

The DTS in that comment doesn't boot, my guess is that renaming the `gsbi` instances causes issues, but I'll need to disassemble my MR52 and attach a serial connection to it to see... right now it looks bricked (white LED on, no network LEDs and AFAICT no network traffic in/out, but the switch does have a link).

rkboni, 2025-01-09:
BTW, one other thing I noticed is that while MR52 doesn't have the TLC leds available, it looks like MR42 doesn't have the LAN leds available (see this issue's original description where @vochong lists the registered LEDs on the MR42).  I'm wondering if come GPIO config got crossed between the devices.

Note also that the I2C config of the OEM kernel includes the following interesting attributes in the DTS:
```
                i2c-gpio,try-slave-recovery;
                i2c-gpio,delay-us = <0x0a>;
```

If I were a betting man, I'd bet the delay could have something to do with the timing / delays of the i2c transactions, but I know very little about i2c, the linux GPIO system, etc.