openwrt/openwrt #15606: FWD: github CI workflow question

CHKDSK88, 2024-05-31:
### Describe the bug

Tim Harvey on ML ask couple days ago about CI in forks. I have the same issue and I'm very interested in solution. Maybe GitHub is better place to ask instead ML. So I quoted original message here:

Tim Harvey (@Gateworks ) on ML [[0]](https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg66255.html):  
> Greetings,
> 
> I have a fork of https://github.com/openwrt/openwrt on github
> (https://github.com/Gateworks/openwrt_upstream) that I use to create
> feature branches to submit pull requests from for upstream OpenWrt.
> 
> I notice that if I enable actions on my fork I get build failures for
> a branch that builds just fine upstream once I issue a PR upstream. Is
> there some reason why the github workflows would be different in my
> fork? I would like to be able to prove I have a successful build in my
> downstream fork 'before' I issue a pull request.
> 
> Best Regards,
> 
> Tim

[0] https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg66255.html

### OpenWrt version

master

### OpenWrt release

snapshot

### OpenWrt target/subtarget

all subtargets

### Device

all

### Image kind

Self-built image

### Steps to reproduce

1. Allow to do GitHub Actions CI tests in repository fork and it allways fail.
2. Reproduce it main tree (doing  PR checks) and it will be ok.

### Expected behaviour

Actions in fork repositories act the same as main repo.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-05-31:
Sync fork first. Your branch is out-of-date

Gateworks, 2024-05-31:
That was the first thing I tried to do but perhaps I don't quite
understand what I'm supposed to sync and how.

The branch I was working on which I wanted CI to build for me before I
issued a pull request was syncd with main and my patches rebased on
top of that. I also went into Github's web UI and selected Sync fork
-> Update branch on the main branch and that didn't seem to help.

CHKDSK88, 2024-06-12:
I prepared simple test. I changed only comments, but CI fail:
https://github.com/CHKDSK88/openwrt-1/actions/runs/9478503040
https://github.com/CHKDSK88/openwrt-1/actions/runs/9478503035

My `master` and `main` branches are up to date, so it isn't an issue.

@Ansuel @ynezz 
Any idea why it is broken?

CHKDSK88, 2024-07-26:
I tried changing name from `https://github.com/CHKDSK88/openwrt-1` to `https://github.com/CHKDSK88/openwrt` but it won't help. This is the only change from other github forks.