openwrt/openwrt #15618: resizing root partition on raspberry pi 5 causes failure to boot

robot-ranger, 2024-06-01:
### Describe the bug

After flashing the snapshot to SD card, openwrt boots fine with 104Mb root partition.

All attempts to resize the partition (incl gparted on another machine) result in the pi not booting or not booting completely. It does not produce a DHCP address on eth0, cannot access ssh.

### OpenWrt version

r26493-dbc9596c7e

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

bcm27xx/bcm2712

### Device

Raspberry Pi 5 Model B Rev 1.0

### Image kind

Official downloaded image

### Steps to reproduce

1. flash SNAPSHOT (r26493-dbc9596c7e) to SD card
2. insert SD card into pi and power on
3. run script to resize root https://openwrt.org/docs/guide-user/advanced/expand_root#automated
4. next boot, nothing works, doesnt fully boot

### Actual behaviour

well... nothing. but thats the problem

### Expected behaviour

expected full boot w/ 100% root partition

### Diffconfig

```text
not self built
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-06-02:
Have you properly followed instructions with `losetup`?
https://openwrt.org/docs/guide-user/advanced/expand_root

robot-ranger, 2024-06-02:
> Have you properly followed instructions with `losetup`?
> https://openwrt.org/docs/guide-user/advanced/expand_root

Yes, many times. It runs and reboots (I assume the first time after `parted`), but fails to actually boot

brada4, 2024-06-02:
The forum post https://forum.openwrt.org/t/expanding-root-soft-bricks-raspberry-pi-5/199784 shows individual resize2fs commands on your part. Please do full script piece by piece manually in failsafe mode boot.  Kind of "works for me".

robot-ranger, 2024-06-02:
> The forum post https://forum.openwrt.org/t/expanding-root-soft-bricks-raspberry-pi-5/199784 shows individual resize2fs commands on your part. Please do full script piece by piece manually in failsafe mode boot.  Kind of "works for me".

That's my forum post - I'm robot-ranger. Ksofen's instructions do work.

... But The issue here is the automated script you linked doesn't work on raspi5, where as it does work on raspi4, cm4 and nano pi r4s.

brada4, 2024-06-02:
I got that part about names ;-)
This is wiki issue, down in forums. Instruction works perfecly on x86.