openwrt/openwrt #15619: ipq40xx: `auc` upgrade - SNAPSHOT not possible

GoetzGoerisch, 2024-06-02:
### Describe the bug

Follow-up to #15528 

attended-sysupgrade from command line using `auc` is not possible on `ipq40xx`.

```
Requesting build.................Error: Received inncorrect version r26497-37ff0ea726 (requested r26492-af329ec389)
Bad message (74)
```

### OpenWrt version

r26445-4f078bf377

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/chromium

### Device

Google WiFi (Gale)

### Image kind

Official downloaded image

### Steps to reproduce

- login to device
- auc -n

### Actual behaviour

```
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r26445-4f078bf377 on ipq40xx/chromium (google,wifi)
Requesting package lists...
 kmod-crypto-gf128: 6.6.30-r1 -> 6.6.32-r1
 kmod-usb-core: 6.6.30-r1 -> 6.6.32-r1
 luci-app-opkg: 24.148.43850~2d2e5c0 -> 24.148.86114~fbbfdb4
 kmod-google-firmware: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-manager: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-ctr: 6.6.30-r1 -> 6.6.32-r1
 kmod-lib-zlib-deflate: 6.6.30-r1 -> 6.6.32-r1
 kmod-nft-fib: 6.6.30-r1 -> 6.6.32-r1
 kmod-nfnetlink: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-hash: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-reject6: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-aead: 6.6.30-r1 -> 6.6.32-r1
 kmod-usb-dwc3: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-flow: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-reject: 6.6.30-r1 -> 6.6.32-r1
 base-files: 1596~4f078bf377 -> 1596~37ff0ea726
 kmod-lib-crc16: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-nat: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-crc32c: 6.6.30-r1 -> 6.6.32-r1
 kmod-ramoops: 6.6.30-r1 -> 6.6.32-r1
 kmod-usb-dwc3-qcom: 6.6.30-r1 -> 6.6.32-r1
 uboot-envtools: 2024.01-r2 -> 2024.01-r3
 kmod-usb-xhci-hcd: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-hmac: 6.6.30-r1 -> 6.6.32-r1
 kmod-lib-crc32c: 6.6.30-r1 -> 6.6.32-r1
 kmod-reed-solomon: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-sha512: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-seqiv: 6.6.30-r1 -> 6.6.32-r1
 kmod-nft-nat: 6.6.30-r1 -> 6.6.32-r1
 kmod-usb3: 6.6.30-r1 -> 6.6.32-r1
 luci-app-firewall: 24.122.52360~fa2aeb7 -> 24.148.86114~fbbfdb4
 kmod-crypto-ghash: 6.6.30-r1 -> 6.6.32-r1
 kmod-lib-zlib-inflate: 6.6.30-r1 -> 6.6.32-r1
 kmod-fs-ext4: 6.6.30-r1 -> 6.6.32-r1
 kmod-hwmon-core: 6.6.30-r1 -> 6.6.32-r1
 ath10k-board-qca4019: 20240220-r1 -> 20240513-r1
 kmod-pstore: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-ccm: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-log: 6.6.30-r1 -> 6.6.32-r1
 luci-proto-ppp: 24.134.38067~b57c6ff -> 24.148.86114~fbbfdb4
 kmod-leds-gpio: 6.6.30-r1 -> 6.6.32-r1
 kmod-gpio-button-hotplug: 6.6.30-r3 -> 6.6.32-r3
 kmod-nf-log6: 6.6.30-r1 -> 6.6.32-r1
 kmod-mac80211: 6.6.30.6.6.15-r1 -> 6.6.32.6.6.15-r1
 luci-proto-ipv6: 24.142.26676~216cbb0 -> 24.148.86114~fbbfdb4
 luci-app-attendedsysupgrade: 24.079.58964~7943616 -> 24.148.86114~d174bb8
 kmod-crypto-geniv: 6.6.30-r1 -> 6.6.32-r1
 kmod-nft-offload: 6.6.30-r1 -> 6.6.32-r1
 kmod-fs-f2fs: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-null: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-conntrack: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-gcm: 6.6.30-r1 -> 6.6.32-r1
 kmod-ath10k-ct: 6.6.30.2023.06.05~fadd0768-r1 -> 6.6.32.2023.06.05~fadd0768-r1
 kmod-crypto-cmac: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-rng: 6.6.30-r1 -> 6.6.32-r1
 kmod-nf-conntrack6: 6.6.30-r1 -> 6.6.32-r1
 kmod-ath: 6.6.30.6.6.15-r1 -> 6.6.32.6.6.15-r1
 kmod-crypto-crc32: 6.6.30-r1 -> 6.6.32-r1
 luci-mod-network: 24.121.67809~2a96307 -> 24.149.31337~7226e13
 kmod-nft-core: 6.6.30-r1 -> 6.6.32-r1
 kmod-nls-base: 6.6.30-r1 -> 6.6.32-r1
 kmod-crypto-sha3: 6.6.30-r1 -> 6.6.32-r1
 kmod-cfg80211: 6.6.30.6.6.15-r1 -> 6.6.32.6.6.15-r1
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.................Error: Received inncorrect version r26497-37ff0ea726 (requested r26492-af329ec389)
```

### Expected behaviour

build, download and install the latest SNAPSHOT

### Additional info

Where does the string `Error: Received inncorrect version` come from?
Would like to help fixing this spelling error, but did not find it.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-06-02:
Just try again later, you caught sysupgrade server mid-downloading next snapshot.

GoetzGoerisch, 2024-06-03:
Thanks @brada4 , where do one check for ths sysupgrade server status?

auc now reports:
`Requesting build..........Error: Received inncorrect version r26504-d4acd05218 (requested r26498-8647ce0800)
Bad message (74)`

Through luci it works:

```
Version	SNAPSHOT r26504-d4acd05218
SHA256	9be2fbacf57f6856395f4c00e1e578962b1cd0ae7ba585a74b2526f0491243cc
Profile	google_wifi
Target	ipq40xx/chromium
Build Date	2024-06-02T20:43:46.000000Z
Filename	openwrt-79bdd5f8c306-ipq40xx-chromium-google_wifi-squashfs-sysupgrade.bin
Filesystem	squashfs
```

brada4, 2024-06-03:
I think just this:
https://sysupgrade.openwrt.org/stats

GoetzGoerisch, 2024-06-03:
Thanks.

Behaviour is similar on other targets and devices.

`luci-attendedsysupgrade` works, as per #15528 
`auc` does not work.

dangowrt, 2024-06-05:
@aparcar Please take a look why the asu server doesn't follow available built versions any more.

pko76, 2024-06-08:
I have been facing the same problem since Thursday (two days).

auc -b 23.05 -B 23.05-SNAPSHOT -y

auc/0.3.2-1
Server: https://sysupgrade.openwrt.org
Running: 23.05.3 r23809-234f1a2efa on ipq40xx/generic (zte,mf286d)
Available: 23.05-SNAPSHOT r23896-6698e6e76a
Requesting package lists...

Error: Received inncorrect version r23897-3287d82fcf (requested r23896-6698e6e76a)
Bad message (74)

pko76, 2024-06-09:
> @aparcar Please take a look why the asu server doesn't follow available built versions any more.

Neither auc nor Luci attended sysupgrade works perfectly OK. To upgrade 23.05-snapshot using Luci I have to use Advance Mode, remove or add some packages and then I am sometimes able to get the newest version than is detected. There are cases when the software package update indicates a new base file, the attended upgrade does not recognize it. Adding or removing packages allows to get the newest version in spite of displaying that the newest is already installed. It needs some more development IMHO.

aparcar, 2024-06-10:
Please try again. I'm afraid this was an issue with the CDN which would sometimes still show old version information

pko76, 2024-06-10:
root@OpenWrt:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05-SNAPSHOT r23898-d0620706f0 on ipq40xx/generic (zte,mf286d)
Available: 23.05-SNAPSHOT r23898-d0620706f0
Requesting package lists...
 base-files: 1555-r23898-d0620706f0 -> 1555-r23899-3001d3c5d4
Are you sure you want to continue the upgrade process? [N/y] y
Requesting build.................................................................................
Downloading image from https://sysupgrade.openwrt.org/store/f43f42a935ded9fb0449b70def1c7202/openwrt-23.05-snapshot-r23898-d0620706f0-4585010d2a33-ipq40xx-generic-zte_mf286d-squashfs-sysupgrade.bin
Writing to 'openwrt-23.05-snapshot-r23898-d0620706f0-4585010d2a33-ipq40xx-generic-zte_mf286d-squashfs-sysupgrade.bin'
ubiattach: error!: cannot attach mtd9
           error 17 (File exists)
invoking sysupgrade

GoetzGoerisch, 2024-06-28:
solved with [owut](https://github.com/efahl/owut/), the replacement for `auc` going forward.

Thank you @efahl ! ❤️

closed 2024-06-28 by GoetzGoerisch

efahl, 2024-06-28:
> solved with [owut](https://github.com/efahl/owut/), the replacement for `auc` going forward.

Actually, I think this was resolved by @aparcar 's switch to webhooks for the ASU server updates.  In the week that I've been watching closely, the version differences between the downloads and sysupgrade sites are only present for 20-30 minutes, updating is very fast.

GoetzGoerisch, 2024-06-29:
Than also thank you to @aparcar. 👍🏻 

For reference, which switch to webhooks do you mean?

efahl, 2024-06-29:
> For reference, which switch to webhooks do you mean?

This commit on the ASU server: https://github.com/openwrt/asu/commit/1e6484d8ed95ffd697779bc1ecba68e282bb7203

Yes, it definitely deserves a 👍🏼 as it seems to have resolved a lot of the auc/FS issues.