openwrt/openwrt #15620: imagebuilder for x86/64 not creating image successfully

w1zz4, 2024-06-02:
### Describe the bug

x86/64 imagebuilder broken, probably related to passage to 6.6 Kernel made in commit 15370.
kmod-dwmac-intel is default but this module is not in current kmods snapshot repo.

### OpenWrt version

 r26186-064d4a8083

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

Try to build latest x86/64 snapshot using imagebuilder

### Actual behaviour

Collected errors:
 * opkg_install_cmd: Cannot install package kmod-dwmac-intel.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:151: _call_image] Error 2
make: *** [Makefile:310: image] Error 2

### Expected behaviour

make is able to dl package and compile it correctly

### Diffconfig

```text
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_x86_64_DEVICE_generic=y
CONFIG_TARGET_DEVICE_PACKAGES_x86_64_DEVICE_generic=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2024-06-03:
The issue is with `dwmac-intel` dependencies: It depends on `stmmac-core`, which has a dependency for Linux 6.6 on `of-mdio` (which is not built for x86). This removes `dwmac-intel` from the build config.
https://github.com/openwrt/openwrt/blob/d4acd052181a71d7486430bd4322dec879b99847/package/kernel/linux/modules/netdevices.mk#L1607-L1610
https://github.com/openwrt/openwrt/blob/d4acd052181a71d7486430bd4322dec879b99847/package/kernel/linux/modules/netdevices.mk#L1252-L1255

Maybe try (untested):
```patch
diff --git a/package/kernel/linux/modules/netdevices.mk b/package/kernel/linux/modules/netdevices.mk
index cf253ff9e05e..1c7df18e3020 100644
--- a/package/kernel/linux/modules/netdevices.mk
+++ b/package/kernel/linux/modules/netdevices.mk
@@ -1607,7 +1607,7 @@ $(eval $(call KernelPackage,pcs-xpcs))
 define KernelPackage/stmmac-core
   SUBMENU:=$(NETWORK_DEVICES_MENU)
   TITLE:=Synopsis Ethernet Controller core (NXP,STMMicro,others)
-  DEPENDS:=@TARGET_x86_64||TARGET_armsr_armv8 +kmod-pcs-xpcs +LINUX_6_6:kmod-of-mdio +kmod-ptp
+  DEPENDS:=@TARGET_x86_64||TARGET_armsr_armv8 +kmod-pcs-xpcs +(LINUX_6_6&&!TARGET_x86_64):kmod-of-mdio +kmod-ptp
   KCONFIG:=CONFIG_STMMAC_ETH \
     CONFIG_STMMAC_SELFTESTS=n \
     CONFIG_STMMAC_PLATFORM \

```

namiltd, 2024-06-03:
@TARGET_x86_64 contradicts !TARGET_x86_64

john-tho, 2024-06-10:
Should be fixed by https://github.com/openwrt/openwrt/commit/fe5fe3c8e708913afc9c1e50bf7347ef1304353c

pgera, 2024-06-11:
Still failing on that commit. The image builder's snapshot is at that commit right now.
```
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.32~703e64376a4102c0534abd39375f9f84-r1) for kmod-stmmac-core
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-stmmac-core found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.32~703e64376a4102c0534abd39375f9f84-r1) for kmod-dwmac-intel
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dwmac-intel found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-dwmac-intel.
 * opkg_install_cmd: Cannot install package luci-app-wireguard.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-dwmac-intel:
 * 	kernel (= 6.6.32~703e64376a4102c0534abd39375f9f84-r1)
 * opkg_install_cmd: Cannot install package kmod-dwmac-intel.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:161: _call_manifest] Error 2
make: *** [Makefile:322: manifest] Error 2
```

ynezz, 2024-06-11:
> Still failing on that commit. The image builder's snapshot is at that commit right now.

I'm not able to reproduce using following:

```
04228e4d3efa43d1fb192e7a7ba6742a95c8cf5af5d3e9b2993de1b15eb067d1  openwrt-imagebuilder-x86-64.Linux-x86_64.tar.zst
```

and simple `make image`. Can you provide full reproducer?

pgera, 2024-06-11:
I haven't set up a local build environment. I was using the public openwrt image builder (https://firmware-selector.openwrt.org/) and selected snapshot for x86_64 and gave some custom packages to build a custom image. The packages I added shouldn't matter, but they were:

`usbutils ethtool kmod-usb-net-rtl8152 luci luci-ssl ca-certificates pbr luci-app-pbr sqm-scripts stubby openconnect luci-proto-openconnect luci-app-sqm luci-app-wireguard ddns-scripts-cloudflare luci-app-ddns luci-proto-wireguard drill bash curl vim-fuller htop luci-app-uhttpd acme-acmesh-dnsapi luci-app-acme libuhttpd-openssl`

robimarko, 2024-06-11:
Locally using the latest snapshot imagebuilder for x86_64 works just fine, so ASU is probably just behind on the IB it uses

ynezz, 2024-06-11:
> Locally using the latest snapshot imagebuilder for x86_64 works just fine, so ASU is probably just behind on the IB it uses

Yep, seems so. Thanks a lot for double checking! @aparcar FYI

ynezz, 2024-06-11:
> I haven't set up a local build environment. I was using the public openwrt image builder (https://firmware-selector.openwrt.org/) and selected snapshot for x86_64 and gave some custom packages to [...]

Trying this package set, I'm not able to build the image due to missing  `luci-app-wireguard` package:

```
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-wireguard.
make[2]: *** [Makefile:220: package_install] Error 255
make[1]: *** [Makefile:151: _call_image] Error 2
make: *** [Makefile:310: image] Error 2
```

and indeed, this package is missing in https://downloads.openwrt.org/snapshots/packages/x86_64/luci/ so not ImageBuilder's issue.

w1zz4, 2024-06-11:
luck-app-wireguard was deprecated few months ago and is now included in luci-proto-wireguard

I guess his hasn’t been working for a while…

ynezz, 2024-06-11:
> luck-app-wireguard was deprecated few months ago and is now included in luci-proto-wireguard
> 
> I guess his hasn’t been working for a while…

Yeah, thanks for the hint, so seems we're good.

lorand-horvath, 2024-06-12:
@ynezz Not sure if this is related, but I haven't been able to create a custom build with https://firmware-selector.openwrt.org for a couple of days now. When I customize the packages and click on "request build", it returns an older snapshot version 26580 from days ago, no matter what I do. I think the custom image builder is broken for some reason.

aparcar, 2024-06-12:
I can reproduce the issue regarding outdated snapshots. I'm investigating what causes the issue. Sorry for the inconvenience

lorand-horvath, 2024-06-12:
@aparcar I've noticed your fix https://github.com/openwrt/openwrt/commit/213bd273edade89e2da471083eb9329c0e9c62f2 and the firmware-selector was unstuck for a while at 26593, but now if I try to request a new custom build, which I expect to be 26608, it's stuck again at 26593.
Am I trying too soon, or has the issue returned?

lorand-horvath, 2024-06-12:
@aparcar Just did another test for https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=ath79%2Fgeneric&id=tplink_tl-wr1043nd-v2
and the default snapshot is 26608, but when I customized it and requested the custom build, it returned the older 26593.
There's definitely something wrong going on with the custom builds.

aparcar, 2024-06-12:
Honestly I'm fairly confused. All components use direct downloads so it can't be a CDN issue. I'll investigate further...

lorand-horvath, 2024-06-12:
@aparcar Probably this doesn't work quite as intended: https://github.com/openwrt/asu/commit/1e6484d8ed95ffd697779bc1ecba68e282bb7203

lorand-horvath, 2024-06-13:
Now the default snapshots are also stuck in the past. Should be 26609, but all of them rebuilt as 26608 - very strange.

lorand-horvath, 2024-06-13:
@aparcar Now the default snapshots are also stuck in the past. Should be 26609, but all of them rebuilt as 26608 - very strange.
Custom snapshots are even more outdated at 26593.

lorand-horvath, 2024-06-14:
@aparcar  Custom builds still at 26593. I need the new LuCI package build, which has been fixed yesterday. Is there a way to fix this issue of outdated builds, please? Should I open a bug report or can I help with anything else?

efahl, 2024-06-14:
Deeper dive into the last few comments over on asu: https://github.com/openwrt/asu/issues/853