openwrt/openwrt #15622: newer Snapshot than r26434 on Z800AX causes system to be unresponsive

5gfreak, 2024-06-03:
### Describe the bug

Hello,

There is an issue in Z800AX when installing a newer snapshot other than r26434.

On rebooting the system remains unresponsive and the console error is:
[ 11.452083] kmodloader: no module folders for kernel version 6.6.30 found

It is only running up to snapshot r26434.

### OpenWrt version

snapshot

### OpenWrt release

r26490

### OpenWrt target/subtarget

qualcommax

### Device

Z800AX

### Image kind

Official downloaded image

### Steps to reproduce

connect to uar and you will see the bootlog up the point:
[ 11.452083] kmodloader: no module folders for kernel version 6.6.30 found

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

5gfreak, 2024-06-03:
yes it is:
it was running normally up to snashot r26434-openwrt-0adba846916d-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin

https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=qualcommax%2Fipq807x&id=zbtlink_zbt-z800ax

5gfreak, 2024-06-03:
I don't know why is marked as invalid but is a valid devide
https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=qualcommax%2Fipq807x&id=zbtlink_zbt-z800ax

robimarko, 2024-06-03:
This is probably AUC issue again

5gfreak, 2024-06-03:
I have tried to update it to newer snapshot either with uart console or attendedsys or auc, the result is the same
it remains unresponsive on[ 11.452083] kmodloader: no module folders for kernel version 6.6.30 found

5gfreak, 2024-06-03:
Yes, I did,  I only use official openwrt images

robimarko, 2024-06-03:
Can you run `ls -l /lib/modules/6.6.*` ?

Because images contain kernel modules

5gfreak, 2024-06-03:
I have to reinstall a newer snapshot because I had to revert to OpenWrt SNAPSHOT, r26403-e7661a808e
as this is my only router and I am working online now, I can try it again

5gfreak, 2024-06-03:
[z800axputty.log](https://github.com/user-attachments/files/15536186/z800axputty.log)
you will find the console log before and after flashing
from r26403 to r26490 by attendedsysupgrade online

robimarko, 2024-06-03:
Thats not a text file, please just provide plain text instead of Putty and like logs

5gfreak, 2024-06-03:
[z800axputty.txt](https://github.com/user-attachments/files/15536457/z800axputty.txt)

this one? can you open?

robimarko, 2024-06-03:
No, its just binary data

5gfreak, 2024-06-03:
here the output:
it seems that some of them are missing:
root@OpenWrt:/# ls -l /lib/modules/6.6.*
-rw-r--r--    1 root     root         35784 Jun  1 01:29 [0;0maquantia.kom
-rw-r--r--    1 root     root         38904 Jun  1 01:29 math.kom
-rw-r--r--    1 root     root        753816 Jun  1 01:29 math11k.kom
-rw-r--r--    1 root     root         27632 Jun  1 01:29 math11k_ahb.kom
-rw-r--r--    1 root     root         23144 Jun  1 01:29 mcdc-wdm.kom
-rw-r--r--    1 root     root        518312 Jun  1 01:29 mcfg80211.kom
-rw-r--r--    1 root     root         14552 Jun  1 01:29 mchacha-neon.kom
-rw-r--r--    1 root     root          7152 Jun  1 01:29 mcmac.kom
-rw-r--r--    1 root     root          3976 Jun  1 01:29 mcompat.kom
-rw-r--r--    1 root     root          5184 Jun  1 01:29 mcrc-ccitt.kom
-rw-r--r--    1 root     root          4784 Jun  1 01:29 mcrc32_generic.kom
-rw-r--r--    1 root     root          4792 Jun  1 01:29 mcrc32c_generic.kom
-rw-r--r--    1 root     root         14824 Jun  1 01:29 mdrbg.kom
-rw-r--r--    1 root     root         20952 Jun  1 01:29 mdwc3-qcom.kom
-rw-r--r--    1 root     root         74360 Jun  1 01:29 mdwc3.kom
-rw-r--r--    1 root     root        726472 Jun  1 01:29 mext4.kom
-rw-r--r--    1 root     root        610080 Jun  1 01:29 mf2fs.kom
-rw-r--r--    1 root     root          6024 Jun  1 01:29 mgeniv.kom
-rw-r--r--    1 root     root         18016 Jun  1 01:29 mgpio-button-hotplug.kom
-rw-r--r--    1 root     root          8232 Jun  1 01:29 mhmac.kom
-rw-r--r--    1 root     root         26296 Jun  1 01:29 mhwmon.kom
-rw-r--r--    1 root     root          5408 Jun  1 01:29 mip6_udp_tunnel.kom
-rw-r--r--    1 root     root        131656 Jun  1 01:29 mjbd2.kom
-rw-r--r--    1 root     root         12400 Jun  1 01:29 mjitterentropy_rng.kom
-rw-r--r--    1 root     root          7024 Jun  1 01:29 mkpp.kom
-rw-r--r--    1 root     root          9208 Jun  1 01:29 mleds-gpio.kom
-rw-r--r--    1 root     root          3496 Jun  1 01:29 mlibchacha.kom
-rw-r--r--    1 root     root         11168 Jun  1 01:29 mlibchacha20poly1305.kom
-rw-r--r--    1 root     root          5128 Jun  1 01:29 mlibcrc32c.kom
-rw-r--r--    1 root     root         28784 Jun  1 01:29 mlibcurve25519-generic.kom
-rw-r--r--    1 root     root          2968 Jun  1 01:29 mlibcurve25519.kom
-rw-r--r--    1 root     root        979168 Jun  1 01:29 mmac80211.kom
-rw-r--r--    1 root     root         12392 Jun  1 01:29 mmbcache.kom
-rw-r--r--    1 root     root          6088 Jun  1 01:29 mmichael_mic.kom
-rw-r--r--    1 root     root         10680 Jun  1 01:29 mmii.kom
-rw-r--r--    1 root     root          2636 Jun  1 01:29 mmodules.builtinm
-rw-r--r--    1 root     root          8957 Jun  1 01:29 mmodules.builtin.modinfom
-rw-r--r--    1 root     root        162032 Jun  1 01:29 mnf_conntrack.kom
-rw-r--r--    1 root     root          6944 Jun  1 01:29 mnf_defrag_ipv4.kom
-rw-r--r--    1 root     root         16904 Jun  1 01:29 mnf_defrag_ipv6.kom
-rw-r--r--    1 root     root         41264 Jun  1 01:29 mnf_flow_table.kom
-rw-r--r--    1 root     root          5456 Jun  1 01:29 mnf_flow_table_inet.kom
-rw-r--r--    1 root     root         24720 Jun  1 01:29 mnf_log_syslog.kom
-rw-r--r--    1 root     root         47712 Jun  1 01:29 mnf_nat.kom
-rw-r--r--    1 root     root          9752 Jun  1 01:29 mnf_reject_ipv4.kom
-rw-r--r--    1 root     root         10688 Jun  1 01:29 mnf_reject_ipv6.kom
-rw-r--r--    1 root     root        288096 Jun  1 01:29 mnf_tables.kom
-rw-r--r--    1 root     root         17968 Jun  1 01:29 mnfnetlink.kom
-rw-r--r--    1 root     root          5808 Jun  1 01:29 mnft_chain_nat.kom
-rw-r--r--    1 root     root         21640 Jun  1 01:29 mnft_ct.kom
-rw-r--r--    1 root     root          6768 Jun  1 01:29 mnft_fib.kom
-rw-r--r--    1 root     root          5184 Jun  1 01:29 mnft_fib_inet.kom
-rw-r--r--    1 root     root          7872 Jun  1 01:29 mnft_fib_ipv4.kom
-rw-r--r--    1 root     root          9096 Jun  1 01:29 mnft_fib_ipv6.kom
-rw-r--r--    1 root     root          9752 Jun  1 01:29 mnft_flow_offload.kom
-rw-r--r--    1 root     root          7552 Jun  1 01:29 mnft_hash.kom
-rw-r--r--    1 root     root          9880 Jun  1 01:29 mnft_limit.kom
-rw-r--r--    1 root     root          6936 Jun  1 01:29 mnft_log.kom
-rw-r--r--    1 root     root          7976 Jun  1 01:29 mnft_masq.kom
-rw-r--r--    1 root     root          7736 Jun  1 01:29 mnft_nat.kom
-rw-r--r--    1 root     root          6568 Jun  1 01:29 mnft_numgen.kom
-rw-r--r--    1 root     root          6904 Jun  1 01:29 mnft_quota.kom
-rw-r--r--    1 root     root          7720 Jun  1 01:29 mnft_redir.kom
-rw-r--r--    1 root     root          5800 Jun  1 01:29 mnft_reject.kom
-rw-r--r--    1 root     root          5472 Jun  1 01:29 mnft_reject_inet.kom
-rw-r--r--    1 root     root          4936 Jun  1 01:29 mnft_reject_ipv4.kom
-rw-r--r--    1 root     root          4936 Jun  1 01:29 mnft_reject_ipv6.kom
[... 1 lines cut ...]
-rw-r--r--    1 root     root         16568 Jun  1 01:29 mppp_async.kom
-rw-r--r--    1 root     root         59128 Jun  1 01:29 mppp_generic.kom
-rw-r--r--    1 root     root         22528 Jun  1 01:29 mpppoe.kom
-rw-r--r--    1 root     root          7264 Jun  1 01:29 mpppox.kom
-rw-r--r--    1 root     root         70144 Jun  1 01:29 mqca-nss-dp.kom
-rw-r--r--    1 root     root       1625696 Jun  1 01:29 mqca-ssdk.kom
[... cut ...]
-rw-r--r--    1 root     root         14832 Jun  1 01:29 musb_wwan.kom
-rw-r--r--    1 root     root         46328 Jun  1 01:29 musbnet.kom
-rw-r--r--    1 root     root         48016 Jun  1 01:29 musbserial.kom
-rw-r--r--    1 root     root         98616 Jun  1 01:29 mwireguard.kom
-rw-r--r--    1 root     root        210056 Jun  1 01:29 mxhci-hcd.kom
-rw-r--r--    1 root     root         18472 Jun  1 01:29 mxhci-pci.kom
-rw-r--r--    1 root     root         16088 Jun  1 01:29 mxhci-plat-hcd.kom

5gfreak, 2024-06-03:
and this is from the working snapshot: (ignore the m at the end of the previous)
root@OpenWrt:~# ls -l /lib/modules/6.6.*
-rw-r--r--    1 root     root         35720 May 25 09:17 aquantia.ko
-rw-r--r--    1 root     root         38904 May 25 09:17 ath.ko
-rw-r--r--    1 root     root        753496 May 25 09:17 ath11k.ko
-rw-r--r--    1 root     root         27632 May 25 09:17 ath11k_ahb.ko
-rw-r--r--    1 root     root         23144 May 25 09:17 cdc-wdm.ko
-rw-r--r--    1 root     root        522240 May 25 09:17 cfg80211.ko
-rw-r--r--    1 root     root         14552 May 25 09:17 chacha-neon.ko
-rw-r--r--    1 root     root          7152 May 25 09:17 cmac.ko
-rw-r--r--    1 root     root          3976 May 25 09:17 compat.ko
-rw-r--r--    1 root     root          5184 May 25 09:17 crc-ccitt.ko
-rw-r--r--    1 root     root          4784 May 25 09:17 crc32_generic.ko
-rw-r--r--    1 root     root          4792 May 25 09:17 crc32c_generic.ko
-rw-r--r--    1 root     root         14824 May 25 09:17 drbg.ko
-rw-r--r--    1 root     root         20952 May 25 09:17 dwc3-qcom.ko
-rw-r--r--    1 root     root         73872 May 25 09:17 dwc3.ko
-rw-r--r--    1 root     root        726824 May 25 09:17 ext4.ko
-rw-r--r--    1 root     root        610040 May 25 09:17 f2fs.ko
-rw-r--r--    1 root     root          6024 May 25 09:17 geniv.ko
-rw-r--r--    1 root     root         18016 May 25 09:17 gpio-button-hotplug.ko
-rw-r--r--    1 root     root          8232 May 25 09:17 hmac.ko
-rw-r--r--    1 root     root         26296 May 25 09:17 hwmon.ko
-rw-r--r--    1 root     root          5408 May 25 09:17 ip6_udp_tunnel.ko
-rw-r--r--    1 root     root        131656 May 25 09:17 jbd2.ko
-rw-r--r--    1 root     root         12400 May 25 09:17 jitterentropy_rng.ko
-rw-r--r--    1 root     root          7024 May 25 09:17 kpp.ko
-rw-r--r--    1 root     root          9208 May 25 09:17 leds-gpio.ko
-rw-r--r--    1 root     root          3432 May 25 09:17 libchacha.ko
-rw-r--r--    1 root     root         11104 May 25 09:17 libchacha20poly1305.ko
-rw-r--r--    1 root     root          5128 May 25 09:17 libcrc32c.ko
-rw-r--r--    1 root     root         28784 May 25 09:17 libcurve25519-generic.ko
-rw-r--r--    1 root     root          2968 May 25 09:17 libcurve25519.ko
-rw-r--r--    1 root     root        980584 May 25 09:17 mac80211.ko
-rw-r--r--    1 root     root         12392 May 25 09:17 mbcache.ko
-rw-r--r--    1 root     root          6088 May 25 09:17 michael_mic.ko
-rw-r--r--    1 root     root         10680 May 25 09:17 mii.ko
-rw-r--r--    1 root     root          2636 May 25 09:17 modules.builtin
-rw-r--r--    1 root     root          8957 May 25 09:17 modules.builtin.modinfo
-rw-r--r--    1 root     root        161904 May 25 09:17 nf_conntrack.ko
-rw-r--r--    1 root     root          6944 May 25 09:17 nf_defrag_ipv4.ko
-rw-r--r--    1 root     root         16904 May 25 09:17 nf_defrag_ipv6.ko
-rw-r--r--    1 root     root         41264 May 25 09:17 nf_flow_table.ko
-rw-r--r--    1 root     root          5456 May 25 09:17 nf_flow_table_inet.ko
-rw-r--r--    1 root     root         24720 May 25 09:17 nf_log_syslog.ko
-rw-r--r--    1 root     root         47584 May 25 09:17 nf_nat.ko
-rw-r--r--    1 root     root          9688 May 25 09:17 nf_reject_ipv4.ko
-rw-r--r--    1 root     root         10688 May 25 09:17 nf_reject_ipv6.ko
-rw-r--r--    1 root     root        287696 May 25 09:17 nf_tables.ko
-rw-r--r--    1 root     root         17992 May 25 09:17 nfnetlink.ko
-rw-r--r--    1 root     root          5808 May 25 09:17 nft_chain_nat.ko
-rw-r--r--    1 root     root         21640 May 25 09:17 nft_ct.ko
-rw-r--r--    1 root     root          6768 May 25 09:17 nft_fib.ko
-rw-r--r--    1 root     root          5184 May 25 09:17 nft_fib_inet.ko
-rw-r--r--    1 root     root          7872 May 25 09:17 nft_fib_ipv4.ko
-rw-r--r--    1 root     root          9096 May 25 09:17 nft_fib_ipv6.ko
-rw-r--r--    1 root     root          9752 May 25 09:17 nft_flow_offload.ko
-rw-r--r--    1 root     root          7552 May 25 09:17 nft_hash.ko
-rw-r--r--    1 root     root          9880 May 25 09:17 nft_limit.ko
-rw-r--r--    1 root     root          6936 May 25 09:17 nft_log.ko
-rw-r--r--    1 root     root          7976 May 25 09:17 nft_masq.ko
-rw-r--r--    1 root     root          7736 May 25 09:17 nft_nat.ko
-rw-r--r--    1 root     root          6568 May 25 09:17 nft_numgen.ko
-rw-r--r--    1 root     root          6904 May 25 09:17 nft_quota.ko
-rw-r--r--    1 root     root          7720 May 25 09:17 nft_redir.ko
-rw-r--r--    1 root     root          5800 May 25 09:17 nft_reject.ko
-rw-r--r--    1 root     root          5472 May 25 09:17 nft_reject_inet.ko
-rw-r--r--    1 root     root          4936 May 25 09:17 nft_reject_ipv4.ko
-rw-r--r--    1 root     root          4936 May 25 09:17 nft_reject_ipv6.ko
-rw-r--r--    1 root     root         13712 May 25 09:17 poly1305-neon.ko
-rw-r--r--    1 root     root         16568 May 25 09:17 ppp_async.ko
-rw-r--r--    1 root     root         59128 May 25 09:17 ppp_generic.ko
-rw-r--r--    1 root     root         22528 May 25 09:17 pppoe.ko
-rw-r--r--    1 root     root          7264 May 25 09:17 pppox.ko
-rw-r--r--    1 root     root         70144 May 25 09:17 qca-nss-dp.ko
-rw-r--r--    1 root     root       1625696 May 25 09:17 qca-ssdk.ko
-rw-r--r--    1 root     root         11872 May 25 09:17 qcserial.ko
[... cut ...]
-rw-r--r--    1 root     root        100432 May 25 09:17 usb-storage.ko
-rw-r--r--    1 root     root         14832 May 25 09:17 usb_wwan.ko
-rw-r--r--    1 root     root         46328 May 25 09:17 usbnet.ko
-rw-r--r--    1 root     root         48016 May 25 09:17 usbserial.ko
-rw-r--r--    1 root     root         98424 May 25 09:17 wireguard.ko
-rw-r--r--    1 root     root        209584 May 25 09:17 xhci-hcd.ko
-rw-r--r--    1 root     root         18472 May 25 09:17 xhci-pci.ko
-rw-r--r--    1 root     root         16088 May 25 09:17 xhci-plat-hcd.ko

robimarko, 2024-06-03:
Well, there is same number of kmods present so I am not seeing an issue at all.

Please post a full bootlog, and make sure to use code block formatting.

To clarify stuff:

1. Are you using official snapshots from https://downloads.openwrt.org ?
2. You are upgrading via sysupgrade?

5gfreak, 2024-06-04:
Hi Robert
yes from selector openwrt
Yes sysupgrade or attendedsysupgrade over gui, it plays no role which one

attached please find the output

[Z800AXbootlog.txt](https://github.com/user-attachments/files/15567256/Z800AXbootlog.txt)

[image]

robimarko, 2024-06-04:
Please try a current image from https://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/

I am trying to figure out what is different in your setup since I cannot reproduce this.

aiamadeus, 2024-06-05:
So which kernel version of the module exists under /lib/modules?
Did you upgrade to openwrt from stock firmware?

5gfreak, 2024-06-05:
6.6.32
no, it is a clean install

tftpboot openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-initramfs-uImage.itb
bootm
copy openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin to /tmp
and then sysupgrade -F openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin

or upgrade from snapshot r26403-openwrt-2f39ce9eae51-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin
after reboot no network is available

5gfreak, 2024-06-05:
@robimarko 
I will try a clean install again.

5gfreak, 2024-06-05:
I have the same issue on cleaning installation:

[   10.083613] kmodloader: no module folders for kernel version 6.6.30 found
[   10.251455] urngd: v1.0.2 started.
[   31.844304] l11: disabling

BusyBox v1.36.1 (2024-06-04 08:19:41 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r26515-6e51b363e9
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# ifconfig
shows only loopback interface:

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:128 errors:0 dropped:0 overruns:0 frame:0
          TX packets:128 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9984 (9.7 KiB)  TX bytes:9984 (9.7 KiB)

robimarko, 2024-06-05:
If you installed 6.6.32, then why does the error message state 6.6.30?

In the broken state, can you post what `lsmod` reports?
Also, then try running `modprobe qca_nss_dp`

5gfreak, 2024-06-05:
because I reinstalled the working one, 
If I upgrade the only connection to router is possible over serial
let me update it again and I will post the lsmod report

5gfreak, 2024-06-05:
lsmod outputs no information
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r26515-6e51b363e9
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# lsmod
root@OpenWrt:/# modprobe qca_nss_dp
no module folders for kernel version 6.6.30 found
root@OpenWrt:/#

5gfreak, 2024-06-05:
and ls /lib/modules shows 6.6.32, I think the router has ghosts inside....
root@OpenWrt:/# ls /lib/modules/
6.6.32
root@OpenWrt:/#

robimarko, 2024-06-05:
Wait, this thing has 2 rootfs partitions, maybe its mounting the wrong one.

@aiamadeus Since you added this device, can you reproduce this?

5gfreak, 2024-06-05:
no, it doesn't either:

root@OpenWrt:/# insmod /lib/modules/6.6.32/qca-ssdk.ko
no module folders for kernel version 6.6.30 found
Failed to find the folder holding the modules
root@OpenWrt:/#

robimarko, 2024-06-05:
What does `uname -a` say?

robimarko, 2024-06-05:
Well, that explains why its trying to look at the wrong folder.

My best guess is that wrong rootfs is being used.

5gfreak, 2024-06-05:
and how this can be corrected?

robimarko, 2024-06-05:
What does `fw_printenv` say?

robimarko, 2024-06-05:
Then you gotta reboot into U-Boot and just run print.

5gfreak, 2024-06-05:
IPQ807x# printenv
baudrate=115200
bootargs=console=ttyMSM0,115200n8
bootcmd=bootipq
bootdelay=2
eth1addr=
eth2addr=
eth3addr=
eth4addr=
ethact=eth0
ethaddr=
fdt_high=0x4A400000
fdtcontroladdr=4a970220
flash_type=7
ipaddr=192.168.1.11
machid=8010008
netmask=255.255.255.0
serverip=192.168.1.10
soc_version_major=2
soc_version_minor=0
stderr=serial@78B3000
stdin=serial@78B3000
stdout=serial@78B3000

Environment size: 510/65532 bytes

I removed the MACs addresses from screen

robimarko, 2024-06-05:
Well, it seems to rely on `bootipq` for the boot logic and `bootargs`.

I think we will have to wait for @aiamadeus if he has better insight

5gfreak, 2024-06-05:
by the way I have a question regarding flashing this router
I am actually using the option:
1.tftpboot openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-initramfs-uImage.itb
2.bootm
3.copy openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin to /tmp
4.and then sysupgrade -F /tmp/openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin

is there any other way to flash the system with the bin file over serial ? 
what is the difference between factory.bin & factory.ubi? images
what image format is required to use the usbboot command

Thank you

aiamadeus, 2024-06-05:
Can you share your "0:bootconfig" and "0:bootconfig1" partition dump?
I think we can set which partition to boot from without breaking the firmware upgrade from sotck.

aiamadeus, 2024-06-05:
Confirm the partition location:
```
root@OpenWrt:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 03400000 00020000 "rootfs"
mtd1: 00800000 00020000 "0:wififw"
mtd2: 03400000 00020000 "rootfs_1"
mtd3: 00800000 00020000 "0:wififw_1"
mtd4: 00050000 00010000 "0:sbl1"
mtd5: 00010000 00010000 "0:mibib"
mtd6: 00020000 00010000 "0:bootconfig"
mtd7: 00020000 00010000 "0:bootconfig1"
mtd8: 00180000 00010000 "0:qsee"
mtd9: 00180000 00010000 "0:qsee_1"
mtd10: 00010000 00010000 "0:devcfg"
mtd11: 00010000 00010000 "0:devcfg_1"
mtd12: 00010000 00010000 "0:apdp"
mtd13: 00010000 00010000 "0:apdp_1"
mtd14: 00040000 00010000 "0:rpm"
mtd15: 00040000 00010000 "0:rpm_1"
mtd16: 00010000 00010000 "0:cdt"
mtd17: 00010000 00010000 "0:cdt_1"
mtd18: 00010000 00010000 "0:appsblenv"
mtd19: 000a0000 00010000 "0:appsbl"
mtd20: 000a0000 00010000 "0:appsbl_1"
mtd21: 00040000 00010000 "0:art"
mtd22: 00080000 00010000 "0:ethphyfw"
```
Dump partition:
```
dd if=/dev/mtd6 of=/tmp/bootconfig.bin
dd if=/dev/mtd7 of=/tmp/bootconfig1.bin
```

5gfreak, 2024-06-05:
[bootconfig0_1.zip](https://github.com/user-attachments/files/15592460/bootconfig0_1.zip)

aiamadeus, 2024-06-05:
[image]
It doesn't seem to be caused by booting from another partition...

5gfreak, 2024-06-05:
if you want to remote access on this router I can use teamsviewer or anydesk or what ever you prefer

5gfreak, 2024-06-05:
here you go
[upgrade process.txt](https://github.com/user-attachments/files/15592894/upgrade.process.txt)

aiamadeus, 2024-06-05:
```
[  170.058371] ubi0: attaching mtd0
[  170.291770] ubi0: scanning is finished
[  170.297823] ubi0: attached mtd0 (name "rootfs", size 52 MiB)
[  170.297865] ubi0: PEB size: 131072 bytes (128 KiB), LEB size: 126976 bytes
[  170.302558] ubi0: min./max. I/O unit sizes: 2048/2048, sub-page size 2048
[  170.309265] ubi0: VID header offset: 2048 (aligned 2048), data offset: 4096
[  170.316104] ubi0: good PEBs: 416, bad PEBs: 0, corrupted PEBs: 0
[  170.322876] ubi0: user volume: 3, internal volumes: 1, max. volumes count: 128
[  170.329121] ubi0: max/mean erase counter: 75/46, WL threshold: 4096, image sequence number: 921923990
[  170.336165] ubi0: available PEBs: 0, total reserved PEBs: 416, PEBs reserved for bad PEB handling: 40
[  170.345454] ubi0: background thread "ubi_bgt0d" started, PID 3862
[  170.346114] block ubiblock0_1: created from ubi0:1(rootfs)
UBI device number 0, total 416 LEBs (52822016 bytes, 50.3 MiB), available 0 LEBs (0 bytes), LEB size 126976 bytes (124.0 KiB)
[  170.447838] ------------[ cut here ]------------
[  170.447874] sysfs group 'power' not found for kobject 'ubiblock0_1'
[  170.451544] WARNING: CPU: 2 PID: 3898 at fs/sysfs/group.c:282 sysfs_remove_group+0x94/0xa0
[  170.457519] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath11k_ahb(O) ath11k(O) pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211(O) cfg80211(O) slhc qrtr_smd qrtr qmi_helpers(O) nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat(O) sha512_generic sha512_arm64 seqiv sha3_generic drbg michael_mic hmac geniv cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp(O) qca_ssdk(O) gpio_button_hotplug(O) ext4 mbcache jbd2 aquantia hwmon crc_ccitt crc32c_generic
[  170.511582] CPU: 2 PID: 3898 Comm: ubirmvol Tainted: G           O       6.6.32 #0
[  170.533820] Hardware name: Zbtlink ZBT-Z800AX (DT)
[  170.541195] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  170.545971] pc : sysfs_remove_group+0x94/0xa0
[  170.552825] lr : sysfs_remove_group+0x94/0xa0
[  170.557338] sp : ffffffc08741b960
[  170.561676] x29: ffffffc08741b960 x28: ffffff8003994800 x27: ffffff8002bc8730
[  170.564983] x26: 000000000000005f x25: ffffff8004308000 x24: 0000000000000000
[  170.572100] x23: ffffffc081378000 x22: ffffffc081378a98 x21: ffffff8002dbf8c8
[  170.579219] x20: ffffffc0808ac8b0 x19: 0000000000000000 x18: 0000000000000146
[  170.586336] x17: 0000000000000004 x16: ffffff8002916080 x15: ffffffc081386ee0
[  170.593455] x14: 00000000000003d2 x13: 0000000000000146 x12: 00000000ffffffea
[  170.600573] x11: 00000000ffffefff x10: ffffffc0813deee0 x9 : ffffffc081386e88
[  170.607691] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[  170.614809] x5 : 0000000000000fff x4 : 0000000000000000 x3 : ffffffc08741b760
[  170.621926] x2 : ffffffc081386e20 x1 : ffffffc081386e20 x0 : 0000000000000037
[  170.629046] Call trace:
[  170.636154]  sysfs_remove_group+0x94/0xa0
[  170.638416]  dpm_sysfs_remove+0x5c/0x6c
[  170.642579]  device_del+0xa0/0x424
[  170.646225]  del_gendisk+0x1d0/0x328
[  170.649697]  ubiblock_remove+0xac/0x14c
[  170.653430]  ubiblock_notify+0x64/0x274
[  170.656988]  blocking_notifier_call_chain+0x6c/0xa0
[  170.660811]  ubi_volume_notify+0x4c/0x60
[  170.665671]  ubi_remove_volume+0x218/0x33c
[  170.669837]  ubi_cdev_ioctl+0x440/0xb34
[  170.673743]  __arm64_sys_ioctl+0x1f4/0xe4c
[  170.677477]  invoke_syscall.constprop.0+0x5c/0x108
[  170.681643]  do_el0_svc+0x40/0xc8
[  170.686416]  el0_svc+0x30/0xb8
[  170.689800]  el0t_64_sync_handler+0x120/0x12c
[  170.692754]  el0t_64_sync+0x178/0x17c
[  170.697181] ---[ end trace 0000000000000000 ]---
[  170.702892] block ubiblock0_1: released
Volume ID 0, size 43 LEBs (5459968 bytes, 5.2 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "kernel", alignment 1
[  170.954958] block ubiblock0_1: created from ubi0:1(rootfs)
Volume ID 1, size 56 LEBs (7110656 bytes, 6.7 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs", alignment 1
Set volume size to 34664448
Volume ID 2, size 273 LEBs (34664448 bytes, 33.0 MiB), LEB size 126976 bytes (124.0 KiB), dynamic, name "rootfs_data", alignment 1
sysupgrade successful
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy
[  174.101856] Suspending MXIC NAND chip failed (-524)
[  174.633073] reboot: Restarting system
```
It's weird, there's a bug that breaks the sysupgrade.
@robimarko What do you think about this?

testuser7, 2024-06-05:
@aiamadeus Are you sure that this bug prevents from sysupgrade?
There is `sysupgrade successful` message at the end.

I have reported this problem before: https://github.com/openwrt/openwrt/pull/14844#issuecomment-1998680617

5gfreak, 2024-06-06:
here are the working snapshots I have and can be sysupgraded and run on z800 normally,
if it helps....
r26403
[r26403-openwrt-2f39ce9eae51-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.zip](https://github.com/user-attachments/files/15605032/r26403-openwrt-2f39ce9eae51-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.zip)

5gfreak, 2024-06-06:
r26434
[r26434-openwrt-0adba846916d-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.zip](https://github.com/user-attachments/files/15605039/r26434-openwrt-0adba846916d-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.zip)

robimarko, 2024-06-06:
@aiamadeus Its just a warning for sysfs that I have not been able to reproduce though it seems like a real bug in the kernel, a similar one was present for PCI and others before.

But it should not affect upgrading.

5gfreak, 2024-06-06:
so until this issue is corrected, I will stay at the snapshot mentioned above.
I have no problem with those.

Thanks to both of you
:-)

robimarko, 2024-06-06:
Well, its not going to fix itself

5gfreak, 2024-06-06:
if you want to test anything on this router I will be more than willing to help

aiamadeus, 2024-06-07:
> I am actually using the option:
> 1.tftpboot openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-initramfs-uImage.itb
> 2.bootm 3.copy openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-sysupgrade.bin [...]

I did this, upgraded from 6.6.29 to 6.6.32, no recurrence. Can you do this:
`tftpboot openwrt-qualcommax-ipq807x-zbtlink_zbt-z800ax-squashfs-factory.ubi` and `flash rootfs`

5gfreak, 2024-06-07:
the same error 
[flash rootfs.txt](https://github.com/user-attachments/files/15709956/flash.rootfs.txt)

5gfreak, 2024-06-07:
since I flash rootfs I cannot revert it to the previous working snapshot...
can you write how I can use my older snapshot again?

5gfreak, 2024-06-07:
by the way I have an original image but the file has an extension .img
how can I install the original .img to z800?

aiamadeus, 2024-06-07:
> [bootconfig0_1.zip](https://github.com/user-attachments/files/15592460/bootconfig0_1.zip)

[image]
> NAND erase: device 0 offset 0x3c00000, size 0x3400000
> NAND write: device 0 offset 0x3c00000, size 0x1180000

Sorry, I misread it before. This is booted from rootfs_1 partition.
Can you try this commit? https://github.com/aiamadeus/openwrt/commit/869669d

aiamadeus, 2024-06-07:
Try this image: https://files.catbox.moe/gydxjy.zip

5gfreak, 2024-06-07:
I could install it but I get the same error
[   12.011390] kmodloader: no module folders for kernel version 6.6.32 found

aiamadeus, 2024-06-07:
```
uname -a
find_mtd_index 0:bootconfig
hexdump -e '1/1 "%01x|"' -n 1 -s 168 -C /dev/mtd6 | cut -f 1 -d "|" | head -n1
```
What is the output of this command?

5gfreak, 2024-06-07:
root@OpenWrt:/# uname -a
Linux OpenWrt 6.6.32 #0 SMP Fri Jun  7 07:00:15 2024 aarch64 GNU/Linux
root@OpenWrt:/# find_mtd_index 0:bootconfig
/bin/ash: find_mtd_index: not found
root@OpenWrt:/# hexdump -e '1/1 "%01x|"' -n 1 -s 168 -C /dev/mtd6 | cut -f 1 -d
"|" | head -n1
1
root@OpenWrt:/#

aiamadeus, 2024-06-07:
Execute the following command in uboot:
```
flasherase 0:BOOTCONFIG
flasherase 0:BOOTCONFIG1
reset
```
Then flash the firmware again.

5gfreak, 2024-06-07:
the firmware you attached above, right?

testuser7, 2024-06-07:
@aiamadeus There is some implementation of changing the boot in the `bootconfig` partition: https://github.com/openwrt/openwrt/blob/main/target/linux/ipq806x/base-files/lib/upgrade/asrock.sh
Maybe something similar should be implemented.

[1 more reply]

PR #15700 (2024-06-13): qualcommax: ipq807x: fix sysupgrade for ZBT-Z800AX
This router has two rootfs partitions and dualboot is used. Vendor firmware may swap the rootfs partition location, and vendor u-boot will append 'ubi.mtd=rootfs' in the end of cmdline. Since we use fixed-partitions, force boot from the first rootfs partition to avoid boot failure.
Fixes: #15622
closed 2024-06-13 by openwrt-bot