openwrt/openwrt #15625: The ubus object exists, but when called by invoke, it returns the status 4

zhengyikeng, 2024-06-03:
### Describe the bug

I added an Ubus object, which can be queried through the Ubus list
```
root@DataJar-D0F1:~# ubus -v list systemd.virtual_led_service
'systemd.virtual_led_service' @44c1359e
        "list":{}
        "set":{"led_name":"String","led_status":"Integer"}
        "get":{"led_name":"String"}
```

The ubus object exists, but when called by invoke, it returns the status 4. 
```
ubus call systemd.virtual_led_service list &
-> b2b0087d #b2b0087d          hello: {}
<- b2b0087d #00000000         lookup: {"objpath":"systemd.virtual_led_service"}
-> b2b0087d #00000000           data: {"objpath":"systemd.virtual_led_service","objid":1153512862,"objtype":-290628632,"signature":{"list":{},"set":{"led_name":3,"led_status":5},"get":{"led_name":3}}}
-> b2b0087d #00000000         status: {"status":0}
<- b2b0087d #44c1359e         invoke: {"objid":1153512862,"method":"list","data":{}}
-> 808599b2 #b2b0087d         invoke: {"objid":1153512862,"method":"list","data":{},"user":"root","group":"root"}
<- 808599b2 #00000000         status: {"status":4,"objid":0}

```

Status 4 means UBUS_STATUS_NOT_FOUND? May I ask how to troubleshoot? 

### OpenWrt version

DISTRIB_REVISION='2.14.51'

### OpenWrt release

DISTRIB_RELEASE='22.03.3'

### OpenWrt target/subtarget

DISTRIB_TARGET='rockchip/armv8'

### Device

RK3308B

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-06-03:
Can you repeat on mainstream stable OpenWRT and copy proper release tags?

zhengyikeng, 2024-06-07:
My problem has been resolved. My code has a mechanism that attempts to subscribe periodically after a subscription failure. The same struct ubus_subscriber has called API ubus_register_subscriber() multiple times, which can cause structural abnormalities in the avl tree

[image]

Just a suggestion, would it be better to do a check before insertion
```c
diff --git a/libubus-obj.c b/libubus-obj.c
index 29cbb2b..aec0a4d 100644
--- a/libubus-obj.c
+++ b/libubus-obj.c
@@ -169,7 +169,9 @@ static void ubus_add_object_cb(struct ubus_request *req, int type, struct blob_a
 
        if (attrbuf[UBUS_ATTR_OBJTYPE])
                obj->type->id = blob_get_u32(attrbuf[UBUS_ATTR_OBJTYPE]);
-
+       /*
+        * Suggest: check if avl node exists in the AVL tree? and if it exists, would it be better to remove it first?
+        */
        obj->avl.key = &obj->id;
        avl_insert(&req->ctx->objects, &obj->avl);

```