openwrt/openwrt #15629: Incorrect dev->cpu_port value for some routers with switches from the rtl8367, rtl8367b family

namiltd, 2024-06-04:
### Describe the bug

kernel: rtl8367, rtl8367b: Invalid dev->cpu_port value
For some switches from the rtl8367, rtl8367b families, the dev->cpu_port variable returns an incorrect value
The problem concerns routers:
Asus RT N56U - the return value is 9 instead of 8
IO Data WN-AC733GR3 - the return value is 5 instead of 6
Dlink DIR-645 - the return value is 5 instead of 6

### OpenWrt version

r26492-af329ec389

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/rt3883

### Device

D-Link DIR-645

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-06-04:
The solution is to apply the patch https://github.com/openwrt/openwrt/pull/15033
It uses setting cpu_port based on the use of realtek, extif0 to extif2.

robimarko, 2024-06-08:
PR has been merged, closing as completed.

closed 2024-06-08 by robimarko