openwrt/openwrt #15632: mount_root does not work for ubi devices

jpsollie, 2024-06-05:
### Describe the bug

when booting to recovery system,
mount_root doesn't mount anything to /overlay.
It also reports no error.
/proc/mounts doesn't say anything.
dwfreed suggested it may not have been updated to work with UBI on snand, so here's the bug report.

### OpenWrt version

r0+26513-65f34e9928

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R4

### Image kind

Self-built image

### Steps to reproduce

reboot into recovery
run mount_root to mount root filesystem into /overlay and access config files

### Actual behaviour

no mount is performed.
cat /proc/mounts is empty
no error is given - not by the command or in dmesg

### Expected behaviour

the root filesystem is mounted read-write in /overlay

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH11K_THERMAL=y
CONFIG_ATHEEPMGR_DANGEROUS_FUNCTIONS=y
CONFIG_BPF_TOOLCHAIN_HOST=y
CONFIG_BPF_TOOLCHAIN_HOST_PATH=""
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUILD_LOG=y
CONFIG_BUILD_PATENTED=y
CONFIG_BUSYBOX_CONFIG_ASH_BASH_NOT_FOUND_HOOK=y
CONFIG_BUSYBOX_CONFIG_ASH_HELP=y
CONFIG_BUSYBOX_CONFIG_ASH_IDLE_TIMEOUT=y
CONFIG_BUSYBOX_CONFIG_ASH_SLEEP=y
CONFIG_BUSYBOX_CONFIG_BASH_IS_ASH=y
# CONFIG_BUSYBOX_CONFIG_BASH_IS_NONE is not set
CONFIG_BUSYBOX_CONFIG_BB_ARCH=y
CONFIG_BUSYBOX_CONFIG_BLKDISCARD=y
CONFIG_BUSYBOX_CONFIG_BLKID=y
[... 1382 lines cut ...]
# CONFIG_LLDPD_WITH_SONMP is not set
CONFIG_MOLD=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_brcmfmac-firmware-usb=m
CONFIG_PACKAGE_libcomerr=y
# CONFIG_PACKAGE_libnghttp2 is not set
CONFIG_PACKAGE_libss=y
# CONFIG_PACKAGE_libxdp is not set
CONFIG_PACKAGE_openwrt-keyring=y
# CONFIG_PACKAGE_perf is not set
# CONFIG_PACKAGE_qosify is not set
CONFIG_PACKAGE_usign=y
# CONFIG_PACKAGE_xdp-filter is not set
# CONFIG_PACKAGE_xdp-loader is not set
# CONFIG_PACKAGE_xdpdump is not set
# CONFIG_RSYNC_acl is not set
# CONFIG_USE_MOLD is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dangowrt, 2024-06-05:
I think there is a small misunderstanding here:
You are talking about the recovery image (= initramfs)  on the BPi-R4.
`mount_root`, however, is a command available in when booting the production image (= kernel+squashfs) when interrupting boot before the read/write overlay has been mounted.

If you want to mount the rootfs while using initramfs you can still do so also, but you have to do it manually:
```
mount /dev/fit0 /mnt
```

Mounting the read/write overlay on top depends on the boot media, it will either be in a ubifs volume when booting from NAND:
```
mount /dev/ubiblock0_X /mnt/overlay
```

Or, when booting from SD card or eMMC:
```
mount /dev/fitfw /mnt/overlay
```

jpsollie, 2024-06-06:
Oops ... Then this is completely irrelevant. Sorry. Topic closef

closed 2024-06-06 by jpsollie