openwrt/openwrt #15642: losetup crashes kernel

M95D, 2024-06-06:
### Describe the bug

Trying to attach a loop device to an openwrt image file causes a kernel null pointer dereference.
I can reproduce the problem with kernels v6.6 and v6.1 (from snapshot, self built), but not v5.15 (from stable release image).

### OpenWrt version

r26290

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Turris Omnia

### Image kind

Self-built image

### Steps to reproduce

- Build busybox with losetup included. Separate losetup package may trigger this bug too - I didn't test.
- Download and gunzip an OpenWrt squasfs image for Turris Omnia
- Run `losetup -fP <image>`

### Actual behaviour

```
# losetup -fP openwrt-23.05.3-mvebu-cortexa9-cznic_turris-omnia-sysupgrade.img
[  175.547525][ T5705] 8<--- cut here ---
[  175.551323][ T5705] Unable to handle kernel NULL pointer dereference at virtual address 00000180
[  175.560150][ T5705] [00000180] *pgd=00000000
[  175.564451][ T5705] Internal error: Oops: 5 [#1] SMP ARM
[  175.569779][ T5705] Modules linked in: ath9k(O) ath9k_common(O) nf_flow_table_inet iptable_nat ath9k_hw(O) ath(O) xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mac80211(O) ipt_REJECT cfg80211(O) xt_time xt_tcpudp xt_tcpmss xt_statistic xt_physdev xt_multiport xt_mark xt_mac xt_limit xt_length xt_ipv4options(O) xt_iprange xt_iface(O) xt_hl xt_ecn xt_dscp xt_comment xt_TRACE xt_TEE xt_TCPMSS xt_PROTO(O) xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbip_host usbip_core nfnetlink_queue nfnetlink nf_reject_ipv4 nf_log_syslog nf_dup_ipv6 nf_dup_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip6table_raw ip_tables compat(O) br_netfilter dvb_usb_rtl28xxu cxd2841er r820t
[  175.569947][ T5705]  dvb_usb_v2 dvb_core mc cryptodev(O) ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 rpcsec_gss_krb5 des_generic dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax mii
[  175.675680][ T5705] CPU: 1 PID: 5705 Comm: losetup Tainted: G           O       6.1.89 #0
[  175.683881][ T5705] Hardware name: Marvell Armada 380/385 (Device Tree)
[  175.690510][ T5705] PC is at 0xc0706004
[  175.694356][ T5705] LR is at 0xc05d985c
[  175.698199][ T5705] pc : [<c0706004>]    lr : [<c05d985c>]    psr: 60000013
[  175.705177][ T5705] sp : f2b29cc8  ip : 00000000  fp : c14b83c0
[  175.711109][ T5705] r10: 00000000  r9 : c14b83c0  r8 : c14b87b0
[  175.717042][ T5705] r7 : c1da7e30  r6 : c14b8780  r5 : 00000001  r4 : c1da7e00
[  175.724280][ T5705] r3 : 00000000  r2 : 00000000  r1 : c0b42ae4  r0 : 00000000
[  175.731519][ T5705] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  175.739369][ T5705] Control: 10c5387d  Table: 065a404a  DAC: 00000051
[  175.745823][ T5705] Register r0 information: NULL pointer
[  175.751237][ T5705] Register r1 information: non-slab/vmalloc memory
[  175.757607][ T5705] Register r2 information: NULL pointer
[  175.763018][ T5705] Register r3 information: NULL pointer
[  175.768429][ T5705] Register r4 information: slab kmalloc-512 start c1da7e00 pointer offset 0 size 512
[  175.777767][ T5705] Register r5 information: non-paged memory
[  175.783527][ T5705] Register r6 information: slab bdev_cache start c14b8780 pointer offset 0
[  175.791991][ T5705] Register r7 information: slab kmalloc-512 start c1da7e00 pointer offset 48 size 512
[  175.801413][ T5705] Register r8 information: slab bdev_cache start c14b8780 pointer offset 48
[  175.809963][ T5705] Register r9 information: slab bdev_cache start c14b83c0 pointer offset 0
[  175.818426][ T5705] Register r10 information: NULL pointer
[  175.823924][ T5705] Register r11 information: slab bdev_cache start c14b83c0 pointer offset 0
[  175.832475][ T5705] Register r12 information: NULL pointer
[  175.837972][ T5705] Process losetup (pid: 5705, stack limit = 0xe835189e)
[  175.844779][ T5705] Stack: (0xf2b29cc8 to 0xf2b2a000)
[  175.849844][ T5705] 9cc0:                   10300000 c0ab044c 00000cc0 c023925c 00000000 00000000
[  175.858739][ T5705] 9ce0: 00000000 23fc6bf6 c0c20344 c1da7e00 00000001 00000800 00000000 00000080
[  175.867633][ T5705] 9d00: c0c20344 00008a00 c3789540 c05d9e90 00008200 00000000 00000000 f2b5b095
[  175.876528][ T5705] 9d20: 00000000 c0b37cb8 c0b42ec0 00000000 00000000 c1da7e0c c0b42bdc c1d99400
[  175.885422][ T5705] 9d40: c39c6000 c1d99400 00000200 c14b83c0 484e001f 00000000 c1d9952c c071c238
[  175.894317][ T5705] 9d60: 00000200 c14b83c0 484e001f 00000000 c39c6000 c071e4d4 c39c6000 00000000
[  175.903211][ T5705] 9d80: 00000000 00000000 c15a7040 c15a6f48 beff2bb8 c14b83c0 beff2bb8 484e001f
[  175.912106][ T5705] 9da0: c1d99400 00000051 c1dc29b0 00000036 002c0d2c c071e930 c4156e58 c4156ea8
[  175.921000][ T5705] 9dc0: 00000003 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  175.929894][ T5705] 9de0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000008
[  175.938789][ T5705] 9e00: 6e65706f 2d747277 302e3332 2d332e35 6265766d 6f632d75 78657472 632d3961
[  175.947683][ T5705] 9e20: 63696e7a 7275745f 2d736972 696e6d6f 79732d61 67707573 65646172 006d692e
[  175.956578][ T5705] 9e40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  175.965472][ T5705] 9e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  175.974366][ T5705] 9e80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  175.983260][ T5705] 9ea0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  175.992154][ T5705] 9ec0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  176.001049][ T5705] 9ee0: 00000000 00000000 00000000 00000000 00100cca 23fc6bf6 00260000 00004c0a
[  176.009944][ T5705] 9f00: c39c60c0 beff2bb8 c39c60c0 00000004 c1dc29b0 00000036 002c0d2c c0277524
[  176.018839][ T5705] 9f20: 00000000 23fc6bf6 002bf000 f2b29fb0 00260998 80000007 c1da8000 00000873
[  176.027734][ T5705] 9f40: 00000004 c29e2200 00000254 c0110388 f2b29f58 23fc6bf6 66616a72 00000000
[  176.036628][ T5705] 9f60: 08f0d180 00000000 66616a72 00000000 08f0d180 00000000 00000036 23fc6bf6
[  176.045523][ T5705] 9f80: f2b29fb0 00000004 beff2bb8 00000002 00000036 c01002c4 c29e2200 00000036
[  176.054417][ T5705] 9fa0: 002c0d2c c01002b4 00000004 beff2bb8 00000004 00004c0a beff2bb8 beff2af0
[  176.063311][ T5705] 9fc0: 00000004 beff2bb8 00000002 00000036 beff2d3c 00000000 00000000 002c0d2c
[  176.072205][ T5705] 9fe0: b6fccf78 beff29c0 0025c144 b6f81fa8 60000010 00000004 00000000 00000000
[  176.081100][ T5705] Function entered at [<c0706004>] from [<c05d985c>]
[  176.087644][ T5705] Function entered at [<c05d985c>] from [<c05d9e90>]
[  176.094187][ T5705] Function entered at [<c05d9e90>] from [<c071c238>]
[  176.100729][ T5705] Function entered at [<c071c238>] from [<c071e4d4>]
[  176.107271][ T5705] Function entered at [<c071e4d4>] from [<c071e930>]
[  176.113814][ T5705] Function entered at [<c071e930>] from [<c0277524>]
[  176.120357][ T5705] Function entered at [<c0277524>] from [<c01002b4>]
[  176.126898][ T5705] Exception stack(0xf2b29fa8 to 0xf2b29ff0)
[  176.132659][ T5705] 9fa0:                   00000004 beff2bb8 00000004 00004c0a beff2bb8 beff2af0
[  176.141553][ T5705] 9fc0: 00000004 beff2bb8 00000002 00000036 beff2d3c 00000000 00000000 002c0d2c
[  176.150447][ T5705] 9fe0: b6fccf78 beff29c0 0025c144 b6f81fa8
[  176.156208][ T5705] Code: 1283000c 05900184 e3a03000 e12fff1e (e5903180)
[  176.163050][ T5705] ---[ end trace 0000000000000000 ]---
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-06-06:
[166782.916104] nbd0: detected capacity change from 0 to 45509
[166782.916446]  nbd0: p1 p2
[166782.916539] nbd0: p2 size 213504 extends beyond EOD, truncated
[166782.916874]  nbd0: p1 p2
[166782.916943] nbd0: p2 size 213504 extends beyond EOD, truncated

M95D, 2024-06-06:
Is this from Turris Omnia with kernel v6.1 or v6.6?

brada4, 2024-06-06:
I know, nbd just illustrates inconsistency losetup dislikes.

M95D, 2024-06-06:
No... it's not that... I can losetup and mount the exact same file on x86.

ynezz, 2024-06-06:
> OpenWrt release: SNAPSHOT

Can you actually try with the snapshot build so anyone interested gets the stack trace resolved already?

```[  176.081100][ T5705] Function entered at [<c0706004>] from [<c05d985c>]
[  176.087644][ T5705] Function entered at [<c05d985c>] from [<c05d9e90>]
[  176.094187][ T5705] Function entered at [<c05d9e90>] from [<c071c238>]
[  176.100729][ T5705] Function entered at [<c071c238>] from [<c071e4d4>]
[  176.107271][ T5705] Function entered at [<c071e4d4>] from [<c071e930>]
[  176.113814][ T5705] Function entered at [<c071e930>] from [<c0277524>]
[  176.120357][ T5705] Function entered at [<c0277524>] from [<c01002b4>]```

M95D, 2024-06-07:
Setup:
- Official OpenWrt image written to internal mmc
- USB stick mounted to /mnt/usb with my own openwrt-mvebu-cortexa9-cznic_turris-omnia-rootfs.tar.gz
- rootfs from my own build extracted to /tmp/root (tmpfs)
- strace and busybox run from /tmp/root directly, without chroot

Note that kernel log and strace log are mixed in the captured serial output, with strace lines periodically interrupted by kernel lines. I did my best to separate the two. I can post the unedited serial capture if needed.

Official OpenWrt v23.05.3:
```
root@OpenWrt:/tmp/root# ./usr/bin/strace -f ./bin/busybox losetup -fP /mnt/usb/openwrt-23.05.3-mvebu-cortexa9-cznic_turris-omnia-sysupgrade.img
execve("./bin/busybox", ["./bin/busybox", "losetup", "-fP", "/mnt/usb/openwrt-23.05.3-mvebu-c"...], 0xbea45de0 /* 11 vars */) = 0
set_tls(0xb6fec4dc)                     = 0
set_tid_address(0xb6feb7d4)             = 3119
brk(NULL)                               = 0x1f6c000
brk(0x1f6e000)                          = 0x1f6e000
mmap2(0x1f6c000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1f6c000
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=45058, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 49152, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6f69000
mmap2(0xb6f73000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0xb6f73000
close(3)                                = 0
mprotect(0xb6f73000, 4096, PROT_READ)   = 0
mprotect(0x2bf000, 4096, PROT_READ)     = 0
prctl(PR_GET_NAME, "busybox")           = 0
open("/dev/loop-control", O_RDWR|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
ioctl(3, LOOP_CTL_GET_FREE)             = 1
close(3)                                = 0
open("/dev/loop1", O_RDONLY|O_LARGEFILE) = 3
ioctl(3, LOOP_GET_STATUS64, 0xbeddfc10) = -1 ENXIO (No such device or address)
close(3)                                = 0
open("/mnt/usb/openwrt-23.05.3-mvebu-cortexa9-cznic_turris-omnia-sysupgrade.img", O_RDWR|O_LARGEFILE) = 3
statx(AT_FDCWD, "/dev/loop1", AT_STATX_SYNC_AS_STAT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFBLK|0600, stx_size=0, ...}) = 0
open("/dev/loop1", O_RDWR|O_LARGEFILE)  = 4
ioctl(4, LOOP_GET_STATUS64, 0xbeddfbb0) = -1 ENXIO (No such device or address)
ioctl(4, LOOP_CONFIGURE, {fd=3, block_size=0, info={lo_offset=0, lo_number=0, lo_flags=LO_FLAGS_PARTSCAN, lo_file_name="/mnt/usb/openwrt-23.05.3-mvebu-cortexa9-cznic_turris-omnia-sysu", ...}}) = 0
close(3)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++

[ 1379.370757] loop1: detected capacity change from 0 to 45508
[ 1379.378063]  loop1: p1 p2
[ 1379.380921] loop1: p2 size 213504 extends beyond EOD, truncated

```

Official OpenWrt snapshot r26518-f1091ef7ac :
```
root@OpenWrt:/tmp/root# ./usr/bin/strace -f ./bin/busybox losetup -fP /mnt/usb/openwrt-23.05.3-mvebu-cortexa9-cznic_turris-omnia-sysupgrade.img
execve("./bin/busybox", ["./bin/busybox", "losetup", "-fP", "/mnt/usb/openwrt-23.05.3-mvebu-c"...], 0xbed40df0 /* 11 vars */) = 0
set_tls(0xb6f12fac)                     = 0
set_tid_address(0xb6f11854)             = 2845
brk(NULL)                               = 0x1b4e000
brk(0x1b50000)                          = 0x1b50000
mmap2(0x1b4e000, 4096, PROT_NONE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x1b4e000
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=45058, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\0\0\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 49152, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6e89000
mmap2(0xb6e93000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0xb6e93000
close(3)                                = 0
mprotect(0xb6e93000, 4096, PROT_READ)   = 0
mprotect(0x2bf000, 4096, PROT_READ)     = 0
prctl(PR_GET_NAME, "busybox")           = 0
[... 84 lines cut ...]
[  278.396262] Kernel panic - not syncing: Fatal exception
[  278.402884] CPU0: stopping
[  278.402888] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D    O       6.6.32 #0
[  278.402894] Hardware name: Marvell Armada 380/385 (Device Tree)
[  278.402898]  unwind_backtrace from show_stack+0x10/0x14
[  278.402909]  show_stack from dump_stack_lvl+0x40/0x4c
[  278.402921]  dump_stack_lvl from do_handle_IPI+0xec/0x120
[  278.402931]  do_handle_IPI from ipi_handler+0x18/0x20
[  278.402937]  ipi_handler from handle_percpu_devid_irq+0x78/0x134
[  278.402949]  handle_percpu_devid_irq from generic_handle_domain_irq+0x28/0x38
[  278.402960]  generic_handle_domain_irq from gic_handle_irq+0x74/0x88
[  278.402967]  gic_handle_irq from generic_handle_arch_irq+0x34/0x44
[  278.402977]  generic_handle_arch_irq from call_with_stack+0x18/0x20
[... cut ...]
[  278.403004] 1f40: c0e04f68 c0e08600 00000000 c0e04fa4 00000000 00000000 c0e04f08 c0e08474
[  278.403008] 1f60: c1379838 c0e01f80 c095fcdc c09603f0 60000013 ffffffff
[  278.403011]  __irq_svc from default_idle_call+0x2c/0x30
[  278.403020]  default_idle_call from do_idle+0x188/0x1d8
[  278.403035]  do_idle from cpu_startup_entry+0x28/0x2c
[  278.403047]  cpu_startup_entry from kernel_init+0x0/0x12c
[  278.403058]  kernel_init from arch_post_acpi_subsys_init+0x0/0x8
[  278.533149] Rebooting in 3 seconds..
```

john-tho, 2024-06-15:
>```
>losetup -fP /mnt/usb/openwrt-23.05.3-mvebu-cortexa9-cznic_turris-omnia-sysupgrade.img
>```
>```
>[  278.300997]  device_get_named_child_node from add_partition+0x3d0/0x428
>[  278.309011]  add_partition from bdev_disk_changed+0x374/0x604
>[  278.316157]  bdev_disk_changed from loop_reread_partitions+0x24/0x68
>[  278.323914]  loop_reread_partitions from loop_configure+0x498/0x534
>```

This partition fwnode code comes from: https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.6/450-09-block-partitions-populate-fwnode.patch:
https://github.com/openwrt/openwrt/blob/22f92cce22401ef7df44357a7e6921d2cf45c840/target/linux/generic/pending-6.6/450-09-block-partitions-populate-fwnode.patch#L29-L38
One for @dangowrt

At a guess, either `ddev` or `ddev->parent` is `NULL`
If you can compile a test image yourself, a quick test for `->parent` would be:
```patch
diff --git a/target/linux/generic/pending-6.6/450-09-block-partitions-populate-fwnode.patch b/target/linux/generic/pending-6.6/450-09-block-partitions-populate-fwnode.patch
index 2e5bcac65309..9a48d7502fb5 100644
--- a/target/linux/generic/pending-6.6/450-09-block-partitions-populate-fwnode.patch
+++ b/target/linux/generic/pending-6.6/450-09-block-partitions-populate-fwnode.patch
@@ -34,7 +34,7 @@ Signed-off-by: Daniel Golle <daniel@makrotopia.org>
 +	u32 partno;
 +
 +	fw_parts = device_get_named_child_node(ddev, "partitions");
-+	if (!fw_parts)
++	if (!fw_parts && ddev->parent)
 +		fw_parts = device_get_named_child_node(ddev->parent, "partitions");
 +
 +	if (!fw_parts)
```

dangowrt, 2024-06-15:
Interesting. I'll have a look now.

commit 94e01900 (2024-06-16): generic: 6.6: update NVMEM-on-MMC series

Allow a wider audience to test this pending series.

Use about to be submitted v3 which factors out block notification support.
Apart from dropping the no longer needed (and problematic) fallback for
for the 'partitions' node being present at the device parent there are
no intended functional changes.

As opening a block device as file is not supported yet in Kernel v6.6,
use the previous method as backporting seems a bit too involving.

Fixes: #15642
closed 2024-06-16 by openwrt-bot