openwrt/openwrt #15643: ramps: eth1 fails to initialise on Edgerouter X

darkxst, 2024-06-06:
### Describe the bug

I just updated my er-x build to latest main and eth1 has stopped working. This would seem to be a regression in the past few weeks and I do notice some patches around ethernet support on rampis.

### OpenWrt version

r26463+81-a1a9572f43

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Ubiquiti EdgeRouter X

### Image kind

Self-built image

### Steps to reproduce

1. sysupgrade to snapshot at commit c67d5189a405
2. reboot

### Actual behaviour

eth1 is missing, kernel logs show:
```
Thu Jun  6 20:40:48 2024 kern.info kernel: [    3.233498] mt7530-mdio mdio-bus:1f: MT7530 adapts as multi-chip module
Thu Jun  6 20:40:48 2024 kern.info kernel: [    3.273510] mt7530-mdio mdio-bus:1f: configuring for fixed/rgmii link mode
Thu Jun  6 20:40:48 2024 kern.info kernel: [    3.288355] mt7530-mdio mdio-bus:1f: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Jun  6 20:40:48 2024 kern.info kernel: [    3.293786] mt7530-mdio mdio-bus:1f eth1 (uninitialized): PHY [mt7530-0:01] driver [MediaTek MT7530 PHY] (irq=21)
Thu Jun  6 20:40:48 2024 kern.err kernel: [    3.324660] mtk_soc_eth 1e100000.ethernet eth0: error -17 registering interface eth1
```

### Expected behaviour

eth1 should exist and be part of lan

### Additional info

My custom build only includes this PR https://github.com/openwrt/openwrt/pull/15194, to allow 6.6 kernel no other customisation.

### Diffconfig

```text
./scripts/diffconfig.sh

CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_ubnt_edgerouter-x=y
CONFIG_GETDNS_ENABLE_STUB_ONLY=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_MBEDTLS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_curl=m
CONFIG_PACKAGE_ddns-scripts=m
CONFIG_PACKAGE_ddns-scripts-cloudflare=m
CONFIG_PACKAGE_ddns-scripts-services=m
CONFIG_PACKAGE_getdns=m
CONFIG_PACKAGE_kmod-crypto-chacha20poly1305=m
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=m
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=m
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=m
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=m
CONFIG_PACKAGE_kmod-udptunnel4=m
CONFIG_PACKAGE_kmod-udptunnel6=m
CONFIG_PACKAGE_kmod-wireguard=m
CONFIG_PACKAGE_libatomic=m
CONFIG_PACKAGE_libcurl=m
CONFIG_PACKAGE_liblua=m
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=m
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libnghttp2=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_libubus-lua=m
CONFIG_PACKAGE_libyaml=m
CONFIG_PACKAGE_lua=m
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-ddns=m
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=m
CONFIG_PACKAGE_luci-lib-ip=m
CONFIG_PACKAGE_luci-lib-jsonc=m
CONFIG_PACKAGE_luci-lib-nixio=m
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-lua-runtime=m
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-wireguard=m
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_stubby=m
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-lua=m
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wireguard-tools=m
# CONFIG_PACKAGE_kmod-crypto-kpp is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john-tho, 2024-06-06:
Could we please get more `dmesg` output?
Also would like to see `ip l` output.

> `Thu Jun  6 20:40:48 2024 kern.err kernel: [    3.324660] mtk_soc_eth 1e100000.ethernet eth0: error -17 registering interface eth1`

from: https://elixir.bootlin.com/linux/v6.6.32/source/net/dsa/slave.c#L2678, error is `#define	EEXIST		17	/* File exists */`

I think this happens because one of the DSA ports labels (`eth1`) for this device collides with an (already) automatically allocated device name:

@dangowrt some blowback from removing our device name hack…
Guessing here:
Following https://github.com/openwrt/openwrt/commit/2a25c6ace8d833cf491a66846a0b9e7c5387b8f0:
- `mtk_soc_eth` no longer registers gmac0 with it's label name `dsa`, and instead gets automatic name `eth%d`. First device gets `eth0`.
- gmac1 gets automatic name `eth1`
- DSA port switch port 1@gmac0 has DSA label `eth1`, which does get applied [`dsa.c`](https://elixir.bootlin.com/linux/v6.6.32/source/net/dsa/dsa.c#L1245), but collides with existing netdev name.

Not sure the preinit script would deal with duplicate netdev names anyway.
Workaround options:
- revert 2a25c6ace8d8
- change DTB DSA port labels to names that do not collide (but migrating OpenWrt netdev names in sysupgrade configs nightmare)
- patch DSA to not use label if netdev name collision (potentially up-streamable)

Drivers at upstream (OpenWrt modifies, so I grep build_dir): [`mtk_eth_soc`](https://elixir.bootlin.com/linux/v6.6.32/source/drivers/net/ethernet/mediatek/mtk_eth_soc.c)  [`mt7530`](https://elixir.bootlin.com/linux/v6.6.32/source/drivers/net/dsa/mt7530-mdio.c)

DTS:
- https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/dts/mt7621.dtsi
- https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
- https://github.com/openwrt/openwrt/blob/main/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts

darkxst, 2024-06-07:
```
root@OpenWrt:/# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
    link/ether ac:8b:a9:a4:56:a0 brd ff:ff:ff:ff:ff:ff
3: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether ac:8b:a9:a4:56:a0 brd ff:ff:ff:ff:ff:ff
4: eth2@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether ac:8b:a9:a4:56:a2 brd ff:ff:ff:ff:ff:ff
5: eth3@dsa: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 10
    link/ether ac:8b:a9:a4:56:a3 brd ff:ff:ff:ff:ff:ff
6: eth4@dsa: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 10
    link/ether ac:8b:a9:a4:56:a4 brd ff:ff:ff:ff:ff:ff
7: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether ac:8b:a9:a4:56:a2 brd ff:ff:ff:ff:ff:ff
```

dangowrt, 2024-06-07:
> patch DSA to not use label if netdev name collision (potentially up-streamable): dirty example:

I think instead of `dev_net(master)` you should just point to `&init_net`, but apart from that this looks like the best possible solution to me, and it's reasonable enough for upstream to deal with it. Would you suggest it to netdev mailing list please?

john-tho, 2024-06-26:
Alternate workaround adding a port level (to minimise patch delta—would prefer switch level) DTB property to prevent DSA applying ifname: patch not refreshed. I can PR if this _looks_ acceptable. I could also send it upstream for comment in reply to my previous submission, if we want to poke the bear again (probably should), even if it is just to hear _don't use label_ again.
```patch
commit dff534e43713f302d959337b3c8a33f2b709d4c5
Author: John Thomson <git@johnthomson.fastmail.com.au>
Date:   Sun Jun 23 09:48:33 2024 +1000

    ramips: workaround DSA ifname collision
    
    Signed-off-by: John Thomson <git@johnthomson.fastmail.com.au>

diff --git a/target/linux/ramips/dts/mt7621_tplink_deco-m4r-v4.dts b/target/linux/ramips/dts/mt7621_tplink_deco-m4r-v4.dts
index 9f30f935509a..246f085bb0e7 100644
--- a/target/linux/ramips/dts/mt7621_tplink_deco-m4r-v4.dts
+++ b/target/linux/ramips/dts/mt7621_tplink_deco-m4r-v4.dts
@@ -189,6 +189,7 @@
 		port@0 {
 			status = "okay";
 			label = "eth0";
+			disable-kernel-ifname;
 		};
 
 		port@1 {
diff --git a/target/linux/ramips/dts/mt7621_tplink_er605-v2.dts b/target/linux/ramips/dts/mt7621_tplink_er605-v2.dts
index 1299b02806ed..699d0513d7d3 100644
--- a/target/linux/ramips/dts/mt7621_tplink_er605-v2.dts
+++ b/target/linux/ramips/dts/mt7621_tplink_er605-v2.dts
@@ -98,6 +98,7 @@
 		port@1 {
 			status = "okay";
 			label = "eth1";
+			disable-kernel-ifname;
 		};
 
 		port@2 {
diff --git a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
index 0902ad56b961..bfc670f03953 100644
--- a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
+++ b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
@@ -30,6 +30,7 @@
 		port@0 {
 			status = "okay";
 			label = "eth0";
+			disable-kernel-ifname;
 		};
 
 		port@1 {
diff --git a/target/linux/ramips/patches-6.6/600-net-dsa-allow-disable-kernel-ifname-via-label.patch b/target/linux/ramips/patches-6.6/600-net-dsa-allow-disable-kernel-ifname-via-label.patch
new file mode 100644
index 000000000000..6b89e6fbe68d
--- /dev/null
+++ b/target/linux/ramips/patches-6.6/600-net-dsa-allow-disable-kernel-ifname-via-label.patch
@@ -0,0 +1,33 @@
+From b6ddb010455569cd02dc6ada92e9c805d739161c Mon Sep 17 00:00:00 2001
+From: John Thomson <git@johnthomson.fastmail.com.au>
+Date: Sun, 23 Jun 2024 09:25:33 +1000
+Subject: [PATCH] net: dsa: allow disable kernel ifname via label
+
+With only DSA supporting network interface names set by DSA label,
+an interface can be brought up (before DSA port) with an autogenerated
+name that clashes with a DSA label. DSA then refuses to bring up that port.
+This DTS port label is the simplest place to detail intended device names,
+so introduce a port level property which skips DSA applying the ifname:
+disable-kernel-ifname;
+
+Link: https://lore.kernel.org/netdev/20240608014724.2541990-1-git@johnthomson.fastmail.com.au/
+
+Signed-off-by: John Thomson <git@johnthomson.fastmail.com.au>
+---
+ net/dsa/dsa.c | 5 ++++-
+ 1 file changed, 4 insertions(+), 1 deletion(-)
+
+--- a/net/dsa/dsa.c
++++ b/net/dsa/dsa.c
+@@ -1263,7 +1263,10 @@ static int dsa_port_parse_of(struct dsa_
+ 	if (link)
+ 		return dsa_port_parse_dsa(dp);
+ 
+-	return dsa_port_parse_user(dp, name);
++	if (!of_property_read_bool(dn, "disable-kernel-ifname"))
++		return dsa_port_parse_user(dp, name);
++	else
++		return dsa_port_parse_user(dp, NULL);
+ }
+ 
+ static int dsa_switch_parse_ports_of(struct dsa_switch *ds,
```

carlo2007, 2024-07-25:
> Workaround options:
> 
>     * revert [2a25c6a](https://github.com/openwrt/openwrt/commit/2a25c6ace8d833cf491a66846a0b9e7c5387b8f0)

I understand the DSA driver issue of naming interfaces but if the bug was introduced with the commit [2a25c6a](https://github.com/openwrt/openwrt/commit/2a25c6ace8d833cf491a66846a0b9e7c5387b8f0), at the moment i think it's better to revert the commit instead of loose the support for this device.

This open issue is blocking the [15194](https://github.com/openwrt/openwrt/pull/15194) pull request necessary to restore device support which is currently lost.

dangowrt, 2024-07-25:
Upstream suggested introducing a downstream property `openwrt,netdev-name` to keep handling this in userspace. Imho this sounds acceptable and doesn't need any kernel changes, and still allows for dropping the downstream patch.

ptpt52, 2024-08-23:
this issue related to the commit https://github.com/openwrt/openwrt/commit/2a25c6ace8d833cf491a66846a0b9e7c5387b8f0 ?
and I have a workaroud for it

ptpt52, 2024-08-23:
```diff
diff --git a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x-sfp.dts b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x-sfp.dts
index 50a77c3877a..f619747d70c 100644
--- a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x-sfp.dts
+++ b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x-sfp.dts
@@ -67,6 +67,10 @@
 
 &switch0 {
 	ports {
+		port@0 {
+			label = "rename_eth0";
+		};
+
 		port@5 {
 			reg = <5>;
 			label = "eth5";
diff --git a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
index 80467c88e98..5fdbb6a6ba6 100644
--- a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
+++ b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dts
@@ -7,7 +7,7 @@
 
 &gmac1 {
 	status = "okay";
-	label = "eth0";
+	label = "rename_eth0";
 	phy-handle = <&ethphy0>;
 
 	nvmem-cells = <&macaddr_factory_22 0>;
diff --git a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
index 0902ad56b96..7dcd4c546a2 100644
--- a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
+++ b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
@@ -34,7 +34,7 @@
 
 		port@1 {
 			status = "okay";
-			label = "eth1";
+			label = "rename_eth1";
 			nvmem-cells = <&macaddr_factory_22 1>;
 			nvmem-cell-names = "mac-address";
 		};
diff --git a/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label b/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label
index 110e023b962..9d7c5a4dad8 100644
--- a/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label
+++ b/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label
@@ -10,6 +10,11 @@ set_netdev_labels() {
 		[ "$netdev" = "$label" ] && continue
 		ip link set "$netdev" name "$label"
 	done
+
+	for dir in /sys/class/net/rename_*; do
+		netdev="${dir##*/}"
+		test -e "$dir" && ip link set "$netdev" name "${netdev##rename_}"
+	done
 }
 
 boot_hook_add preinit_main set_netdev_labels

```

ptpt52, 2024-08-23:
or simple:
```diff
diff --git a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
index 0902ad56b96..713cce4a89d 100644
--- a/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
+++ b/target/linux/ramips/dts/mt7621_ubnt_edgerouter-x.dtsi
@@ -29,12 +29,12 @@
 	ports {
 		port@0 {
 			status = "okay";
-			label = "eth0";
+			label = "rename_eth0";
 		};
 
 		port@1 {
 			status = "okay";
-			label = "eth1";
+			label = "rename_eth1";
 			nvmem-cells = <&macaddr_factory_22 1>;
 			nvmem-cell-names = "mac-address";
 		};
diff --git a/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label b/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label
index 110e023b962..9d7c5a4dad8 100644
--- a/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label
+++ b/target/linux/ramips/mt7621/base-files/lib/preinit/04_set_netdev_label
@@ -10,6 +10,11 @@ set_netdev_labels() {
 		[ "$netdev" = "$label" ] && continue
 		ip link set "$netdev" name "$label"
 	done
+
+	for dir in /sys/class/net/rename_*; do
+		netdev="${dir##*/}"
+		test -e "$dir" && ip link set "$netdev" name "${netdev##rename_}"
+	done
 }
 
 boot_hook_add preinit_main set_netdev_labels
```

dangowrt, 2024-08-23:
Let's use another property name in device tree, such as `openwrt,netdev-name` like upstream reviewer Vladimir Oltean [has suggested](https://lore.kernel.org/netdev/20240709124503.pubki5nwjfbedhhy@skbuf/). For starters we can simply support both, renaming according to `label` as well as `openwrt,netdev-name`. In the long run we should abandon using `label` for renaming in userspace and only use `openwrt,netdev-name`.

ptpt52, 2024-08-23:
I would like to change the scripts and dts instead of changing the C code

namiltd, 2024-08-23:
```
 	ports {
 		port@0 {
 			status = "okay";
-			label = "eth0";
+			openwrt,netdev-name = "eth0";
 		};
 
 		port@1 {
 			status = "okay";
-			label = "eth1";
+			openwrt,netdev-name = "eth1";
 			nvmem-cells = <&macaddr_factory_22 1>;
 			nvmem-cell-names = "mac-address";
```

dangowrt, 2024-08-23:
> I would like to change the scripts and dts instead of changing the C code

That's exactly what has been suggested upstream. To handle this with a *custom downstream property* and *in a script in userland*, avoiding having to change any C code.

LynxChaus, 2024-08-24:
Guys, you're fighting with windmills.

Just rename "eth[0-5]" in mt7621_ubnt_edgerouter-x.dtsi to "lan[0-5]"/"swp[0-5]"/"enet[0-5]" and forget about this issue. 
Instead, you try to write "yet another incompatible with upstream specific owrt hack" and re-apply it on every kernel upgrade. 
For what?

darkxst, 2024-11-09:
> Just rename "eth[0-5]" in mt7621_ubnt_edgerouter-x.dtsi to "lan[0-5]"/"swp[0-5]"/"enet[0-5]" and forget about this issue.

Given the new partition layout that just landed in main https://github.com/openwrt/openwrt/pull/15194, requires a fresh reinstall anyway, seems like a good chance to simply rename without any issues?

I will create a PR

PR #17062 (2024-11-25): ramips: mt7621: support openwrt,netdev-name for renaming interfaces
Edgerouter X currently has its eth1 port on the switch missing since there is a naming conflict currently.

So, as the root cause is mixing kernel support for DSA interfaces having predictable names set via "label" property vs others having it assigned dynamically lets avoid the conflict by using our own custom property as suggested upstream [1].

So, add support via "openwrt,netdev-name" property and use it on ERX.

Fixes: 2a25c6ace8d8 ("ramips: get rid of downstream network device label patch")
Fixes: #15643
commit 141ccfd5 (2024-11-25): ramips: mt7621: support openwrt,netdev-name for renaming interfaces

Edgerouter X currently has its eth1 port on the switch missing since there
is a naming conflict currently.

So, as the root cause is mixing kernel support for DSA interfaces having
predictable names set via "label" property vs others having it assigned
dynamically lets avoid the conflict by using our own custom property as
suggested upstream [1].

So, add support via "openwrt,netdev-name" property and use it on ERX.

Fixes: 2a25c6ace8d8 ("ramips: get rid of downstream network device label patch")
Fixes: #15643
Link: https://g [... cut]