openwrt/openwrt #15644: qualcommax: NBG7815: tftpbooted initramfs images does not working properly

pwned-pixel, 2024-06-06:
### Describe the bug

The latest snapshots for this device is broken somehow. I've tftpbooted the last snapshot to recover mmc partitions. The gdisk/sgdisk tool didn't work due to this error (they were just segfaulting).

This only happens booting initramfs image through tftp. I didn't try to tftpboot a stable branch image yet nor did I install a snapshot image yet. Because my own installed builds are working properly. So I think this is soley an initramfs issue.

The initramfs image is throwing out (I can provide full log if needed):
The last part of the log looks familiar to me from a bug apk packaging related we had in the past.

`[    0.000000] Linux version 6.1.33 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23375-cdfcac6e24) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Wed Jun 14 21:27:29 2023`

```
[    1.393181] ------------[ cut here ]------------
[    1.398577] mmcblk0: mmc0:0001 M62704 3.53 GiB
[    1.403276] gcc_usb1_master_clk status stuck at 'on'
[    1.403295] WARNING: CPU: 2 PID: 1 at clk_branch_wait+0x11c/0x130
[    1.408783]  mmcblk0: p1 p2 p3 p4 p5 p6 p7 p8 p9 p10 p11
[    1.412131] Modules linked in:
[    1.418172] mmcblk0boot0: mmc0:0001 M62704 2.00 MiB
[    1.423332] CPU: 2 PID: 1 Comm: swapper/0 Not tainted 6.1.33 #0
[    1.429188] mmcblk0boot1: mmc0:0001 M62704 2.00 MiB
[    1.431577] Hardware name: Zyxel NBG7815 (DT)
[    1.431580] pstate: 604000c5 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    1.436971] mmcblk0rpmb: mmc0:0001 M62704 512 KiB, chardev (246:0)
[    1.442340] pc : clk_branch_wait+0x11c/0x130
[    1.464821] lr : clk_branch_wait+0x11c/0x130
[    1.469248] sp : ffffffc0093fbc70
[    1.473499] x29: ffffffc0093fbc70 x28: 0000000000000000 x27: ffffffc008a500c8
[    1.476719] x26: ffffffc008aa6068 x25: ffffffc008aa6078 x24: ffffffc008a355a0
[    1.483837] x23: ffffffc0089afc68 x22: 0000000000000000 x21: ffffffc008409c50
[    1.490955] x20: 0000000000000000 x19: ffffffc009354708 x18: fffffffffffe0000
[    1.498074] x17: ffffffc00935ce20 x16: ffffffc00935cdb0 x15: fffffffffffc3270
[    1.505192] x14: 0000000000000000 x13: 6f27207461206b63 x12: 7574732073757461
[    1.512309] x11: 0000000000000001 x10: 0000000000000027 x9 : ffffffc0093fbc70
[    1.519428] x8 : ffffffc0093203c0 x7 : 0000000000000000 x6 : 00000000fffff0b7
[    1.526545] x5 : 40000000fffff0b7 x4 : 0000000000000000 x3 : 0000000000000000
[    1.533664] x2 : ffffffc0092c71f0 x1 : ffffffc0092c71f0 x0 : 0000000000000028
[    1.540782] Call trace:
[    1.547891]  clk_branch_wait+0x11c/0x130
[    1.550151]  clk_branch2_disable+0x2c/0x40
[    1.554317]  clk_disable_unused_subtree+0xa8/0x108
[    1.558225]  clk_disable_unused_subtree+0x2c/0x108
[    1.562999]  clk_disable_unused_subtree+0x2c/0x108
[    1.567773]  clk_disable_unused_subtree+0x2c/0x108
[    1.572548]  clk_disable_unused_subtree+0x2c/0x108
[    1.577322]  clk_disable_unused+0x50/0xd8
[    1.582096]  do_one_initcall+0x50/0x1d0
[    1.586173]  kernel_init_freeable+0x234/0x29c
[    1.589821]  kernel_init+0x28/0x130
[    1.594333]  ret_from_fork+0x10/0x20
[    1.597633] ---[ end trace 0000000000000000 ]---

[    1.601797] ------------[ cut here ]------------
[    1.606055] gcc_usb0_master_clk status stuck at 'on'
[    1.606073] WARNING: CPU: 2 PID: 1 at clk_branch_wait+0x11c/0x130
[    1.615606] Modules linked in:
[    1.621591] CPU: 2 PID: 1 Comm: swapper/0 Tainted: G        W          6.1.33 #0
[    1.624550] Hardware name: Zyxel NBG7815 (DT)
[    1.632096] pstate: 604000c5 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    1.636354] pc : clk_branch_wait+0x11c/0x130
[    1.643120] lr : clk_branch_wait+0x11c/0x130
[    1.647633] sp : ffffffc0093fbc70
[    1.651885] x29: ffffffc0093fbc70 x28: 0000000000000000 x27: ffffffc008a500c8
[    1.655105] x26: ffffffc008aa6068 x25: ffffffc008aa6078 x24: ffffffc008a355a0
[    1.662223] x23: ffffffc0089afd20 x22: 0000000000000000 x21: ffffffc008409c50
[    1.669340] x20: 0000000000000000 x19: ffffffc009354a88 x18: 00000000000000dd
[    1.676458] x17: ffffffc00935ce20 x16: ffffffc00935cdb0 x15: ffffffc0092c7298
[    1.683577] x14: 0000000000000297 x13: 00000000000000dd x12: 00000000ffffffea
[    1.690694] x11: 00000000ffffefff x10: ffffffc00931f298 x9 : ffffffc0092c7240
[    1.697813] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000057fa8
[    1.704931] x5 : 0000000000000fff x4 : 0000000000000000 x3 : ffffffc0093fba30
[    1.712049] x2 : ffffffc0092c71f0 x1 : ffffffc0092c71f0 x0 : 0000000000000028
[    1.719167] Call trace:
[    1.726275]  clk_branch_wait+0x11c/0x130
[    1.728535]  clk_branch2_disable+0x2c/0x40
[    1.732703]  clk_disable_unused_subtree+0xa8/0x108
[    1.736611]  clk_disable_unused_subtree+0x2c/0x108
[    1.741384]  clk_disable_unused_subtree+0x2c/0x108
[    1.746158]  clk_disable_unused_subtree+0x2c/0x108
[    1.750934]  clk_disable_unused_subtree+0x2c/0x108
[    1.755707]  clk_disable_unused+0x50/0xd8
[    1.760480]  do_one_initcall+0x50/0x1d0
[    1.764559]  kernel_init_freeable+0x234/0x29c
[    1.768207]  kernel_init+0x28/0x130
[    1.772718]  ret_from_fork+0x10/0x20
[    1.776018] ---[ end trace 0000000000000000 ]---
```
```
[   13.684279] ath11k c000000.wifi: ipq8074 hw2.0
[   13.684312] ath11k c000000.wifi: FW memory mode: 0
[   13.688282] remoteproc remoteproc0: powering up cd00000.q6v5_wcss
[   13.692872] remoteproc remoteproc0: Booting fw image IPQ8074/q6_fw.mdt, size 668
[   14.043894] remoteproc remoteproc0: remote processor cd00000.q6v5_wcss is now up
[   14.047274] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.087850] ath11k c000000.wifi: qmi ignore invalid mem req type 3
[   14.095303] ath11k c000000.wifi: chip_id 0x0 chip_family 0x0 board_id 0xff soc_id 0xffffffff
[   14.095339] ath11k c000000.wifi: fw_version 0x290c84a5 fw_build_timestamp 2023-03-25 07:34 fw_build_id WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
[   14.167490] ath11k c000000.wifi: failed to fetch board data for bus=ahb,qmi-chip-id=0,qmi-board-id=255,variant=Zyxel-NBG7815 from ath11k/IPQ8074/hw2.0/board-2.bin
[   14.167553] ath11k c000000.wifi: failed to fetch board data for bus=ahb,qmi-chip-id=0,qmi-board-id=255 from ath11k/IPQ8074/hw2.0/board-2.bin
[   14.180975] ath11k c000000.wifi: failed to fetch board.bin from IPQ8074/hw2.0
[   14.193692] ath11k c000000.wifi: qmi failed to fetch board file: -12
[   14.200707] ath11k c000000.wifi: failed to load board data file: -12
```

### OpenWrt version

r23375-cdfcac6e24

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Zyxel NBG7815

### Image kind

Official downloaded image

### Steps to reproduce

Download latest snapshot.
Tftpboot the snapshot.
System is unstable.

### Actual behaviour

The system is not working properly. E. g. starting apps (installed via opkg) like gdisk/sgdisk dying with segfaults. Others are working (e. g. parted) fortunately.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pwned-pixel, 2024-06-06:
If I look at this `gcc_usb1_master_clk status stuck at 'on'` it seems to me there is sth. wrong with usb. While usb can be handy for any recovery attempts I would rather want a stable recovery environment. Would it help to strip out usb from dts and/or from kernel config?

My own build itb file is stucking on decompressing the kernel with reboot. Don't know why. Does the size matter? I guess so.

robimarko, 2024-06-06:
Just ignore those, they dont do impact anything.

pwned-pixel, 2024-06-06:
Hmm. But what is cuasing then the segfault of gdisk/sdisk. There is nothing within the logs if I call this binaries. Am I right if I then assume it is an issue related to this programs?

robimarko, 2024-06-06:
That I cannot comment on, just on the USB clock warnings

pwned-pixel, 2024-06-07:
Well I will close this. Because this looks now more like a bug of gptfdisk.

closed 2024-06-07 by pwned-pixel