openwrt/openwrt #15656: apk:

Ra2-IFV, 2024-06-08:
### Describe the bug

```
/openwrt/staging_dir/host/bin/fakeroot /openwrt/staging_dir/host/bin/apk mkpkg --info "name:luci-app-cloudflared" --info "version:1.2-1" --info "description:" --info "arch:all" --info "license:Apache-2.0" --info "origin:feeds/luci/applications/luci-app-cloudflared" --info "provides:" --script "post-install:/openwrt/build_dir/target-aarch64_generic_musl/luci-app-cloudflared/apk-all/luci-app-cloudflared/post-install" --script "pre-deinstall:/openwrt/build_dir/target-aarch64_generic_musl/luci-app-cloudflared/apk-all/luci-app-cloudflared/pre-deinstall" --info "depends:libc cloudflared" --files "/openwrt/build_dir/target-aarch64_generic_musl/luci-app-cloudflared/ipkg-all/luci-app-cloudflared" --output "/openwrt/bin/packages/aarch64_generic/luci/luci-app-cloudflared-1.2-1.apk" --sign "/openwrt/private-key.pem"
ERROR: info field 'version' has invalid value: package version is invalid
ERROR: failed to create package: /openwrt/bin/packages/aarch64_generic/luci/luci-app-cloudflared-1.2-1.apk: package version is invalid
make[3]: *** [../../luci.mk:375: /openwrt/bin/packages/aarch64_generic/luci/luci-app-cloudflared-1.2-1.apk] Error 99
make[3]: Leaving directory '/openwrt/feeds/luci/applications/luci-app-cloudflared'
time: package/feeds/luci/luci-app-cloudflared/compile#0.41#0.23#0.59
    ERROR: package/feeds/luci/luci-app-cloudflared failed to build.
make[2]: *** [package/Makefile:173: package/feeds/luci/luci-app-cloudflared/compile] Error 1
make[2]: Leaving directory '/openwrt'
make[1]: *** [package/Makefile:167: /openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt'
make: *** [/openwrt/include/toplevel.mk:233: world] Error 2
```

### OpenWrt version

3de3c2bdfa6b757a1ac9fa4d9905e0d268d98e5f

### OpenWrt release

3de3c2bdfa6b757a1ac9fa4d9905e0d268d98e5f

### OpenWrt target/subtarget

aarch64 generic

### Device

no

### Image kind

Self-built image

### Steps to reproduce

run make -j1 V=s

### Actual behaviour

Failing to build. 

### Expected behaviour

Should pass. the version name followed the semver rules, so the error quite strange

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ra2-IFV, 2024-06-08:
Another package that was having the same issue is `luci-app-adblock-fast`, In Makefile `PKG_VERSION:=1.1.2`  
Pretty weird...

Ra2-IFV, 2024-06-09:
Does someone know the name syntax? So that I can create a simple PR to fix them all

CHKDSK88, 2024-06-10:
Here you can find scheme:
https://github.com/openwrt/openwrt/pull/14918

@aparcar is apk specialist.

aparcar, 2024-06-10:
APK expects an `r` before the revision, so the command you're trying `--info "version:1.2-1"` fails. It doesn't understand the trailing `-1`. Within the build system we switched version handling so revisions receive a prefixed `r`. The correct version would be `1.2-r1`

Ra2-IFV, 2024-06-10:
Understood. Thanks for explaining

closed 2024-06-10 by Ra2-IFV

Ra2-IFV, 2024-06-10:
So we are going to format `PKG_VERSION` in all the packages' Makefile, to support the new package manager, called `apk`?

aparcar, 2024-06-10:
Please just keep using `PKG_RELEASE` and `PKG_VERSION` and the build system does the magic itself.

Ra2-IFV, 2024-06-10:
Many thanks.