openwrt/openwrt #15658: Exception stack in dmesg during R7800 booting (snapshot kernel 6.6 with mainline ath10 firmware/driver)

vochong, 2024-06-08:
### Describe the bug

If the mainline (non-CT) ath10k driver/firmware are used for R7800 (master snapshot with kernel 6.6), the following dump is seen in dmesg after every booting.     However, WiFI works properly in spite of this issue.

```
{
	"kernel": "6.6.32",
	"hostname": "xxx",
	"system": "ARMv7 Processor rev 0 (v7l)",
	"model": "Netgear Nighthawk X4S R7800",
	"board_name": "netgear,r7800",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r26551-b9650de249",
		"target": "ipq806x/generic",
		"description": "OpenWrt SNAPSHOT r26551-b9650de249"
	}
}
```

```
[  124.700224] ------------[ cut here ]------------
[  124.700260] WARNING: CPU: 0 PID: 5716 at backports-6.6.15/drivers/net/wireless/ath/ath10k/wmi.c:3630 ath10k_wmi_event_host_swba+0x68c/0x6ec [ath10k_core]
[  124.703933] memcpy: detected field-spanning write (size 64) of single field "tim->virtual_map" at backports-6.6.15/drivers/net/wireless/ath/ath10k/wmi.c:3630 (size 1)
[  124.717651] Modules linked in: xt_connlimit qcserial pppoe ppp_async option nft_fib_inet nf_flow_table_inet nf_conncount l2tp_ppp ipw ath10k_pci(O) ath10k_core(O) ath(O) xt_state xt_helper xt_conntrack xt_connmark xt_connbytes wireguard usb_wwan usb_serial_simple ti_usb_3410_5052 sierra pppox ppp_generic pl2303 oti6858 nft_xfrm nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_compat nft_chain_nat nf_tables nf_nat_ftp nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack_ftp nf_conntrack mos7840 mos7720 mct_u232 mac80211(O) libchacha20poly1305 keyspan ipt_REJECT io_edgeport garmin_gps ftdi_sio cypress_m8 curve25519_neon cp210x ch341 cfg80211(O) belkin_sa ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_policy xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_esp xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY xfrm_interface visor
[  124.718011]  usbserial usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc sch_cake poly1305_arm nfnetlink nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcurve25519_generic libcrc32c iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables ezusb dmx_usb(O) crc_ccitt compat(O) chacha_neon fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport ppdev parport cryptodev(O) ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 ksmbd nls_ucs2_utils cifs_arc4 asn1_decoder nfs msdos ifb ip6_vti ip_vti sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel oid_registry netlink_diag tunnel6 tunnel4 ip_tunnel tun snd_rawmidi snd_seq_device snd_pcm_oss snd_mixer_oss snd_hwdep snd_compress snd_pcm snd_timer snd soundcore ovpn_dco_v2(O) xfrm_user xfrm_ipcomp af_key
[  124.800449]  xfrm_algo lockd sunrpc grace hfsplus hfs cdrom dns_resolver nls_utf8 nls_iso8859_15 nls_iso8859_1 nls_cp850 nls_cp437 nls_cp1250 vxlan macsec geneve udp_tunnel ip6_udp_tunnel wp512 twofish_generic twofish_common tea serpent_generic khazad cast6_generic cast5_generic cast_common camellia_generic blowfish_generic blowfish_common anubis xts crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_arm sha1_generic sha1_arm seqiv echainiv geniv ecb des_generic libdes cmac chacha20poly1305 authencesn authenc uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom uhci_hcd ohci_platform ohci_hcd phy_qcom_ipq806x_usb ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod scsi_common ehci_hcd ramoops reed_solomon pstore gpio_button_hotplug(O) vfat fat f2fs ext4 mbcache jbd2 exfat dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax tpm crc32c_generic crc32_generic cbc encrypted_keys trusted
[  124.976096] CPU: 0 PID: 5716 Comm: nft Tainted: G           O       6.6.32 #0
[  124.998268] Hardware name: Generic DT based system
[  125.005303]  unwind_backtrace from show_stack+0x10/0x14
[  125.009988]  show_stack from dump_stack_lvl+0x40/0x4c
[  125.015109]  dump_stack_lvl from __warn+0x94/0xbc
[  125.020318]  __warn from warn_slowpath_fmt+0xf8/0x15c
[  125.025003]  warn_slowpath_fmt from ath10k_wmi_event_host_swba+0x68c/0x6ec [ath10k_core]
[  125.030121]  ath10k_wmi_event_host_swba [ath10k_core] from ath10k_wmi_10_4_op_rx+0x418/0x56c [ath10k_core]
[  125.038205]  ath10k_wmi_10_4_op_rx [ath10k_core] from ath10k_htc_rx_completion_handler+0x104/0x160 [ath10k_core]
[  125.047668]  ath10k_htc_rx_completion_handler [ath10k_core] from ath10k_pci_fw_dump_work+0x4f0/0x538 [ath10k_pci]
[  125.058015]  ath10k_pci_fw_dump_work [ath10k_pci] from ath10k_ce_per_engine_service+0x64/0x84 [ath10k_core]
[  125.068154]  ath10k_ce_per_engine_service [ath10k_core] from ath10k_ce_per_engine_service_any+0x74/0x194 [ath10k_core]
[  125.077703]  ath10k_ce_per_engine_service_any [ath10k_core] from ath10k_pci_napi_poll+0x40/0x118 [ath10k_pci]
[  125.088465]  ath10k_pci_napi_poll [ath10k_pci] from __napi_poll.constprop.0+0x2c/0x180
[  125.098446]  __napi_poll.constprop.0 from net_rx_action+0x140/0x2e8
[  125.106256]  net_rx_action from __do_softirq+0x100/0x270
[  125.112418]  __do_softirq from irq_exit+0x88/0xb4
[  125.117972]  irq_exit from call_with_stack+0x18/0x20
[  125.122573]  call_with_stack from __irq_usr+0x78/0xa0
[  125.127607] Exception stack(0xc8523fb0 to 0xc8523ff8)
[  125.132557] 3fa0:                                     b56200d8 b56200dc b6e1bb44 ffffffff
[  125.137596] 3fc0: b56200d8 00000002 b56200dc b6e1bb44 0000000d b6ee5080 b562a778 fffffffc
[  125.145754] 3fe0: 3cf7369a beb5b9c8 b6e1bb7c b6ee4de8 a0000010 ffffffff
[  125.153962] ---[ end trace 0000000000000000 ]---
[  125.160930] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[  125.165817] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
```

### OpenWrt version

r26551-b9650de249

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq806x/generic

### Device

NETGEAR Nighthawk X4S R7800

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-06-08:
@Ansuel
@robimarko
@namiltd

namiltd, 2024-06-08:
Please try to ask a question/report a bug according to https://wireless.wiki.kernel.org/en/users/documentation/reporting_bugs

vochong, 2024-06-16:
I don't think adding (void *) will get rid of the warning.

[  124.703933] memcpy: detected field-spanning write (size 64) of single field "tim->virtual_map" at backports-6.6.15/drivers/net/wireless/ath/ath10k/wmi.c:3630 (size 1)

I think the problem lies in the fact that tim->virtual_map is declared with a size of 1 byte, but the memcpy operation is trying to copy 64 bytes into it. This discrepancy is the cause of the warning.   The length pvm_len should match the actual size of tim->virtual_map to avoid overwriting adjacent memory.

john-tho, 2024-06-27:
This has been detailed on linux-wireless; https://lore.kernel.org/linux-wireless/CAPh3n83zb1PwFBFijJKChBqY95zzpYh=2iPf8tmh=YTS6e3xPw@mail.gmail.com/, and was already fixed upstream: 2ae5c9248e06 ("wifi: mac80211: Use flexible array in struct ieee80211_tim_ie"): https://github.com/torvalds/linux/commit/2ae5c9248e06, but has not yet reached stable (requested ~yesterday) https://lore.kernel.org/stable/fc31dd6f-ec32-4911-921f-1f34e9ad2449@quicinc.com/

vochong, 2024-06-27:
@john-tho
Thanks a lot for the info.