openwrt/openwrt #15660: MBIM Cellular Pin Code Rejected

swvalenti194, 2024-06-08:
Hello, have latest openwrt build using Quectel RM520N-GL and have tried MBIM and QMI but keep getting the error on both "pin code rejected". I am using Verizon with apn ne01.vzwstatic

Any help would be appreciated.  Thanks

swvalenti194, 2024-06-10:
I contacted Verizon and no such pin code is associated with the SIM.

swvalenti194, 2024-06-10:
Tried AT&T with the same results.

robimarko, 2024-06-11:
Are you using ModemManager?

Cause, I am using RM520N-GL with MM and it works fine

swvalenti194, 2024-06-11:
I used MM and it fired up however a reboot dropped connection cause the modem kept getting a new Mac address.

robimarko, 2024-06-11:
MM is kind of the only option when it comes to 5G modems, why does the MAC address of the underlying interface matter to you?

swvalenti194, 2024-06-11:
It's frustrating cause I spent close to $600 and pay a monthly data plan...hoping someone can help.  Anyone take Venmo or PayPal?  HA!

ptpt52, 2024-06-17:
@swvalenti194  you may try x-wrt with QMAP Cellular support for you RM520N-GL, tell me what is your openwrt device model?

swvalenti194, 2024-06-17:
@ptpt52  I am using PRI 4b+ with lastest openwrt build. Hope that's what you mean and not sure I know about xwrt and qmap

swvalenti194, 2024-06-17:
@ptpt52 https://sixfab.com/product/sixfab-jumpstart-5g/

ptpt52, 2024-06-17:
@swvalenti194  I mean which openwrt firmware you are running?

ptpt52, 2024-06-17:
have you try the `QMI Cellular ` protocol

swvalenti194, 2024-06-17:
I have and that works but the speeds are brutally slow

ptpt52, 2024-06-17:
@swvalenti194  you have to switch to mbim mode via AT command to the modem?

swvalenti194, 2024-06-17:
Do you mind telling me how?  And for some weird reason only the at apn and restart command take. All others error out

ptpt52, 2024-06-17:
@swvalenti194 hmm, RM520N is not cap with mbim mode, it could be used as QMI mode. or use QMAP Cellular
what speed you get with QMI?

swvalenti194, 2024-06-17:
Maybe 5/1 it's unusable. Ha. On my firstnet phone I get 140/50. What is qmap?

ptpt52, 2024-06-17:
QMI could be 350~500 Mbps but it depends on the signal condition. 5Mbps low is really bug...

swvalenti194, 2024-06-17:
Will do thank you. I'll report back.

swvalenti194, 2024-06-17:
@ptpt52 none of my AT commands take except AT responds ok all others error out.  Can't get the modem in MBIM mode

swvalenti194, 2024-06-18:
@ptpt52 I was able to get the modem into mbim mode and connect to internet but it is still only grabbing LTE and slow speeds.  QMI was even worse. Any ideas?

ptpt52, 2024-06-18:
@swvalenti194  have you try the x-wrt firmware. for my case, mt7621 + RM520N with QMI get 200+Mbps

swvalenti194, 2024-06-18:
I'm running xwrt and switched to r520n gl as it's compatible with firstnet. Still only LTE and slow speeds. I don't get it. Even with qmi. Sorry to be a bother on this just at a loss.

ptpt52, 2024-06-18:
Maybe your 5G module has no antenna and the signal is not good?

swvalenti194, 2024-06-18:
Nah just checked and all 4 antenna leads are connected. Oh well. Thanks for the try.

azooz2007, 2024-10-22:
I don't know if you solve this issue or not, but I used to have smaller issue some AT commands not working probably, after upgrading my modem to latest using USB stick that support M.2 and following some guides on internet to figure which port and drivers then after upgrading the modem works on every mode with no issue even on openwrt. I think quectel are new on modems so you have to update sometimes if you are seeing other people has their modem working fine and yours showing some issue just like what happened to me on RM502Q-GL.

RoninTommy, 2026-09-10:
> Hello, have latest openwrt build using Quectel RM520N-GL and have tried MBIM and QMI but keep getting the error on both "pin code rejected". I am using Verizon with apn ne01.vzwstatic
> 
> Any help would be appreciated. Thanks

2026 Here's how I fixed this very issue

Fixed my ZBT Z8102AX v1 + Quectel RM520N-GL MBIM connection:

1. In the router's LuCI interface, set the 4G interface protocol to "MBIM Cellular".
2. In LuCI → Modem → AT Commands, check the modem USB mode:
   AT+QCFG="usbnet"
3. Mine returned:
   +QCFG: "usbnet",0
   This meant the RM520N-GL was still in QMI mode even though LuCI was configured for MBIM.
4. Changed it to MBIM:
   AT+QCFG="usbnet",2
   Wait for OK.
5. Restarted the RM520N-GL from AT Commands:
   AT+CFUN=1,1
6. Waited for the modem to reconnect.

After that, the 4G interface came back with a MAC address and RX/TX traffic. MBIM was working.

The "PIN code rejected", "no subscriber", and "MBIM failed to read modem caps" errors were occurring while the modem was still in the wrong USB mode. AT+CPIN? showed +CPIN: READY, so the SIM PIN wasn't actually the problem.