openwrt/openwrt #15662: Newer snapshots for Mediatek Filelogic seems to be broken!

UHeinz13, 2024-06-09:
### Describe the bug

I have a Xiaomi AX3000T and the snapshots have been broken for about 2-3 days. The update process runs perfectly, but after a restart the end devices are no longer assigned an IP by the DHCP server! The router is no longer accessible or pingable. Only a TFTP flash to an older version helps.

### OpenWrt version

Newest Snapshots

### OpenWrt release

Newest Snapshots

### OpenWrt target/subtarget

Newest Snapshots

### Device

Xiaomi AX3000T

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-06-09:
Without the log from the serial console, it will be difficult to find the cause.

RadioOperator, 2024-06-09:
Tested my build from snapshots, admin IP is 192.168.31.1, all ok.

share it here:
[ax3000t-snapshot-other-files.zip](https://github.com/user-attachments/files/15751985/ax3000t-snapshot-other-files.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-factory.zip](https://github.com/user-attachments/files/15751986/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-factory.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-kernel.zip](https://github.com/user-attachments/files/15751987/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-kernel.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-squashfs-sysupgrade.zip](https://github.com/user-attachments/files/15751990/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-squashfs-sysupgrade.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-bl31-uboot.zip](https://github.com/user-attachments/files/15751992/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-bl31-uboot.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-initramfs-factory.zip](https://github.com/user-attachments/files/15751993/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-initramfs-factory.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-initramfs-recovery.zip](https://github.com/user-attachments/files/15751995/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-initramfs-recovery.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-preloader.zip](https://github.com/user-attachments/files/15751996/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-preloader.zip)
[ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-squashfs-sysupgrade.zip](https://github.com/user-attachments/files/15751997/ax3000t-snapshot-r26580-dd6bbbabd3-mediatek-filogic-xiaomi_mi-router-ax3000t-ubootmod-squashfs-sysupgrade.zip)

adamast0r, 2024-06-09:
I setup my new ax3000t and I noticed exactly the same problem, the folowing image is definilty broken, tried like 5 different times just in case:

https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-factory.ubi

Didn't get a proper IP from the router, could not ping, looked dead.

End up using the Immortalwrt version and it worked without any issues:
https://downloads.immortalwrt.org/releases/23.05-SNAPSHOT/targets/mediatek/filogic/immortalwrt-23.05-snapshot-r27716-ce2b2bccbc-mediatek-filogic-xiaomi_mi-router-ax3000t-initramfs-factory.ubi

s23317, 2024-06-09:
I have just installed the latest snapshot. It looks like the previous error has already been fixed. 
Working build is from Sun Jun 9 19:49:58 2024.

f0ff886f, 2024-06-11:
This bug is fixed, it can be closed.

https://forum.openwrt.org/t/asus-tuf-gaming-ax6000-support/163002/786?u=f0ff886f has the broken and fixing commit.

RadioOperator, 2024-06-18:
The wireless / 802.11ax device driver is dead? cannot join to a network now.