openwrt/openwrt #15666: libext2fs.so wrong library path to libc

DBLouis, 2024-06-09:
### Describe the bug

My build is working, except `libext2fs` which has strange dependency with a path in the build environment:
It was working before, `ext2fs` tools were built and used to resize the root partition. Now I just updated the git HEAD and changed a few unrelated things and got this:

```
louis@openwrt:~$ ldd /usr/lib/libext2fs.so.2
	ldd (0x7fdf23b28000)
	libuuid.so.1 => /usr/lib/libuuid.so.1 (0x7fdf23ac5000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7fdf23aba000)
Error loading shared library /home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib/libc.so: No such file or directory (needed by /usr/lib/libext2fs.so.2)
	libc.so => ldd (0x7fdf23b28000)
Error relocating /usr/lib/libext2fs.so.2: _et_list: symbol not found
Error relocating /usr/lib/libext2fs.so.2: com_err: symbol not found
```

### OpenWrt version

r23897-3287d82fcf

### OpenWrt release

23.05

### OpenWrt target/subtarget

x86/64

### Device

PC Engines apu2

### Image kind

Self-built image

### Steps to reproduce

make distclean
cp ../diffconfig .config
make defconfig
./scripts/feeds update -a
./scripts/feeds install -a
make download
make -j $(nproc)

### Diffconfig

```text
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K_LEDS=y
CONFIG_ATH10K_THERMAL=y
CONFIG_ATH9K_HWRNG=y
CONFIG_ATH9K_SUPPORT_PCOEM=y
CONFIG_ATH_USER_REGD=y
CONFIG_BIND_ENABLE_DOH=y
CONFIG_BPF_TOOLCHAIN_BUILD_LLVM=y
# CONFIG_BPF_TOOLCHAIN_NONE is not set
CONFIG_BUSYBOX_CONFIG_ACPID=y
CONFIG_BUSYBOX_CONFIG_BC=y
CONFIG_BUSYBOX_CONFIG_BLKID=y
CONFIG_BUSYBOX_CONFIG_BUSYBOX=y
CONFIG_BUSYBOX_CONFIG_DIFF=y
CONFIG_BUSYBOX_CONFIG_ETHER_WAKE=y
CONFIG_BUSYBOX_CONFIG_FEATURE_BLKID_TYPE=y
[... 439 lines cut ...]
CONFIG_VERSION_CODE_FILENAMES=y
CONFIG_VERSION_DIST=""
CONFIG_VERSION_FILENAMES=y
CONFIG_VERSION_HOME_URL="https://pcengines.github.io/"
CONFIG_VERSION_HWREV=""
CONFIG_VERSION_MANUFACTURER="PC Engines"
CONFIG_VERSION_MANUFACTURER_URL="https://www.pcengines.ch/apu2e4.htm"
CONFIG_VERSION_NUMBER="23.05"
CONFIG_VERSION_PRODUCT="APU2E4"
CONFIG_VERSION_REPO="https://downloads.openwrt.org/releases/%V"
CONFIG_VERSION_SUPPORT_URL=""
CONFIG_WPA_MBO_SUPPORT=y
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_shadow-all=y
# CONFIG_NUT_DRIVER_NEON is not set
# CONFIG_NUT_DRIVER_SERIAL is not set
# CONFIG_OPENSSH_LIBFIDO2 is not set
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
# CONFIG_PACKAGE_libcbor is not set
# CONFIG_PACKAGE_libfido2 is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2024-06-09:
Any chance you can try master?

Because on master its finding libc just fine

DBLouis, 2024-06-09:
Interesting, did you try it? I'd rather not use master.

robimarko, 2024-06-10:
Well, I tried both master and current 23.05 and its finding libc just fine

DBLouis, 2024-06-10:
Somehow I end up with `libxext2fs.so` with dependency with local path:

```
louis@ananas:~/Projects/openwrt$ for f in $(find openwrt/ -iname '*libext2fs.so*'); do echo $f; readelf -d "$f" | grep Projects; done
openwrt/staging_dir/target-x86_64_musl/root-x86/usr/lib/libext2fs.so.2
openwrt/staging_dir/target-x86_64_musl/root-x86/usr/lib/libext2fs.so.2.4
openwrt/staging_dir/target-x86_64_musl/usr/lib/libext2fs.so.2
openwrt/staging_dir/target-x86_64_musl/usr/lib/libext2fs.so.2.4
openwrt/staging_dir/target-x86_64_musl/usr/lib/libext2fs.so
openwrt/tmp/mklibs-out/libext2fs.so.2
 0x0000000000000001 (NEEDED)             Shared library: [/home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib/libc.so]
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/.pkgdir/libext2fs/usr/lib/libext2fs.so.2
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/.pkgdir/libext2fs/usr/lib/libext2fs.so.2.4
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/ipkg-install/usr/lib/libext2fs.so.2
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/ipkg-install/usr/lib/libext2fs.so.2.4
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/ipkg-install/usr/lib/libext2fs.so
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/lib/libext2fs.so.2
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/lib/libext2fs.so.2.4
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/lib/ext2fs/libext2fs.so.2.4
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/lib/libext2fs.so
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/ipkg-x86_64/libext2fs/usr/lib/libext2fs.so.2
openwrt/build_dir/target-x86_64_musl/e2fsprogs-1.47.0/ipkg-x86_64/libext2fs/usr/lib/libext2fs.so.2.4
openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/libext2fs.so.2
 0x0000000000000001 (NEEDED)             Shared library: [/home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib/libc.so]
openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/libext2fs.so.2.4
 0x0000000000000001 (NEEDED)             Shared library: [/home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib/libc.so]
openwrt/build_dir/target-x86_64_musl/root.orig-x86/usr/lib/libext2fs.so.2
openwrt/build_dir/target-x86_64_musl/root.orig-x86/usr/lib/libext2fs.so.2.4
louis@ananas:~/Projects/openwrt$ ldd openwrt/tmp/mklibs-out/libext2fs.so.2
	linux-vdso.so.1 (0x00007ffdd158a000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007e9faa956000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007e9faa932000)
	/home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib/libc.so (0x00007e9faa8ba000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007e9faa600000)
	/lib64/ld-linux-x86-64.so.2 (0x00007e9faa9e4000)
```

robimarko, 2024-06-10:
Well, `USE_MKLIBS` would explain the issue

e9hack, 2024-06-10:
A dirty fix is to add 
```
mkdir -p /home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib
ln -s /lib/libc.so /home/louis/Projects/openwrt/openwrt/staging_dir/target-x86_64_musl/root-x86/lib/libc.so
```
to /etc/rc.local on the router.