openwrt/openwrt #15669: zsh: termcap.c:45:14: error: conflicting types for 'boolcodes'; have 'char *[]'

Ra2-IFV, 2024-06-09:
### Describe the bug

```text
aarch64-openwrt-linux-musl-gcc -c -I. -I../../Src -I../../Src -I../../Src/Zle -I. -I/openwrt/staging_dir/target-aarch64_generic_musl/usr/include -I/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/usr/include -I/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include -I/openwrt/staging_dir/toolchain-aarch64_generic_gcc-14.1.0_musl/include/fortify  -DHAVE_CONFIG_H -Ofast -pipe -march=armv8-a -mtune=cortex-a72.cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/openwrt/build_dir/target-aarch64_generic_musl/zsh-5.9=zsh-5.9 -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fPIC   -o termcap.o termcap.c               termcap.c:45:14: error: conflicting types for 'boolcodes'; have 'char *[]'                                   45 | static char *boolcodes[] = {                                                                            |              ^~~~~~~~~
In file included from ../../Src/zshterm.h:1,
                 from ../../Src/zsh_system.h:932,                                                                          from ../../Src/zsh.mdh:17,
                 from termcap.mdh:17,
                 from termcap.c:38:
/openwrt/staging_dir/target-aarch64_generic_musl/usr/include/term.h:783:56: note: previous declaration of 'boolcodes' with type 'const char * const[]'
  783 | S_EXPORT_VAR(NCURSES_CONST char * const ) boolcodes[];
      |                                           ^~~~~~~~~
                                                                                                          make[6]: *** [Makefile:230: termcap.o] Error 1
make[6]: Leaving directory '/openwrt/build_dir/target-aarch64_generic_musl/zsh-5.9/Src/Modules'
make[5]: *** [Makemod:372: modobjs] Error 1
make[5]: Leaving directory '/openwrt/build_dir/target-aarch64_generic_musl/zsh-5.9/Src'
make[4]: *** [Makefile:449: modobjs] Error 2
make[4]: Leaving directory '/openwrt/build_dir/target-aarch64_generic_musl/zsh-5.9/Src'
make[3]: *** [Makefile:188: all] Error 1
make[3]: Leaving directory '/openwrt/build_dir/target-aarch64_generic_musl/zsh-5.9'
make[2]: *** [Makefile:133: /openwrt/build_dir/target-aarch64_generic_musl/zsh-5.9/.built] Error 2
make[2]: Leaving directory '/openwrt/feeds/packages/utils/zsh'
time: package/feeds/packages/zsh/compile#0.45#0.13#0.54
    ERROR: package/feeds/packages/zsh failed to build.
make[1]: *** [package/Makefile:173: package/feeds/packages/zsh/compile] Error 1
make[1]: Leaving directory '/openwrt'
```

### OpenWrt version

3de3c2bdfa6b757a1ac9fa4d9905e0d268d98e5f

### OpenWrt release

3de3c2bdfa6b757a1ac9fa4d9905e0d268d98e5f

### OpenWrt target/subtarget

aarch64 generic

### Device

aarch64 generic

### Image kind

Official downloaded image

### Steps to reproduce

`make package/feeds/packages/zsh/compile V=s`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Ra2-IFV, 2024-06-09:
Build environment: GCC 14, experimental LTO for all packages enabled, experimental linker enabled

Ra2-IFV, 2024-06-09:
Confirmed not caused by LTO.

brada4, 2024-06-09:
fixed yesterday
https://github.com/openwrt/packages/issues/24344
syntax checker, nothing to do with lto

Ra2-IFV, 2024-06-09:
Closing as fixed.

closed 2024-06-09 by Ra2-IFV