openwrt/openwrt #15687: why there no libx264 option  to select

fareign, 2024-06-11:
### Describe the bug

[image]

I reinstall feeds ../scripts/feeds install libx264
 and still no menu

### OpenWrt version

23.05

### OpenWrt release

snap

### OpenWrt target/subtarget

make menuconfig

### Device

rpi cm4

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

fareign, 2024-06-11:
> Invalid Release reported. `snap` Is this from a clean repository?

I buidl from master yesterday.use office config.

fareign, 2024-06-11:
I read the Makefile ,can not find anything wrong,so I want some help.thanks.

fareign, 2024-06-12:
> You must enable BUILD_PATENTED ( disabled by default ) to build those packages. You can see it clearly in those Makefiles about dependency on BUILD_PATENTED.

thanks，solved

raenye, 2024-06-12:
> thanks，solved

Then this issue can be closed?

closed 2024-06-12 by jow-