openwrt/openwrt #15692: Failed to build `tools/tar'

ArielHeleneto, 2024-06-12:
### Describe the bug

`make menuconfig` and choose MediaTek Ralink ARM Filogic 8x0 (MT798x) GL.iNet GL-MT6000 and save it.

Then build tar with `make tools/tar/compile V=s | tee ~/mm.log`

It failed.

`make` failed too.

### OpenWrt version

b2a84c86e3a2bd399761055f84a81e2aebe436e2

### OpenWrt release

b2a84c86e3a2bd399761055f84a81e2aebe436e2

### OpenWrt target/subtarget

MediaTek Ralink ARM Filogic 8x0 (MT798x)

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Steps to reproduce

`make menuconfig` and choose MediaTek Ralink ARM Filogic 8x0 (MT798x) GL.iNet GL-MT6000 and save it. `make tools/tar/compile V=s | tee ~/mm.log`

[mm.log](https://github.com/user-attachments/files/15800981/mm.log)

```log
/home/ariel/openwrt/staging_dir/host/bin/gcc -DHAVE_CONFIG_H -I. -I..  -I../gnu -I../ -I../gnu -I../lib -I../lib -I/home/ariel/openwrt/staging_dir/host/include    -O2 -I/home/ariel/openwrt/staging_dir/host/include  -c -o suffix.o suffix.c
/home/ariel/openwrt/staging_dir/host/bin/gcc -DHAVE_CONFIG_H -I. -I..  -I../gnu -I../ -I../gnu -I../lib -I../lib -I/home/ariel/openwrt/staging_dir/host/include    -O2 -I/home/ariel/openwrt/staging_dir/host/include  -c -o system.o system.c
In file included from system.c:23:
system.c: In function 'mtioseek':
system.c:59:49: error: passing argument 3 of 'rmt_ioctl__' from incompatible pointer type [-Wincompatible-pointer-types]
   59 |           && (0 <= rmtioctl (archive, MTIOCTOP, &operation)
      |                                                 ^~~~~~~~~~
      |                                                 |
      |                                                 struct mtop *
../lib/rmt.h:86:57: note: in definition of macro 'rmtioctl'
   86 |   (_isrmt (fd) ? rmt_ioctl__ (fd - __REM_BIAS, request, argument) \
      |                                                         ^~~~~~~~
../lib/rmt.h:28:28: note: expected 'char *' but argument is of type 'struct mtop *'
   28 | int rmt_ioctl__ (int, int, char *);
      |                            ^~~~~~
system.c:61:56: error: passing argument 3 of 'rmt_ioctl__' from incompatible pointer type [-Wincompatible-pointer-types]
   61 |                   && 0 <= rmtioctl (archive, MTIOCTOP, &operation))))
      |                                                        ^~~~~~~~~~
      |                                                        |
      |                                                        struct mtop *
../lib/rmt.h:86:57: note: in definition of macro 'rmtioctl'
   86 |   (_isrmt (fd) ? rmt_ioctl__ (fd - __REM_BIAS, request, argument) \
      |                                                         ^~~~~~~~
../lib/rmt.h:28:28: note: expected 'char *' but argument is of type 'struct mtop *'
   28 | int rmt_ioctl__ (int, int, char *);
      |                            ^~~~~~
make[5]: *** [Makefile:1581: system.o] Error 1
make[5]: Leaving directory '/home/ariel/openwrt/build_dir/host/tar-1.34/src'
make[4]: *** [Makefile:1546: all-recursive] Error 1
make[4]: Leaving directory '/home/ariel/openwrt/build_dir/host/tar-1.34'
make[3]: *** [Makefile:1486: all] Error 2
make[3]: Leaving directory '/home/ariel/openwrt/build_dir/host/tar-1.34'
make[2]: *** [Makefile:36: /home/ariel/openwrt/build_dir/host/tar-1.34/.built] Error 2
make[2]: Leaving directory '/home/ariel/openwrt/tools/tar'
time: tools/tar/compile#34.77#9.43#48.50
    ERROR: tools/tar failed to build.
make[1]: *** [tools/Makefile:231: tools/tar/compile] Error 1
make[1]: Leaving directory '/home/ariel/openwrt'
make: *** [/home/ariel/openwrt/include/toplevel.mk:248：tools/tar/compile] 错误 2
```

### Actual behaviour

Failed to build. Log is attached.

### Expected behaviour

Build with no error

### Additional info

Revert https://github.com/openwrt/openwrt/commit/2951e0a80e9f9a8d11e5bc04a89d9c7aa0623919 will fix this bug.

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y
CONFIG_FEED_immortalwrt_luci=y
CONFIG_FEED_immortalwrt_packages=y
CONFIG_FEED_kenzok8_packages=y
CONFIG_FEED_nezha=y
CONFIG_FEED_small_package=y
CONFIG_FEED_immortalwrt_luci=y
CONFIG_FEED_immortalwrt_packages=y
CONFIG_FEED_kenzok8_packages=y
CONFIG_FEED_nezha=y
CONFIG_FEED_small_package=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ArielHeleneto, 2024-06-12:
How does this bot work? I'd say that b2a84c86e3a2bd399761055f84a81e2aebe436e2 means https://github.com/openwrt/openwrt/commit/b2a84c86e3a2bd399761055f84a81e2aebe436e2. The target is `MediaTek Ralink ARM`. The subtarget is `Filogic 8x0 (MT798x)`.

brada4, 2024-06-12:
Go ask immortalwrt, this is for openwrt issues. Bot is 101% correct

ynezz, 2024-06-12:
> How does this bot work?

It seems, that it works well. You just need to actually read the issue template completely, mainly the `use command bellow` part 

[image]

ArielHeleneto, 2024-06-12:
> It seems, that it works well. You just need to actually read the issue template completely, mainly the `use command bellow` part [...]

it should be something like `r0-b2a84c8`. I run the command on my openwrt(

ynezz, 2024-06-12:
Now I see, this is a problem with tools, so yeah, good point.

ynezz, 2024-06-12:
```
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y
CONFIG_FEED_immortalwrt_luci=y
CONFIG_FEED_immortalwrt_packages=y
CONFIG_FEED_kenzok8_packages=y
CONFIG_FEED_nezha=y
CONFIG_FEED_small_package=y
CONFIG_FEED_immortalwrt_luci=y
CONFIG_FEED_immortalwrt_packages=y
CONFIG_FEED_kenzok8_packages=y
CONFIG_FEED_nezha=y
CONFIG_FEED_small_package=y
```

Seems not complete, provide complete output from the `./scripts/diffconfig.sh` command

ArielHeleneto, 2024-06-12:
> ```
> CONFIG_TARGET_mediatek=y
> CONFIG_TARGET_mediatek_filogic=y
> CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y
> CONFIG_FEED_immortalwrt_luci=y
> CONFIG_FEED_immortalwrt_packages=y [...]

Wait a while. let me revert my revert.

ArielHeleneto, 2024-06-12:
> ```
> CONFIG_TARGET_mediatek=y
> CONFIG_TARGET_mediatek_filogic=y
> CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y
> CONFIG_FEED_immortalwrt_luci=y
> CONFIG_FEED_immortalwrt_packages=y [...]

```log
tmp/.config-package.in:115652:error: recursive dependency detected!
tmp/.config-package.in:115652:	symbol PACKAGE_strongswan is selected by PACKAGE_luci-app-ipsec-vpnd
tmp/.config-package.in:59804:	symbol PACKAGE_luci-app-ipsec-vpnd depends on PACKAGE_strongswan-mod-kdf
tmp/.config-package.in:116300:	symbol PACKAGE_strongswan-mod-kdf depends on PACKAGE_strongswan
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:61447:error: recursive dependency detected!
tmp/.config-package.in:61447:	symbol PACKAGE_luci-app-qbittorrent depends on PACKAGE_luci-app-qbittorrent
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:25743:error: recursive dependency detected!
tmp/.config-package.in:25743:	symbol PACKAGE_node-cylon-gpio is selected by PACKAGE_node-cylon-gpio
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:25768:error: recursive dependency detected!
tmp/.config-package.in:25768:	symbol PACKAGE_node-cylon-i2c is selected by PACKAGE_node-cylon-i2c
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:126135:error: recursive dependency detected!
tmp/.config-package.in:126135:	symbol PACKAGE_tor is selected by PACKAGE_luci-app-torbp
tmp/.config-package.in:63463:	symbol PACKAGE_luci-app-torbp depends on PACKAGE_tor
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:98881:error: recursive dependency detected!
tmp/.config-package.in:98881:	symbol PACKAGE_mentohust is selected by PACKAGE_mentohust
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:115652:error: recursive dependency detected!
tmp/.config-package.in:115652:	symbol PACKAGE_strongswan is selected by PACKAGE_luci-app-ipsec-vpnd
tmp/.config-package.in:59804:	symbol PACKAGE_luci-app-ipsec-vpnd depends on PACKAGE_strongswan-mod-kdf
tmp/.config-package.in:116300:	symbol PACKAGE_strongswan-mod-kdf depends on PACKAGE_strongswan
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:61447:error: recursive dependency detected!
tmp/.config-package.in:61447:	symbol PACKAGE_luci-app-qbittorrent depends on PACKAGE_luci-app-qbittorrent
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:25743:error: recursive dependency detected!
tmp/.config-package.in:25743:	symbol PACKAGE_node-cylon-gpio is selected by PACKAGE_node-cylon-gpio
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:25768:error: recursive dependency detected!
tmp/.config-package.in:25768:	symbol PACKAGE_node-cylon-i2c is selected by PACKAGE_node-cylon-i2c
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:126135:error: recursive dependency detected!
tmp/.config-package.in:126135:	symbol PACKAGE_tor is selected by PACKAGE_luci-app-torbp
tmp/.config-package.in:63463:	symbol PACKAGE_luci-app-torbp depends on PACKAGE_tor
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:98881:error: recursive dependency detected!
tmp/.config-package.in:98881:	symbol PACKAGE_mentohust is selected by PACKAGE_mentohust
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_glinet_gl-mt6000=y

```

ynezz, 2024-06-12:
So this is system using gcc 14 as a compiler?

ArielHeleneto, 2024-06-12:
> So this is system using gcc 14 as a compiler?

yes and gcc 14.1.1 with my Archlinux

```
[ariel@archlinux ~]$ gcc -v
使用内建 specs。
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/14.1.1/lto-wrapper
目标：x86_64-pc-linux-gnu
配置为：/build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,m2,objc,obj-c++,rust --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/issues --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
线程模型：posix
支持的 LTO 压缩算法：zlib zstd
gcc 版本 14.1.1 20240522 (GCC) 
[ariel@archlinux ~]$ g++ -v
使用内建 specs。
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/14.1.1/lto-wrapper
目标：x86_64-pc-linux-gnu
配置为：/build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,m2,objc,obj-c++,rust --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/issues --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
线程模型：posix
支持的 LTO 压缩算法：zlib zstd
gcc 版本 14.1.1 20240522 (GCC) 
```

sudoBash418, 2024-06-12:
Same issue here, also on Arch Linux w/ GCC 14.1.1.

commit 95597dc2 (2024-06-12): tools: tar: backport patch for paxlib shipped in tar

Tar 1.34 ship an old version of paxlib with rtapelib.c that produce some
compilation warning. This library got updated in 1.35 but we still can't
use the new Tar version.

GCC 14 then made these compilarion warning errors.

Manually backport the fixes to rtapelib.c and patch the version shipped
in 1.34 to fix these compilation warning.

Fixes: #15692
Fixes: 2951e0a80e9f ("tools: tar: backport patches fixing broken --delete")
closed 2024-06-12 by openwrt-bot