openwrt/openwrt #15694: qmi.sh should set network modes before try to register network

dc3l1ne, 2024-06-12:
### Describe the bug

I found that my device can't connect to LTE network, the error code is NETWORK_REGISTRATION_FAILED. I used the command `uqmi -s -d /dev/cdc-wdm0 --get-serving-system` , it returned `{"registration":"searching"}`.

It seems that the modem failed the registration process with carrier. I changed the network mode to LTE and the registration process can be completed with command `uqmi -s -d /dev/cdc-wdm0  --set-network-modes lte`. 

I noticed that in `/lib/netifd/proto/qmi.sh` this command executed after `uqmi -s -d "$device" --network-register > /dev/null 2>&1` which cause the network registration failed even I specify `option modes 'lte'` in config file.

Solve:
I moved `[ -n "$modes" ] && uqmi -s -d "$device" --set-network-modes "$modes" > /dev/null 2>&1`  between `uqmi -s -d "$device" --sync > /dev/null 2>&1` and `uqmi -s -d "$device" --network-register > /dev/null 2>&1`
Problem solved!__

### OpenWrt version

r19803-9a599fee93

### OpenWrt release

22.03.2

### OpenWrt target/subtarget

ath79/nand

### Device

GL.iNet GL-XE300

### Image kind

Official downloaded image

### Steps to reproduce

Maybe it cannot be reproduced, it depends on your carrier.

### Actual behaviour

LUCI report LTE connect failed and raise error code NETWORK_REGISTRATION_FAILED, qmi.sh can do the network modes change, but it should be done before start to register network.

### Expected behaviour

LTE network should connect successfully without any error.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2024-06-13:
issue only happens on LTE? what about 5gnr?

dc3l1ne, 2024-06-13:
> issue only happens on LTE? what about 5gnr?

@ptpt52  Sorry, I can't test that because my router only support 4G network.

BTW, there is a certain probability that it may failed the registration process even I put the command after `uqmi -s -d "$device" --sync > /dev/null 2>&1` , system log attatched

> Thu Jun 13 19:06:12 2024 daemon.notice netifd: Interface 'LTE' is setting up now
Thu Jun 13 19:06:12 2024 daemon.notice netifd: LTE (5938): Waiting for SIM initialization
Thu Jun 13 19:06:13 2024 daemon.notice netifd: LTE (5938): Failed to parse message data
Thu Jun 13 19:06:13 2024 daemon.notice netifd: LTE (5938): PIN verification is disabled
Thu Jun 13 19:06:13 2024 daemon.notice netifd: LTE (5938): Device does not support 802.3 mode. Informing driver of raw-ip only for wwan0 ..
Thu Jun 13 19:06:14 2024 daemon.notice netifd: LTE (5938): Waiting for network registration
Thu Jun 13 19:06:37 2024 daemon.notice netifd: LTE (5938): Network registration failed, registration timeout reached
Thu Jun 13 19:06:37 2024 daemon.notice netifd: LTE (6876): Stopping network LTE
Thu Jun 13 19:06:37 2024 daemon.notice netifd: LTE (6876): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "LTE" } (Permission denied)
Thu Jun 13 19:06:37 2024 daemon.notice netifd: Interface 'LTE' is now down

So I choose the second way, put that in registration state checking loop, and it can finish the registration everytime. System log as below

> Thu Jun 13 19:15:52 2024 daemon.notice netifd: Interface 'LTE' is setting up now
Thu Jun 13 19:15:52 2024 daemon.notice netifd: LTE (7265): Waiting for SIM initialization
Thu Jun 13 19:15:52 2024 daemon.notice netifd: LTE (7265): Failed to parse message data
Thu Jun 13 19:15:52 2024 daemon.notice netifd: LTE (7265): PIN verification is disabled
Thu Jun 13 19:15:53 2024 daemon.notice netifd: LTE (7265): Device does not support 802.3 mode. Informing driver of raw-ip only for wwan0 ..
Thu Jun 13 19:15:53 2024 daemon.notice netifd: LTE (7265): Waiting for network registration
Thu Jun 13 19:15:55 2024 daemon.notice netifd: LTE (7265): Set modes to lte during registration #this my code output
Thu Jun 13 19:15:58 2024 daemon.notice netifd: LTE (7265): Starting network LTE
Thu Jun 13 19:15:59 2024 daemon.notice netifd: LTE (7265): Setting up wwan0
Thu Jun 13 19:15:59 2024 daemon.notice netifd: Interface 'LTE' is now up
Thu Jun 13 19:15:59 2024 daemon.notice netifd: Network device 'wwan0' link is up
Thu Jun 13 19:15:59 2024 daemon.notice netifd: Network alias 'wwan0' link is up
Thu Jun 13 19:15:59 2024 daemon.notice netifd: Interface 'LTE_4' is enabled
Thu Jun 13 19:15:59 2024 daemon.notice netifd: Interface 'LTE_4' has link connectivity
Thu Jun 13 19:15:59 2024 daemon.notice netifd: Interface 'LTE_4' is setting up now
Thu Jun 13 19:16:00 2024 daemon.notice netifd: LTE_4 (7390): udhcpc: started, v1.35.0
Thu Jun 13 19:16:00 2024 daemon.notice netifd: LTE_4 (7390): udhcpc: broadcasting discover
Thu Jun 13 19:16:00 2024 daemon.notice netifd: LTE_4 (7390): udhcpc: broadcasting select for 100.111.189.142, server 100.111.189.141
Thu Jun 13 19:16:00 2024 daemon.notice netifd: LTE_4 (7390): udhcpc: lease of 100.111.189.142 obtained from 100.111.189.141, lease time 7200

Code block:
```
while true; do
	registration_state=$(uqmi -s -d "$device" --get-serving-system 2>/dev/null | jsonfilter -e "@.registration" 2>/dev/null)

	[ "$registration_state" = "registered" ] && break

	if [ "$registration_state" = "searching" ] || [ "$registration_state" = "not_registered" ]; then
		if [ "$registration_timeout" -lt "$timeout" ] || [ "$timeout" = "0" ]; then
			[ "$registration_state" = "searching" ] || {
				echo "Device stopped network registration. Restart network registration"
				uqmi -s -d "$device" --network-register > /dev/null 2>&1
			}
			let registration_timeout++
			sleep 1
			[ -n "$modes" ] && {
				uqmi -s -d "$device" --set-network-modes "$modes"
				echo "Set modes to $modes during registration"
				# Sleep for 3 before check again, and avoid the modem is in busy state
				sleep 3
			}
			continue
		fi
		echo "Network registration failed, registration timeout reached"
	else
		# registration_state is 'registration_denied' or 'unknown' or ''
		echo "Network registration failed (reason: '$registration_state')"
	fi

	proto_notify_error "$interface" NETWORK_REGISTRATION_FAILED
	proto_block_restart "$interface"
	return 1
done
```

ptpt52, 2024-06-13:
could you send a PR?

dc3l1ne, 2024-06-13:
When I try to send a PR, and I think this bug has been fixed in latest qmi.sh
Line 258 of qmi.sh in main branch, the script will do network scan after set network mode for 30s. I just test it and works good on my device!
```
[ -n "$modes" ] && {
	uqmi -s -d "$device" -t 1000 --set-network-modes "$modes" > /dev/null 2>&1
	sleep 3
	# Scan network to not rely on registration-timeout after RAT change
	uqmi -s -d "$device" -t 30000 --network-scan > /dev/null 2>&1
}
```

closed 2024-06-14 by dc3l1ne