openwrt/openwrt #15709: libelf NLS supports issues

vortexilation, 2024-06-14:
### Describe the bug

When trying to compile [off tree](https://github.com/muink/openwrt-einat-ebpf) OpenWrt Packages I have stumbled this issue of libelf cann't found libintl.so.8 .

The application is a rust daemon, it stops at libelf not being able to found libintl.so.8 : 
```
  = note: LC_ALL="C" PATH="/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/bin:/home/user/works/openwrt/staging_dir/hostpkg/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/toolchain-x86_64_gcc-14.1.0_glibc/bin:/home/user/works/openwrt/staging_dir/host/bin:/home/user/works/openwrt/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" VSLANG="1033" "x86_64-openwrt-linux-gnu-gcc" "-m64" "/home/user/works/openwrt/tmp/rustca9rJSt/symbols.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec/build_script_build-13037c25982734ec.build_script_build.900026c213c08482-cgu.0.rcgu.o" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/einat-13037c25982734ec/build_script_build-13037c25982734ec.2kuwii56gzittnew.rcgu.o" "-Wl,--as-needed" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/libbpf-sys-de20753a55c71e06/out" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-L" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/build/vsprintf-f6a1be3b8a629995/out" "-L" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_cargo-fc7173ecf7072a22.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_metadata-77b698b469ccad23.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libserde_json-d18e1952940c9d6c.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libitoa-c99bf860a1243733.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libryu-42733ab1749368f9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libthiserror-aa231274481bae73.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcargo_platform-c20f7cad56b30019.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcamino-a1fd712e042bd596.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libmemmap2-73fca818d05ca4c9.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_rs-12af648727bedc4e.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libvsprintf-52f5d1afc74d566c.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibc-a50be116079b6680.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblibbpf_sys-6ce4fe8ed806fc57.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libsemver-d9af4a6bbaa40b09.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libserde-73abfa3b10e52ca3.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libregex-662c0e78adcebbf3.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libregex_automata-e8e6e085a55c8ab3.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libregex_syntax-f1d457fe75bcbfd6.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libtempfile-e4833a2ad6c4f429.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libfastrand-906c084e5ef6039e.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/librustix-4d3b20d2716f8231.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libbitflags-3a495cf5bbfede0b.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/liblinux_raw_sys-ba02cfb0a81adfba.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libcfg_if-1c8851ce38e8a7c1.rlib" "/home/user/works/openwrt/build_dir/target-x86_64_glibc_custom/einat-ebpf-0.1.3/target/release/deps/libanyhow-0deb8dd3c263e7fc.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libstd-18d1c84127d04dc5.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libpanic_unwind-bc53c6e5d5f4111f.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libobject-a79bc22e6ae9ee83.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libmemchr-9c256084533d6dad.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libaddr2line-e53b44d64d10bb35.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libgimli-9d0b3274a5b15e82.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_demangle-b0fb619b58a6601a.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libstd_detect-1319f451087d6793.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libhashbrown-e7ba5787d3fb181b.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_std_workspace_alloc-e23167ed65317fed.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libminiz_oxide-52af0786d05d61a5.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libadler-28f9d348427e8d43.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libunwind-689892118ddcc60e.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcfg_if-a1f86f0be6e775e6.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/liblibc-cbc93ea126f55c98.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/liballoc-390f90785cbfdde3.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/librustc_std_workspace_core-aded7817e4ea9283.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcore-6c16e1294e732b84.rlib" "/home/user/works/openwrt/staging_dir/target-x86_64_glibc_custom/host/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcompiler_builtins-a8a94ae78107d089.rlib" "-Wl,-Bdynamic" "-lelf" "-lz" "-lgcc_s" "-lutil" "-lrt" "-lpthread" "-lm" "-ldl" "-lc" "-Wl,--eh-frame-hdr" "-Wl,-z [... cut ...]
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f146c7fb000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f146c616000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f146c872000)
```

Except if I am defining the library path to it with LD_LIBRARY_PATH, but still compilation will failed.

### OpenWrt version

r26593-b47fbca97f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

generic

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

neheb, 2024-06-24:
are you building on Alpine Linux?

vortexilation, 2024-06-29:
No, it's Debian SID glibc compiling/building/targeting OpenWrt x86_64 glibc.