openwrt/openwrt #15728: Client Isolation not working with WDS

XDjackieXD, 2024-06-16:
### Describe the bug

When enabling client isolation in the WiFi settings this does not work for WDS clients (as those create separate bridge ports).
It would be nice if enabling client isolation on WiFi would enable port isolation for these bridge ports.

I worked around it with the following (hacky) script in `/etc/hotplug.d/net`, but I'm fairly sure that there is a better way:
```
#!/bin/sh

. /lib/functions.sh

[ "${ACTION}" = "add" ] && {
	echo ${DEVICENAME} | grep "wlan0" && echo 1 > /sys/devices/virtual/net/br-funkfeuer/lower_${DEVICENAME}/brport/isolated
}
```

I currently don't have a device with a more up to date release at hand where I could easily test if this issue is still present (sorry :/) but I could not find any hints that it had been solved in the meantime.

### OpenWrt version

r19803-9a599fee93

### OpenWrt release

22.03.2

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti PowerBeam 5AC Gen2

### Image kind

Self-built image

### Steps to reproduce

- set up AP with WDS and enable client isolation
- connect two WDS clients
- ping client two from client one

### Actual behaviour

all WDS clients can reach each other

### Expected behaviour

WDS clients can not reach each other like "normal" wifi clients

### Additional info

Device is actually an Isostation AC (same hardware as powerbeam as far as the firmware is concerned).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-06-16:
I think that's expected behavior, client isolation feature in hostapd isn't really advanced. For a complete solution you want to use per_sta_vif and bridge filtering rules (either using ebtables or nftables, depending on your OpenWRT version) or the built-in port isolation using a hotplug script to achieve this client isolation (like you're doing now). The client isolation feature in hostapd also doesn't support other use cases like different VLANs needing different rules or sophisticated rules to allow certain things to be forwarded.

Note that there is a small bug in iwinfo with per_sta_vif where it doesn't show all associated stations that's fixed by my PR here: https://github.com/openwrt/iwinfo/pull/5

XDjackieXD, 2024-06-16:
as far as I know using bridge port isolation is faster than ebtables (but that might've changed in the last years - it's been a long time since I tested that).
WDS basically behaves as if per_sta_vif is active for all WDS clients.
That's the reason why the checkbox currently is useless for this.

If this is expected behaviour it would probably make sense to add a short line to the help text of the "isolate clients" checkbox in LUCI (something along the lines "this will not have an effect if per station VIF is activated or for WDS stations").

I'd really like to have a "can be setup entirely from LUCI" way to configure the bridge port isolation for WiFi station VIFs.
One other possibility would be to have the "isolated" flag in the device configuration for the parent wifi interface also apply to all VIFs from this interface? (I tried setting "isolated" on wifi0 bridge port options but this only applies to the parent interface, not all the WDS stations)

rany2, 2024-06-16:
> If this is expected behaviour it would probably make sense to add a short line to the help text of the "isolate clients" checkbox in LUCI (something along the lines "this will not have an effect if per station VIF is activated or for WDS stations").

I think this should be done to protect users from having an insecure configuration if it is decided that no change is done on OpenWRT/hostapd side to fix this issue. AFAIK the isolate option just maps directly to the hostapd option atm.

> I'd really like to have a "can be setup entirely from LUCI" way to configure the bridge port isolation for WiFi station VIFs. One other possibility would be to have the "isolated" flag in the device [...]

Agreed, and I think this could be done by a hotplug script that checks the UCI config of the WiFi interface. Something that removes the last digits of `.wds{digits}`, maps the stripped interface to the UCI config and checks for isolate. At least conceptually it seems simple.

brada4, 2024-06-17:
Isolation stretches to passing sta sta traffic through first l2 bridge ie one on the extender.
You just need to duplicate your l2 filter there (ebtables-nft or table bridge)

XDjackieXD, 2024-06-17:
> Isolation stretches to passing sta sta traffic through first l2 bridge ie one on the extender. You just need to duplicate your l2 filter there (ebtables-nft or table bridge)

In my case WDS is not used for a wifi extender but for a L3 mesh network (funkfeuer.at). We have one router per node and the router has multiple wireless devices connected (we need the "4 MAC mode" aka WDS so that the wireless device can be separate from the router without hacks like relayd and 802.11s sadly still is troublesome in many cases).
On the AP side I need to prevent all clients to talk to each other, because then the routing protocol will think that all clients have direct links between each other.

But yes. if WDS is used for classic WiFi range extenders than the client isolation also needs to be implemented on the repeater in the same way (on the repeater, enabling bridge port isolation and hostapd's client isolation would work as well instead of L2 filtering with ebtables/nft)

brada4, 2024-06-17:
Yes, there are 2 different settings - one is hostapd isolation other is bridge port isolation
`echo 1 | tee /sys/class/net/br-lan/brif/phy?-ap?/isolated`
Which will permit the wifi port to talk only to non-isolated ports. Though not sure if it can talk back to self hostapd.

XDjackieXD, 2024-06-17:
Yes I know. And in WDS Mode - even if bridge port isolation for the parent interface wlan0 is enabled - the separate bridge interfaces automatically created for WDS stations is not set to isolated.

The whole issue is that I'd like to have some way to set the bridge port isolation from UCI for these automatically created "sub interfaces" for WDS stations (probably a viable solution would be to inherit it from the parent interface).

For my specific case the client isolation integrated in hostapd is irrelevant as I only have WDS stations.