openwrt/openwrt #15737: [23.05] Consistent ujail crash at reboot

systemcrash, 2024-06-17:
### Describe the bug

```
root@OpenWrt:/# opkg remove luci-proto-gre
Removing package luci-proto-gre from root..
root@OpenWrt:/#
[100.594878] br-lan: port 1(eth0) entered disabled state
[100.6418341 br-lan: port 1(eth0) entered disabled state
[100.791137] device eth0 left promiscuous mode
[102.040418] EXT4-fs (sda2): re-mounted. Opts: (null). Quota mode: disabled.
[102.102836] ujail[3206]: segfault at 5e ip 00007f7cd6d5e950 sp 00007ffd2e579110 error 4 in libubox.so.20230523[7f7cd6d5d000+5000]
[ 102.177578] Code: 5b ff 25 6b 75 00 00 31 c0 5b c3 41 54 55 53 48 89 fb 48 83 ec 10 40 f6 c6 03 75 0e 48 83 c4 10 5b 5d 41 5c
e9 6e ff ff ff 90 <80> 7f 0e 00 89 f5 74 07 be 03 00 00 00 eb 38 40 f6 c6 08 75 2d 31
```

### OpenWrt version

r23809-234f1a2efa

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

x86/64

### Device

innotek GmbH VirtualBox

### Image kind

Official downloaded image

### Steps to reproduce

Reboot

### Actual behaviour

ujail crash

### Expected behaviour

no crash

### Additional info

reproducible

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

systemcrash, 2024-06-17:
@brada4 any tips to localise the error?

brada4, 2024-06-17:
You have ubox from older version, that happens when you upgrade ext4 version partially.
Should be libucode.so.20230711 for 23.05.3

brada4, 2024-06-17:
You may install gdb and try to run crashing process from within it, then add respective openwrt source files relative to current directory to determine code that crashes
errno 4 -> interrupted syscall

systemcrash, 2024-06-17:
Seems a bit involved to figure out a crash at reboot. 🙈 No other way to decode the `Code: ` part?

brada4, 2024-06-17:
dmesg did not decode backtrace, thus not really.
just function 0x5000 bytes into libubox crashed, without any idea what called it.

closed 2024-06-17 by systemcrash