openwrt/openwrt #15739: (kernel 6.6) rtl8366s: chip detection failed, err=-19 (WNDRMAC v2 / WNDR3800)

lorand-horvath, 2024-06-18:
### Describe the bug

The eth1 interface is not detected in new kernel 6.6, no WAN available on ath79/generic WNDRMAC v2 (WNDR3800) - it has an RTL8366s chip. Here's the log from the last SNAPSHOT 26674 with kernel 6.6:

```
Sun Jun 16 22:11:07 2024 kern.notice kernel: [    6.449246] Realtek RTL8366S ethernet switch driver version 0.2.2
Sun Jun 16 22:11:07 2024 kern.err kernel: [    6.455468] rtl8366s rtl8366s: cannot find mdio node phandle
Sun Jun 16 22:11:07 2024 kern.info kernel: [    6.461132] rtl8366s rtl8366s: using GPIO pins 5 (SDA) and 7 (SCK)
Sun Jun 16 22:11:07 2024 kern.err kernel: [    6.467389] rtl8366s rtl8366s: unknown chip id (0000)
Sun Jun 16 22:11:07 2024 kern.err kernel: [    6.472435] rtl8366s rtl8366s: chip detection failed, err=-19
```

### OpenWrt version

snapshot

### OpenWrt release

snapshot 26657 - 26674

### OpenWrt target/subtarget

ath79/generic

### Device

WNDRMAC v2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

namiltd, 2024-06-18:
Please attach the log also from version 6.1

lorand-horvath, 2024-06-18:
@namiltd The relevant log from working snapshot 26620 based on kernel 6.1:
```
Tue Jun 18 12:16:22 2024 kern.notice kernel: [    6.287600] Realtek RTL8366S ethernet switch driver version 0.2.2
Tue Jun 18 12:16:22 2024 kern.err kernel: [    6.293785] rtl8366s rtl8366s: cannot find mdio node phandle
Tue Jun 18 12:16:22 2024 kern.info kernel: [    6.299481] rtl8366s rtl8366s: using GPIO pins 5 (SDA) and 7 (SCK)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    6.305759] rtl8366s rtl8366s: RTL8366 ver. 1 chip found
Tue Jun 18 12:16:22 2024 kern.info kernel: [    6.820296] ag71xx 1a000000.eth: connected to PHY at rtl8366s:04 [uid=001cc960, driver=Generic PHY]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    6.830397] eth1: Atheros AG71xx at 0xba000000, irq 5, mode: rgmii
```

markmentovai, 2024-06-19:
I’m seeing this on a WNDR3800 too. The LED mappings also seem wrong. And there are even more ominous messages in `dmesg`:

```
[    1.569227] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
```

(I see that message on kernel 6.1 too, but it eventually succeeds. Not on 6.6.)

markmentovai, 2024-06-19:
@DragonBluep, @PolynomialDivision: 4fdc7bb8f13fe7a3eff2e551f0fe621a6dee706e switched ath79 to kernel 6.6 but there are two reports that it’s broken for the WNDR3700 family.

lorand-horvath, 2024-06-19:
> I’m seeing this on a WNDR3800 too. The LED mappings also seem wrong.

The LEDs seem fine here, except the Power LED is off, I assume because of the broken WAN / eth1.

DragonBluep, 2024-06-19:
Please attach the full `dmesg`. I'm pretty sure it's caused by the broken gpio driver.

lorand-horvath, 2024-06-19:
@DragonBluep Full dmesg for build 26657 (kernel 6.6):
```
Sun Jun 16 22:11:07 2024 kern.notice kernel: [    0.000000] Linux version 6.6.33 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26657-531b3f667c) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Sun Jun 16 22:10:53 2024
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] MIPS: machine is Netgear WNDRMAC v2
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Zone ranges:
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Early memory node ranges
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sun Jun 16 22:11:07 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sun Jun 16 22:11:07 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Sun Jun 16 22:11:07 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] Memory: 119416K/131072K available (6713K kernel code, 595K rwdata, 1560K rodata, 1236K init, 215K bss, 11656K reserved, 0K cma-reserved)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] CPU clock: 680.000 MHz
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5621354254 ns
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.007857] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.074099] pid_max: default: 32768 minimum: 301
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.090300] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.097615] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.115103] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.127850] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.137734] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.147530] pinctrl core: initialized pinctrl subsystem
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.155847] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.162410] thermal_sys: Registered thermal governor 'step_wise'
Sun Jun 16 22:11:07 2024 kern.warn kernel: [    0.164937] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.492348] PCI host bridge to bus 0000:00
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.496459] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.503370] pci_bus 0000:00: root bus resource [io  0x0000]
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.508938] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.517370] pci 0000:00:11.0: [168c:ff1d] type 00 class 0x020000
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.523385] pci 0000:00:11.0: reg 0x10: [mem 0x00000000-0x0000ffff]
Sun Jun 16 22:11:07 2024 kern.info kernel: [    0.529932] pci 0000:00:12.0: [168c:ff1d] type 00 class 0x020000
[... cut ...]
Sun Jun 16 22:12:43 2024 daemon.err mount_root: failed to sync jffs2 overlay
Sun Jun 16 22:12:45 2024 daemon.info procd: - init complete -
Sun Jun 16 22:12:45 2024 daemon.info urandom_seed[2626]: Seed saved (/etc/urandom.seed)
Sun Jun 16 22:12:49 2024 daemon.err uhttpd[1895]: [info] luci: accepted login on / for root from 192.168.1.147
Sun Jun 16 22:13:18 2024 daemon.info dnsmasq-dhcp[1]: DHCPINFORM(br-lan) 192.168.1.147 40:61:86:2a:70:c4
Sun Jun 16 22:13:18 2024 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.147 40:61:86:2a:70:c4 Admin-PC
```

lorand-horvath, 2024-06-19:
@DragonBluep Could this patch/hack be the cause? https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/patches-6.6/340-register_gpio_driver_earlier.patch

DragonBluep, 2024-06-19:
Maybe you can try this patch.
```
--- a/target/linux/ath79/dts/ar7161_netgear_wndr.dtsi
+++ b/target/linux/ath79/dts/ar7161_netgear_wndr.dtsi
@@ -19,8 +19,7 @@
 		compatible = "reset-leds";
 
 		usb_led {
-			function = LED_FUNCTION_USB;
-			color = <LED_COLOR_ID_GREEN>;
+			label = "green:usb";
 			resets = <&rst 12>;
 			trigger-sources = <&usb_ohci_port>, <&usb_ehci_port>;
 			linux,default-trigger = "usbport";

```

lorand-horvath, 2024-06-19:
@DragonBluep I see you reverted the usb_led changes from https://github.com/openwrt/openwrt/commit/e3ddfcc70c604c6eb9172a497df2291489dc1890#diff-464cddd5aab58bfbed5a335da5f1b53133b772ecce7ce01b849d6657dd53302a
I would like to try the patch, what is the easiest way of doing so? Do I have to create a local build? I'm afraid I can't do it at the moment.
Would it be possible for you to create a build for WNDRMAC v2, with LuCI included, if possible (if WAN is broken, I can't install it with opkg). I could test it.

lorand-horvath, 2024-06-19:
@DragonBluep I've just checked the log on a WR1043ND v3 which works fine with the latest 6.6 snapshot (it doesn't use the RTL8366S switch, though) and I also see the `gpio_button_hotplug: loading out-of-tree module taints kernel`, which is unexpected:
```
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.105054] usbcore: registered new interface driver usbfs
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.110821] usbcore: registered new interface driver hub
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.116348] usbcore: registered new device driver usb
Tue Jun 18 12:32:31 2024 kern.warn kernel: [    4.125964] gpio_button_hotplug: loading out-of-tree module taints kernel.
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.145498] ehci-platform 1b000000.usb: EHCI Host Controller
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.151283] ehci-platform 1b000000.usb: new USB bus registered, assigned bus number 1
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.159419] ehci-platform 1b000000.usb: irq 11, io mem 0x1b000000
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.195179] ehci-platform 1b000000.usb: USB 2.0 started, EHCI 1.00
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.202878] hub 1-0:1.0: USB hub found
Tue Jun 18 12:32:31 2024 kern.info kernel: [    4.207305] hub 1-0:1.0: 1 port detected
```

DragonBluep, 2024-06-19:
[Testing images](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1616685941)

lorand-horvath, 2024-06-19:
> [Testing images](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1616685941)

Thank you, I have flashed the WNDRMAC v2 sysupgrade you provided, but it's still broken. Here are some more details, maybe something will help:
1. The Power Led is turned off. It used to be green
2. The Wireless Leds (green 2.4G and blue 5G) are turned on, even though they should be completely off because the router was reset. In LuCI the WiFi interfaces are on, with regulatory set to CN (should be US after a reset).
3. The Switch page doesn't detect "RTL8366S" as it should, but only "switch0":
[image]

4. The WAN interface shows Error: Network device is not present:
[image]

5. The log still shows the same errors `rtl8366s: unknown chip id (0000)` and `rtl8366s: chip detection failed, err=-19`, here's the log:
```
Wed Jun 19 18:20:02 2024 kern.notice kernel: [    0.000000] Linux version 6.6.33 (runner@fv-az1426-451) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0-8009c6bb) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Wed Jun 19 10:19:50 2024
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] MIPS: machine is Netgear WNDRMAC v2
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Zone ranges:
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Early memory node ranges
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun 19 18:20:02 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Wed Jun 19 18:20:02 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Wed Jun 19 18:20:02 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] Memory: 122512K/131072K available (4604K kernel code, 574K rwdata, 744K rodata, 1236K init, 190K bss, 8560K reserved, 0K cma-reserved)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] CPU clock: 680.000 MHz
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5621354254 ns
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.007861] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.074117] pid_max: default: 32768 minimum: 301
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.086683] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.094015] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.111158] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.123271] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.133162] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.140027] pinctrl core: initialized pinctrl subsystem
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.147822] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Wed Jun 19 18:20:02 2024 kern.warn kernel: [    0.157198] gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Wed Jun 19 18:20:02 2024 kern.info kernel: [    0.479825] PCI host bridge to bus 0000:00
[... cut ...]
Wed Jun 19 18:20:48 2024 kern.info kernel: [   73.477057] br-lan: port 3(phy1-ap0) entered forwarding state
Wed Jun 19 18:20:48 2024 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun 19 18:20:48 2024 daemon.notice netifd: Wireless device 'radio1' is now up
Wed Jun 19 18:20:48 2024 daemon.notice netifd: Network device 'phy1-ap0' link is up
Wed Jun 19 18:20:48 2024 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ENABLED
Wed Jun 19 18:20:48 2024 daemon.notice hostapd: phy1-ap0: AP-ENABLED
```

namiltd, 2024-06-19:
GPIO doesn't work, everything else follows from that

DragonBluep, 2024-06-19:
> `gpio_button_hotplug: loading out-of-tree module taints kernel.`

Please test if removing the button hotplug package can help this.

> 
> @DragonBluep Could this patch/hack be the cause of the above? https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/patches-6.6/340-register_gpio_driver_earlier.patch

It's working on AR724x and AR934x. So I think nothing to do with it.

> 
> Or perhaps the USB phys patch (the gpio_button_hotplug tainting occurs during USB driver load)? https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/patches-6.6/700-phy-add-ath79-usb-phys.patch

What's the behavior of the USB LED?

> > [Testing images](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1616685941)
> 
> Thank you, I have flashed the WNDRMAC v2 sysupgrade you provided, but it's still broken. Here are some more details, maybe something will help:
> 
> 1. The Power Led is turned off. It used to be green

Does red led blinks during the boot?

> 2. The Wireless Leds (green 2.4G and blue 5G) are turned on, even though they should be completely off because the router was reset. In LuCI the WiFi interfaces are on, with regulatory set to CN (should be US after a reset).

It is expected.

PLease try this [image](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1617180032).

lorand-horvath, 2024-06-19:
> > `gpio_button_hotplug: loading out-of-tree module taints kernel.`
> 
> Please test if removing the button hotplug package can help this.

I have removed the gpio-button-hotplug package, restarted - I see no improvement.

> > Or perhaps the USB phys patch (the gpio_button_hotplug tainting occurs during USB driver load)? https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/patches-6.6/700-phy-add-ath79-usb-phys.patch
> 
> What's the behavior of the USB LED?

The USB LED has been off all the time.

> > 1. The Power Led is turned off. It used to be green
> 
> Does red led blinks during the boot?

The power led blinks orange only a few times on cold boot, then blinks green several times on and off, then it turns off completely and the WiFi leds turn on (I wasn't expecting the WiFi leds to be on after a full reset. It used to be that the radios were always off after reset).

> PLease try this [image](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1617180032).

Have flashed this image, but absolutely nothing changed, same problem, here's the log:
```
Wed Jun 19 20:47:26 2024 kern.notice kernel: [    0.000000] Linux version 6.6.33 (runner@fv-az1116-181) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0-7f2ac3ac) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Wed Jun 19 12:47:14 2024
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] MIPS: machine is Netgear WNDRMAC v2
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Zone ranges:
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Early memory node ranges
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Wed Jun 19 20:47:26 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Wed Jun 19 20:47:26 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Wed Jun 19 20:47:26 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] Memory: 122512K/131072K available (4604K kernel code, 574K rwdata, 744K rodata, 1236K init, 190K bss, 8560K reserved, 0K cma-reserved)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] CPU clock: 680.000 MHz
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5621354254 ns
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.007860] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.074117] pid_max: default: 32768 minimum: 301
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.086682] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.094015] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.111122] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.123205] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.133091] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.139955] pinctrl core: initialized pinctrl subsystem
Wed Jun 19 20:47:26 2024 kern.info kernel: [    0.147757] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... cut ...]
Wed Jun 19 20:48:07 2024 kern.info kernel: [   68.512085] br-lan: port 3(phy0-ap0) entered forwarding state
Wed Jun 19 20:48:07 2024 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun 19 20:48:07 2024 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
Wed Jun 19 20:48:07 2024 daemon.notice hostapd: phy0-ap0: AP-ENABLED
Wed Jun 19 20:48:07 2024 daemon.notice netifd: Wireless device 'radio0' is now up
Wed Jun 19 20:48:07 2024 daemon.notice netifd: Network device 'phy0-ap0' link is up
```

lorand-horvath, 2024-06-20:
@DragonBluep Do you think this will help? https://github.com/openwrt/openwrt/commit/150d0ceab7380538798df76a5d7d3c12a3632050

DragonBluep, 2024-06-20:
> then blinks green several times on and off, then it turns off completely and the WiFi leds turn on (I wasn't expecting the WiFi leds to be on after a full reset. It used to be that the radios were always off after reset).

No. It's a GPIO issue.

> The power led blinks orange only a few times on cold boot, then blinks green several times on and off, then it turns off completely

The green LED should never blink in theory.

I have no idea now. My TL-WR2543N with RTL8367 works fine. These ath79 SoCs use the same GPIO driver, so they should have the same behavior...

lorand-horvath, 2024-06-20:
> I have no idea now. My TL-WR2543N with RTL8367 works fine. These ath79 SoCs use the same GPIO driver, so they should have the same behavior...

There are differences between your WR2543N (AR7242) and the WNDR3700/3800 family (AR7161) in that the SOC and also the switch (RTL8366S) are older. I was specifically looking at this diagram: https://openwrt.org/toh/netgear/wndr3700#switch_ports_for_vlans

lorand-horvath, 2024-06-20:
@DragonBluep I'm not sure if it has anything to do with the issue, but I've found this: https://lore.kernel.org/netdev/87r19e5e8w.fsf@bang-olufsen.dk/T/
 Did they remove kernel support for RTL8366S specifically?
```
  		/* FIXME: add support for RTL8366S and more */
  		.compatible = "realtek,rtl8366s",
  		.data = NULL,
```

DragonBluep, 2024-06-20:
> @DragonBluep I'm not sure if it has anything to do with the issue, but I've found this: https://lore.kernel.org/netdev/87r19e5e8w.fsf@bang-olufsen.dk/T/ Was kernel support removed for RTL8366S specifically?
> 
> ```
>   		/* FIXME: add support for RTL8366S and more */
>   		.compatible = "realtek,rtl8366s",
>   		.data = NULL,
> ```

We are using downstream driver: https://github.com/openwrt/openwrt/blob/main/target/linux/generic/files/drivers/net/phy/rtl8366s.c

BTW, can you control power LEDs on/off status via LuCI?

lorand-horvath, 2024-06-20:
> BTW, can you control power LEDs on/off status via LuCI?

I'm not sure, never tried. What should I test?

However, I see some interesting changes: I've flashed the latest snapshot 26716 from this morning and now the green power led never lights up/doesn't blink when booting. Also the WiFi leds do not turn on at all and the WiFi interfaces are disabled, as should be.
So these seem to be some improvements. I wonder where are they coming from? Only major difference I see since yesterday is the new kernel 6.6.34 https://github.com/openwrt/openwrt/commit/a09a72d86dab6022181c9fb2e7a24ab76101782b

DragonBluep, 2024-06-21:
> > BTW, can you control power LEDs on/off status via LuCI?
> 
> I'm not sure, never tried. What should I test?

Just setting LED status "always on/off" in the OpenWrt web page.

lorand-horvath, 2024-06-22:
This issue is not caused by the LEDs.
The Realtek RTL8366S switch is not recognized/detected properly, it doesn't show any connectivity (no link).
[image]

I've reverted to 26620 based on kernel 6.1, it works just fine.

namiltd, 2024-06-22:
Change the title to: broken gpio driver. This is not a problem with the rtl8366s driver, but GPIO without which it is impossible to communicate with the chip.

lorand-horvath, 2024-06-22:
@namiltd It seems you've found the issue, indeed related to GPIOs 0-15, which have several differences: missing rtl8366s and also different behavior of the LEDs which I have noticed on the device: the LAN led that I've connected my PC to is not blinking anymore.

Please see below the output of the commands, which I listed for build 26620 (kernel 6.1 - running OK) and for latest 26741 (kernel 6.6 - broken).

`SNAPSHOT 26620 (kernel 6.1 - running OK)`
```
gpiochip0
gpiochip492
gpiochip502

----
/sys/class/gpio/gpiochip0/ngpio
/sys/class/gpio/gpiochip492/ngpio
/sys/class/gpio/gpiochip502/ngpio

----
gpiochip0: GPIOs 0-31, parent: platform/18040000.gpio, 18040000.gpio:
 gpio-0   (                    |orange:wps          ) out hi ACTIVE LOW
 gpio-1   (                    |orange:power        ) out hi ACTIVE LOW
 gpio-2   (                    |green:power         ) out lo ACTIVE LOW
 gpio-3   (                    |wps                 ) in  hi IRQ ACTIVE LOW
 gpio-4   (                    |green:wps           ) out hi ACTIVE LOW
 gpio-5   (                    |rtl8366s            ) in  hi
 gpio-6   (                    |green:wan           ) out hi ACTIVE LOW
 gpio-7   (                    |rtl8366s            ) in  hi
 gpio-8   (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-11  (                    |rfkill              ) in  hi IRQ ACTIVE LOW

gpiochip2: GPIOs 492-501, parent: pci/0000:00:12.0, ath9k-phy1:
 gpio-497 (                    |blue:wlan5g         ) out hi ACTIVE LOW

gpiochip1: GPIOs 502-511, parent: pci/0000:00:11.0, ath9k-phy0:
 gpio-507 (                    |green:wlan2g        ) out hi ACTIVE LOW
 gpio-508 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-509 (                    |fixed antenna group ) out hi
 gpio-510 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-511 (                    |fixed antenna group ) out hi

----
blue:wlan5g   green:wan     green:wps     orange:wps
green:power   green:wlan2g  orange:power
```

`SNAPSHOT 26741 (kernel 6.6 - broken)`
```
gpiochip0
gpiochip512
gpiochip522

----
/sys/class/gpio/gpiochip0/ngpio
/sys/class/gpio/gpiochip512/ngpio
/sys/class/gpio/gpiochip522/ngpio

----
gpiochip0: GPIOs 0-15, parent: platform/18040000.gpio, 18040000.gpio:
 gpio-0   (                    |orange:wps          ) in  lo ACTIVE LOW
 gpio-1   (                    |orange:power        ) in  lo ACTIVE LOW
 gpio-2   (                    |green:power         ) in  lo ACTIVE LOW
 gpio-3   (                    |wps                 ) in  lo IRQ ACTIVE LOW
 gpio-4   (                    |green:wps           ) in  lo ACTIVE LOW
 gpio-6   (                    |green:wan           ) in  lo ACTIVE LOW
 gpio-8   (                    |reset               ) in  lo IRQ ACTIVE LOW
 gpio-11  (                    |rfkill              ) in  lo IRQ ACTIVE LOW

gpiochip1: GPIOs 512-521, parent: pci/0000:00:12.0, ath9k-phy0:
 gpio-517 (                    |blue:wlan5g         ) out hi ACTIVE LOW

gpiochip2: GPIOs 522-531, parent: pci/0000:00:11.0, ath9k-phy1:
 gpio-527 (                    |green:wlan2g        ) out hi ACTIVE LOW
 gpio-528 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-529 (                    |fixed antenna group ) out hi
 gpio-530 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-531 (                    |fixed antenna group ) out hi

----
blue:wlan5g   green:usb     green:wlan2g  orange:power
green:power   green:wan     green:wps     orange:wps
```

namiltd, 2024-06-22:
cat /sys/class/gpio/gpiochip*/ngpio

namiltd, 2024-06-22:
Strange, all gpio ports 0-15 are set to "in lo".

lorand-horvath, 2024-06-22:
@namiltd I've updated the output from the commands, including the corrected 2nd one: cat /sys/class/gpio/gpiochip*/ngpio
Yes, gpios 0-15 are all in lo and missing 2 lines for rtl8366s. That's definitely the problem.

namiltd, 2024-06-22:
Interestingly, according to ar7100.dtsi there should be 16 gpio and for 6.1 it is 32

lorand-horvath, 2024-06-22:
> Interestingly, according to ar7100.dtsi there should be 16 gpio and for 6.1 it is 32

Not sure if that's causing it specifically, considering that only ports 0-11 are used anyway. What bothers me is that gpio-5 and gpio-7 are missing entirely. Also everything is set to `in lo` - that can't be correct.

namiltd, 2024-06-22:
rtl8366s requires "in hi"

namiltd, 2024-06-22:
Full log for kernel 6.1, please.

lorand-horvath, 2024-06-22:
Snapshot 26620 (kernel 6.1) log:
```
Tue Jun 18 12:16:22 2024 kern.notice kernel: [    0.000000] Linux version 6.1.92 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26593-b47fbca97f) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Tue Jun 11 12:27:47 2024
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] MIPS: machine is Netgear WNDRMAC v2
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Zone ranges:
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Early memory node ranges
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Tue Jun 18 12:16:22 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Tue Jun 18 12:16:22 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Tue Jun 18 12:16:22 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] Memory: 119728K/131072K available (6484K kernel code, 594K rwdata, 1492K rodata, 1212K init, 220K bss, 11344K reserved, 0K cma-reserved)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] CPU clock: 680.000 MHz
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5621354254 ns
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.007855] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.074099] pid_max: default: 32768 minimum: 301
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.079709] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.087016] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.097472] cblist_init_generic: Setting adjustable number of callback queues.
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.104701] cblist_init_generic: Setting shift to 0 and lim to 1.
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.114736] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.124614] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.131573] pinctrl core: initialized pinctrl subsystem
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.138675] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.145193] thermal_sys: Registered thermal governor 'step_wise'
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.457311] PCI host bridge to bus 0000:00
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.467437] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.474302] pci_bus 0000:00: root bus resource [io  0x0000]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.479868] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.488304] pci 0000:00:11.0: [168c:ff1d] type 00 class 0x020000
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.494327] pci 0000:00:11.0: reg 0x10: [mem 0x00000000-0x0000ffff]
Tue Jun 18 12:16:22 2024 kern.info kernel: [    0.500931] pci 0000:00:12.0: [168c:ff1d] type 00 class 0x020000
[... cut ...]
Tue Jun 18 12:16:50 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Jun 18 12:16:50 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Jun 18 12:16:50 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Jun 18 12:16:50 2024 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Tue Jun 18 12:16:50 2024 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Jun 18 12:16:50 2024 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
```

DragonBluep, 2024-06-22:
Another try [images](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1627546881)

lorand-horvath, 2024-06-22:
> Another try [images](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1627546881)

@DragonBluep Just tried it (new kernel 6.6.35), but the rtl8366s chip detection still fails. However, I see the power led is now turned on orange (not green as it was with kernel 6.1), the wifi interfaces are back to enabled again (and their leds turned on), but none of the leds (power, wifi, lan) are blinking at all, they're just staying fully lit up.
```
Sat Jun 22 19:36:16 2024 kern.notice kernel: [    0.000000] Linux version 6.6.35 (runner@fv-az1756-422) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0-48356d22) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 Sat Jun 22 11:36:05 2024
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] MIPS: machine is Netgear WNDRMAC v2
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Zone ranges:
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Early memory node ranges
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sat Jun 22 19:36:16 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sat Jun 22 19:36:16 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Sat Jun 22 19:36:16 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] Memory: 122512K/131072K available (4606K kernel code, 572K rwdata, 744K rodata, 1236K init, 190K bss, 8560K reserved, 0K cma-reserved)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] CPU clock: 680.000 MHz
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5621354254 ns
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.000002] sched_clock: 32 bits at 340MHz, resolution 2ns, wraps every 6316128254ns
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.007861] Calibrating delay loop... 452.19 BogoMIPS (lpj=2260992)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.074117] pid_max: default: 32768 minimum: 301
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.086677] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.094016] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.111062] RCU Tasks Trace: Setting shift to 0 and lim to 1 rcu_task_cb_adjust=1.
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.123102] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.132987] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.139847] pinctrl core: initialized pinctrl subsystem
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.147646] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.470135] PCI host bridge to bus 0000:00
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.474250] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.481170] pci_bus 0000:00: root bus resource [io  0x0000]
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.486737] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.495335] pci 0000:00:11.0: [168c:ff1d] type 00 class 0x020000
Sat Jun 22 19:36:16 2024 kern.info kernel: [    0.501357] pci 0000:00:11.0: reg 0x10: [mem 0x00000000-0x0000ffff]
[... cut ...]
Sat Jun 22 19:36:51 2024 kern.info kernel: [   60.937949] br-lan: port 3(phy1-ap0) entered forwarding state
Sat Jun 22 19:36:51 2024 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Jun 22 19:36:51 2024 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->ENABLED
Sat Jun 22 19:36:51 2024 daemon.notice netifd: Network device 'phy1-ap0' link is up
Sat Jun 22 19:36:51 2024 daemon.notice hostapd: phy1-ap0: AP-ENABLED
Sat Jun 22 19:36:51 2024 daemon.notice netifd: Wireless device 'radio1' is now up
```

namiltd, 2024-06-22:
Always add the output of "cat /sys/kernel/debug/gpio" to each attached log. It will be easier for us to see the result

markmentovai, 2024-06-23:
Give #15779 a try. It does the trick for me.

DragonBluep, 2024-06-23:
> Give #15779 a try. It does the trick for me.

@lorand-horvath reports it doesn't work...

Already included in
> Another try [images](https://github.com/DragonBluep/Actions/actions/runs/9298412061/artifacts/1627546881)

markmentovai, 2024-06-23:
You’re right, somehow I managed to substitute an old 6.1 image. Doesn’t work.

markmentovai, 2024-06-23:
OK, I’ve got something for real this time.

Noting that `ngpio` was reported as 32 in 6.1 and 16 (matching [ar7100.dtsi](https://github.com/openwrt/openwrt/blob/b07b8aade7174f915932dc16120bfe0c02de7133/target/linux/ath79/dts/ar7100.dtsi#L70)) in 6.6 (also mentioned [here](https://github.com/openwrt/openwrt/issues/15739#issuecomment-2183976414)), I found that the change was due to [torvalds/linux 55b2395e4e92](https://github.com/torvalds/linux/commit/55b2395e4e92adc492c6b30ac109eb78250dcd9d). Reverting that, GPIOs work correctly again on a WNDR3800.

```
# uname -r
6.6.35
# grep '"id"' /tmp/board.json
		"id": "netgear,wndr3800",
# cat /sys/class/gpio/gpiochip0/ngpio
32
# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-31, parent: platform/18040000.gpio, 18040000.gpio:
 gpio-0   (                    |orange:wps          ) out hi ACTIVE LOW
 gpio-1   (                    |orange:power        ) out hi ACTIVE LOW
 gpio-2   (                    |green:power         ) out lo ACTIVE LOW
 gpio-3   (                    |wps                 ) in  hi IRQ ACTIVE LOW
 gpio-4   (                    |green:wps           ) out hi ACTIVE LOW
 gpio-5   (                    |rtl8366s            ) in  hi 
 gpio-6   (                    |green:wan           ) out hi ACTIVE LOW
 gpio-7   (                    |rtl8366s            ) in  hi 
 gpio-8   (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-11  (                    |rfkill              ) in  hi IRQ ACTIVE LOW

gpiochip1: GPIOs 512-521, parent: pci/0000:00:11.0, ath9k-phy0:
 gpio-517 (                    |green:wlan2g        ) out lo ACTIVE LOW
 gpio-518 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-519 (                    |fixed antenna group ) out hi 
 gpio-520 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-521 (                    |fixed antenna group ) out hi 

gpiochip2: GPIOs 522-531, parent: pci/0000:00:12.0, ath9k-phy1:
 gpio-527 (                    |blue:wlan5g         ) out hi ACTIVE LOW
 gpio-528 (                    |netgear:antenna:wlan) out lo 
 gpio-529 (                    |netgear:antenna:wlan) out hi 
 gpio-530 (                    |netgear:antenna:wlan) out hi 
 gpio-531 (                    |netgear:antenna:wlan) out lo 
# dmesg | grep -Ei 'rtl8366s|ag71xx'
[    1.220658] ag71xx 19000000.eth: connected to PHY at fixed-0:00 [uid=00000000, driver=Generic PHY]
[    1.230335] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: rgmii
[    1.569470] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
[    1.979458] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
[    2.319449] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
[    6.794011] Realtek RTL8366S ethernet switch driver version 0.2.2
[    6.800229] rtl8366s rtl8366s: cannot find mdio node phandle
[    6.805892] rtl8366s rtl8366s: using GPIO pins 5 (SDA) and 7 (SCK)
[    6.812175] rtl8366s rtl8366s: RTL8366 ver. 1 chip found
[    6.962010] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
[    7.344757] ag71xx 1a000000.eth: connected to PHY at rtl8366s:04 [uid=001cc960, driver=Generic PHY]
[    7.355162] eth1: Atheros AG71xx at 0xba000000, irq 5, mode: rgmii
[   98.364903] ag71xx 19000000.eth eth0: entered allmulticast mode
[   98.371149] ag71xx 19000000.eth eth0: entered promiscuous mode
```

That commit is good and it fixes a real bug, so it shouldn’t be reverted. But now it should be possible to debug the difference in behavior with and without that commit, and fix it. At first glance, it looks like `gpio_chip::bgpio_bits`, which was 32 in 6.1, is now going to be calculated (based on `gpio_chip::ngpio`) as 16 instead. That appears to be enough to cause this problem.

One solution is to “lie” in ar7100.dtsi by setting `ngpios` to 32, but that won’t work because of [this check in gpio-ath79.c](https://github.com/torvalds/linux/blob/e91d0f05e66afba9a3804bccf3d0a4310bb30024/drivers/gpio/gpio-ath79.c#L255). Seems like too much hacking around.

Another idea is to partially preserve gpio-mmio.c’s historic behavior of setting `bgpio_bits` by taking the larger of the traditional and newly computed values. That could look like https://github.com/markmentovai/openwrt/commit/f31c67bbb888672e222a8b369db9d3cc4049db35.

And that works too, without having to misrepresent `ngpios`. That same WNDR3800 with this patch:

```
# uname -r
6.6.35
# grep '"id"' /tmp/board.json
		"id": "netgear,wndr3800",
# cat /sys/class/gpio/gpiochip0/ngpio
16
# cat /sys/kernel/debug/gpio
[... 16 lines cut ...]
 gpio-520 (                    |netgear:antenna:wlan) out hi 
 gpio-521 (                    |netgear:antenna:wlan) out lo 

gpiochip2: GPIOs 522-531, parent: pci/0000:00:11.0, ath9k-phy1:
 gpio-527 (                    |green:wlan2g        ) out hi ACTIVE LOW
 gpio-528 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-529 (                    |fixed antenna group ) out hi 
 gpio-530 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-531 (                    |fixed antenna group ) out hi 
# dmesg | grep -Ei 'rtl8366s|ag71xx'
[    1.230774] ag71xx 19000000.eth: connected to PHY at fixed-0:00 [uid=00000000, driver=Generic PHY]
[    1.240462] eth0: Atheros AG71xx at 0xb9000000, irq 4, mode: rgmii
[    1.579587] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
[... cut ...]
[    6.816065] rtl8366s rtl8366s: using GPIO pins 5 (SDA) and 7 (SCK)
[    6.822352] rtl8366s rtl8366s: RTL8366 ver. 1 chip found
[    6.994294] ag71xx 1a000000.eth: Could not connect to PHY device. Deferring probe.
[    7.413851] ag71xx 1a000000.eth: connected to PHY at rtl8366s:04 [uid=001cc960, driver=Generic PHY]
[    7.424291] eth1: Atheros AG71xx at 0xba000000, irq 5, mode: rgmii
[   98.749695] ag71xx 19000000.eth eth0: entered allmulticast mode
[   98.755895] ag71xx 19000000.eth eth0: entered promiscuous mode
```

lorand-horvath, 2024-06-23:
> Always add the output of "cat /sys/kernel/debug/gpio" to each attached log. It will be easier for us to see the result

@namiltd Nothing has really changed with the build provided by @DragonBluep yesterday in https://github.com/openwrt/openwrt/issues/15739#issuecomment-2184018103 , except for the wireless gpios `green:wlan2g` and `green:wlan5g` changing from `out hi` to `out lo` (as I mentioned above, both WiFi interfaces and their leds got turned on by default, even after reset).
```
gpiochip0: GPIOs 0-15, parent: platform/18040000.gpio, 18040000.gpio:
 gpio-0   (                    |orange:wps          ) in  lo ACTIVE LOW
 gpio-1   (                    |orange:power        ) in  lo ACTIVE LOW
 gpio-2   (                    |green:power         ) in  lo ACTIVE LOW
 gpio-3   (                    |wps                 ) in  lo IRQ ACTIVE LOW
 gpio-4   (                    |green:wps           ) in  lo ACTIVE LOW
 gpio-6   (                    |green:wan           ) in  lo ACTIVE LOW
 gpio-8   (                    |reset               ) in  lo IRQ ACTIVE LOW
 gpio-11  (                    |rfkill              ) in  lo IRQ ACTIVE LOW

gpiochip1: GPIOs 512-521, parent: pci/0000:00:11.0, ath9k-phy0:
 gpio-517 (                    |green:wlan2g        ) out lo ACTIVE LOW
 gpio-518 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-519 (                    |fixed antenna group ) out hi
 gpio-520 (                    |fixed antenna group ) out lo ACTIVE LOW
 gpio-521 (                    |fixed antenna group ) out hi

gpiochip2: GPIOs 522-531, parent: pci/0000:00:12.0, ath9k-phy1:
 gpio-527 (                    |blue:wlan5g         ) out lo ACTIVE LOW
```

lorand-horvath, 2024-06-23:
> Always add the output of "cat /sys/kernel/debug/gpio" to each attached log. It will be easier for us to see the result

@namiltd I've updated https://github.com/openwrt/openwrt/issues/15739#issuecomment-2184030860

> OK, I’ve got something for real this time.
>
> Noting that `ngpio` was reported as 32 in 6.1 and 16 (matching [...]

@markmentovai Great find! I'd like to test a build (LuCI included) on WNDRMAC v2, if possible.

DragonBluep, 2024-06-23:
@markmentovai Can you test if GCC12 or GCC14 can fix this issue? The code should be fine. And change ngpios from 16 to 17 should also workaround it. BTW, AR71xx chips only have 12 gpios.

namiltd, 2024-06-23:
Ports work when their number is incorrect, strange. I think we should find out why and not force an inappropriate number of them.

DragonBluep, 2024-06-23:
I guess `bgpio_read16` and `bgpio_write16` don‘t work correctly on ath79 target.
https://github.com/torvalds/linux/blob/ffc253263a1375a65fa6c9f62a893e9767fbebfa/drivers/gpio/gpio-mmio.c#L435-L436

markmentovai, 2024-06-23:
https://github.com/openwrt/openwrt/issues/15739#issuecomment-2184757919:
> I'd like to test a build (LuCI included) on WNDRMAC v2, if possible.

I don't use or build with LuCI, so I'd have no way of knowing whether I was giving you something that works.

https://github.com/openwrt/openwrt/issues/15739#issuecomment-2184911644:
> I think we should find out why and not force an inappropriate number of them.

https://github.com/openwrt/openwrt/issues/15739#issuecomment-2184757919:
> BTW, AR71xx chips only have 12 gpios.

Thanks! I don’t have data sheets so I didn’t know that.

In that case, we shouldn't have to lie in ar7100.dtsi, we should be able to set it to `<12>` there, and deal with driving the hardware correctly in code.

Two theories now:
 1. The ar71xx GPIO registers may not be usable with 16-bit accesses and may require 32.
 2. There's a high bit (or bits) in there functioning as a disable, set by default. The 32-bit write would clear it, but the 16-bit write would leave it alone.

markmentovai, 2024-06-23:
I’m also curious about whether other ar71xx-based or ath79-family devices are affected. It doesn’t seem that any of this bug would be WNDR-specific, although the WNDR controls its switch over GPIO so it may be more noticeable there. But many devices do control at least LEDs via GPIO, which may be a bit less obvious if broken. Has anyone checked any other ath79-family devices for GPIO function since switching to kernel 6.6?

lorand-horvath, 2024-06-23:
> Has anyone checked any other ath79-family devices for GPIO function since switching to kernel 6.6?

@markmentovai I've been running the latest snapshot on a WR1043ND v3 ever since ath79 devices were moved to kernel 6.6, i.e. build 26657 from June 17. Everything seems fine, leds and whatnot.

markmentovai, 2024-06-23:
https://github.com/openwrt/openwrt/issues/15739#issuecomment-2185065828:
> > Has anyone checked any other ath79-family devices for GPIO function since switching to kernel 6.6?
> 
> @markmentovai I've been running the latest snapshot on a WR1043ND v3 ever since ath79 devices were moved to kernel 6.6, i.e. build 26654 from June 17. Everything seems fine, leds, switch, wifi and whatnot.
> 
> I wonder if this issue occurs only on certain ar71xx based units...

Thanks for checking!

tplink,tl-wr1043nd-v3 is based on qca9558, which is newer than the ar7161-based WNDR family. Even though they both use the same basic GPIO driver, crucially the WR1043ND has more than 16 GPIOs ([qca955x.dtsi declares 24](https://github.com/openwrt/openwrt/blob/bb6d11e53487e397b1c1cec1767b593dd1025c18/target/linux/ath79/dts/qca955x.dtsi#L93) and [qca9558_tplink_tl-wr1043nd.dtsi uses indices up to 21](https://github.com/openwrt/openwrt/blob/74f2df9dbccd9840edb25d55ad70d58e3ed8303c/target/linux/ath79/dts/qca9558_tplink_tl-wr1043nd.dtsi#L74)), so even with kernel 6.6, gpio-mmio would be interfacing with the GPIO registers using 32-bit accesses.

The only other ath79 device I have is a tplink,archer-c7-v2, which is also qca9558-based, and so I wouldn’t expect it to be susceptible either.

Only ar7100.dtsi has ngpio ≤ 16, so only the first-generation ar71xx-based devices would be using 16-bit accesses to the GPIO registers since switching to kernel 6.6, while they were previously using 32-bit accesses. The rest of the ath79 family continues to use 32-bit accesses. I would only expect those first-generation ar71xx-based devices (ar7141, ar7161) to be affected, the rest of ath79 should be fine.

lorand-horvath, 2024-06-23:
@markmentovai For future reference, here are the GPIO command results for WR1043ND v3 running snapshot 26748 (kernel 6.6)
```
root@OpenWrt:~# ls -1a /sys/class/gpio
.
..
export
gpiochip0
tp-link:power:usb
unexport

root@OpenWrt:~# cat /sys/class/gpio/gpiochip*/ngpio
24

root@OpenWrt:~# cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-23, parent: platform/18040000.gpio, 18040000.gpio:
 gpio-12  (                    |green:wlan          ) out lo ACTIVE LOW
 gpio-15  (                    |green:usb           ) out hi ACTIVE LOW
 gpio-16  (                    |Reset button        ) in  hi IRQ ACTIVE LOW
 gpio-17  (                    |RFKILL button       ) in  hi IRQ ACTIVE LOW
 gpio-18  (                    |green:wps           ) out hi ACTIVE LOW
 gpio-19  (                    |green:system        ) out lo ACTIVE LOW
 gpio-21  (                    |tp-link:power:usb   ) out hi

root@OpenWrt:~# ls /sys/class/leds
ath9k-phy0    green:system  green:usb     green:wlan    green:wps
```

willmo, 2024-06-23:
> Another idea is to partially preserve gpio-mmio.c’s historic behavior of setting `bgpio_bits` by taking the larger of the traditional and newly computed values. That could look like [markmentovai@f31c67b](https://github.com/markmentovai/openwrt/commit/f31c67bbb888672e222a8b369db9d3cc4049db35).

I think something like this (or just deleting the `else`?) is the correct solution. The current logic effectively ignores `sz` if `ngpios` is provided, and that can't be right. Even if the GPIO chip supports register accesses smaller than `sz`, it's broken with big-endian registers, where higher-numbered GPIO bits are at lower addresses, so performing smaller accesses at the same address means accessing only the higher-numbered GPIOs. So in this case, the CPU is probably accessing the register bits for nonexistent GPIOs 16-31 instead of 0-15.

markmentovai, 2024-06-24:
Yup, I just came here to say the same thing. Of course this is broken on ath79, which is big-endian MIPS. The bits aren’t even going to the right place.

namiltd, 2024-06-24:
Maybe in gpio-ath79.c lines 265 to 271:
```
	err = bgpio_init(&ctrl->gc, dev, 4,
			ctrl->base + AR71XX_GPIO_REG_IN,
			ctrl->base + AR71XX_GPIO_REG_SET,
			ctrl->base + AR71XX_GPIO_REG_CLEAR,
			oe_inverted ? NULL : ctrl->base + AR71XX_GPIO_REG_OE,
			oe_inverted ? ctrl->base + AR71XX_GPIO_REG_OE : NULL,
			0);
```
should be:
```
	err = bgpio_init(&ctrl->gc, dev, 4,
			ctrl->base + AR71XX_GPIO_REG_IN,
			ctrl->base + AR71XX_GPIO_REG_SET,
			ctrl->base + AR71XX_GPIO_REG_CLEAR,
			oe_inverted ? NULL : ctrl->base + AR71XX_GPIO_REG_OE,
			oe_inverted ? ctrl->base + AR71XX_GPIO_REG_OE : NULL,
			1);
```

KanjiMonster, 2024-06-24:
> @DragonBluep Is it possible to test a build with your latest version of the gpio patch [DragonBluep@50b3865](https://github.com/DragonBluep/openwrt/commit/50b3865fd00f5235e162d350d84fc7d7967b2ea8) ?

Looks like a reasonable change, though I would think that @namiltd's change would also be required. How did this driver ever work with LE byte order accesses? Also the readl()/writel() accesses in `ath79_gpio_read()`/`ath79_gpio_write()` look wrong.

willmo, 2024-06-24:
That seems like a highly theoretical problem, since this driver is only used on platforms with big-endian CPUs, right? (And if there were an ath79-ish platform with a little-endian CPU, it's possible that it would also have little-endian GPIO registers.)

KanjiMonster, 2024-06-24:
> That seems like a highly theoretical problem, since this driver is only used on platforms with big-endian CPUs, right? (And if there were an ath79-ish platform with a little-endian CPU, it's possible that it would also have little-endian GPIO registers.)

Why a theoretical problem? `readl()`/`writel()` when used in a big endian kernel swap the byte order, and without the flag generic gpio will do the same.

So e.g. instead of accessing bit 3 for GPIO 3, you would actually access bit 27 (`0x00000008` becomes `0x08000000` on accesses).

willmo, 2024-06-24:
I see, that is confusing...I see that other ath79 platform devices seem to use the `__raw` functions to avoid the byte swap so I agree that is odd...

namiltd, 2024-06-24:
> So e.g. instead of accessing bit 3 for GPIO 3, you would actually access bit 27 (`0x00000008` becomes `0x08000000` on accesses).

One byte has the same bit order (left to right) no matter if we're talking about little endian or big endian systems.

namiltd, 2024-06-24:
The GPIO port numbers are correct. Patch applied: https://github.com/DragonBluep/openwrt/commit/00344c9b559c15b199215817563c0d826645110c

[24 more replies]

closed 2024-07-04 by openwrt-bot